dresden-elektronik/deconz-rest-plugin #4013: Device request for NYCE NCZ-3041-HA motion sensors

glc650, 2020-12-23:
## Device

- Product name: Motion Sensor
- Manufacturer: NYCE
- Model identifier: 3041
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

wvuyk, 2021-01-06:
Is there anything missing for this request?

Mimiix, 2021-01-07:
Seems correct to me

SwoopX, 2021-02-15:
Added in 2.10.00-beta

closed 2021-02-15 by SwoopX