dresden-elektronik/deconz-rest-plugin #4014: Device request for Kwikset 916 door lock

glc650, 2020-12-23:
## Device

- Product name: SmartCode 916 touchscreen Electronic Deadbolt
- Manufacturer: Kwikset
- Model identifier: SMARTCODE_DEADBOLT_10T
- Device type : Please remove all unrelated device types. 
  - Lock

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2020-12-24:
Hello, I have started the integration for a device that look like yours https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3152

But without return, do you have an unix machine to test code modifications ?

glc650, 2020-12-24:
Closest thing I have to a unix machine is a Macbook Air.

Smanar, 2020-12-25:
And with real OS not hassio ?
Macbook Air run on unix ?

wvuyk, 2021-01-25:
Is this not possible to create it for the REST API?

Smanar, 2021-01-25:
Nope, It s a long story, but I m rewriting all the code that support doorlock, to create a ZHADoorlock device.
So it s a big change, and I can make a PR without someone to do some test.

The code on this issue is out of date.

closed 2021-03-05 by github-actions[bot]