dresden-elektronik/deconz-rest-plugin #4016: Sonoff ZBmini not working in Phoson and HA

minke890, 2020-12-24:
## Describe the bug
I have issues with 4 of my 5 Sonoff Zigbee mini switches (ZBmini) in Deconz, also in combination with HA. The ZBmini are recognized in Deconz in phoscon and in the VNC viewer and also in HA. Also the connected regular switches connected to the ZBmini’s work fine (you can hear the relay). However, the switching in phoscon and HA does not works:

Switching in HA or Deconz to on or off only works one time.
Switching triggers multiple events in HA history and in Deconz. Example:
2873: 14:10:49:898 Websocket 172.29.0.1:41700 send message: {"e":"changed","id":"35","r":"lights","state":{"alert":null,"on":true,"reachable":true},"t":"event","uniqueid":"84:71:27:ff:fe:93:13:58-01"} (ret = 140)
2874: 14:10:49:899 Websocket 192.168.2.20:54220 send message: {"e":"changed","id":"35","r":"lights","state":{"alert":null,"on":true,"reachable":true},"t":"event","uniqueid":"84:71:27:ff:fe:93:13:58-01"} (ret = 140)
2875: 14:10:49:900 Websocket 192.168.2.20:54434 send message: {"e":"changed","id":"35","r":"lights","state":{"alert":null,"on":true,"reachable":true},"t":"event","uniqueid":"84:71:27:ff:fe:93:13:58-01"} (ret = 140)

One of my ZBmini is working, but still triggering multiple events. I do not know why this ZBmini is working. I do not see a difference in Deconz.

My problems looks the same as described here: #3632 and https://community.home-assistant.io/t/zigbee2mqtt-switch-automatically-toggles-from-off-to-on-after-1-second/244833

## Steps to reproduce the behavior
The issue is seen for 4 out of the 5 Sonoff ZBmini's. Restarting de HA and deCONZ containers does not help. Same for restarting the host (Qnap NAS). There also is no difference between deCONZ 2.7.1 and 2.8

## Expected behavior
In HA and deCONZ is should be possible to swith the ZBmini on and off (independent of the regular switch connected to the ZB mini).

## Screenshots
For screenshots see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3632

For me the screenshots form issue 3632 are the same. Only the Date code is difference (20200927) and the Power Source (unkown).

Additional screenshot:

[image]

Please let me know if any other screenshot is needed.

## Environment
 - Host system: Qnap 253be NAS
 - Running method: HA core Docker container / Marthoc Docker container 
 - Firmware version: 26680700
 - deCONZ version: 2.7.1
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? yes, DSMR to P1 cable. And Zwave stick using extension cable.

## deCONZ Logs
I do no see errors in the debug INFO, ERROR,ERROR_L2,APS,APS_L2 (besides the multiple entries shown above).

Also see https://pastebin.com/jNQzdf2E

During the log I have tried switching the light (loggerkamerlamp) in phoscon en the regular wall light switch attached to the ZBmini.

## Additional context
According HA I have 90 nodes and 150 entities.

minke890, 2020-12-24:
Today I have made a new debug INFO, ERROR,ERROR_L2,APS,APS_L2

see https://pastebin.com/RS2RmbEv

During the log I have switched the lights logeerkamerlamp and kluskamerlamp in HA end using the regular wall light switch attached to the ZBmini. Strangely they now both work, but still showing multiple entries in HA and in the log.

Addition

SwoopX, 2020-12-24:
Thanks for the logs. You mean like those 2 here?

```
10:58:09:097 ZCL attribute report 0x842E14FFFE9E1404 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
10:58:09:098 	payload: 00002801
10:58:09:098 APS-DATA.request id: 117, addrmode: 0x02, addr: 0x493b, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
10:58:09:100 Websocket 172.29.0.1:58844 send message: {"e":"changed","id":"33","r":"lights","state":{"alert":null,"on":false,"reachable":true},"t":"event","uniqueid":"84:2e:14:ff:fe:9e:14:04-01"} (ret = 141)
10:58:09:100 enqueue event event/checkgroupanyon for /groups/65520
10:58:09:138 APS-DATA.indication srcAddr: 0x493b, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -43
10:58:09:139 	asdu: 086f0a00001001
10:58:09:140 Node data 0x842e14fffe9e1404 profileId: 0x0104, clusterId: 0x0006
10:58:09:140 0x842E14FFFE9E1404: update ZCL value 0x01/0x0006/0x0000 after 0 s
10:58:09:141 enqueue event state/on for /lights/33
10:58:09:142 ZCL attribute report 0x842E14FFFE9E1404 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
10:58:09:142 	payload: 00001001
10:58:09:143 APS-DATA.request id: 119, addrmode: 0x02, addr: 0x493b, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
10:58:09:144 Websocket 172.29.0.1:58844 send message: {"e":"changed","id":"33","r":"lights","state":{"alert":null,"on":true,"reachable":true},"t":"event","uniqueid":"84:2e:14:ff:fe:9e:14:04-01"} (ret = 140)
```

The remarkable thing is that the device seems to send 2 attribute reports shortly after another. The first payload (`00002801`) contains an incorrect data type being signed 8-bit integer, the second (`00001001`) the correct data type boolean.

So on the one hand, the device does sometrong odd, on the other hand, we should evaluate to let only bool values through...

thundergreen, 2020-12-26:
I am facing same issue her. Thught i WOuld give it a try before buyinf sonnof zigbee bridge and flash this with tasmota :(

manup, 2020-12-27:
I guess this is a case where it's better to have the real device for testing. I'll put it on the order list for January.
The two faced attribute report looks really strange, note that our unholy group0 `0xfff0` is also at play here `/groups/65520`.

The group0 bindings will be removed in the next release, and /group/0 will use a normal broadcast to all routers, however every other group to which the device is added leads to the same effect.

KodeCR, 2020-12-27:
>note that our unholy group0 0xfff0 is also at play here /groups/65520.
>The group0 bindings will be removed in the next release, and /group/0 will use a normal broadcast to all routers, however every other group to which the device is added leads to the same effect.

@manup, note that I implemented it like that because of the suggestion by you and @ebaauw in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1005. (and only because I wanted to add Hue scenes support https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1471, which still isn't merged...).

I only recently found out that group 65520 is invalid according to the zigbee specs (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3744#issuecomment-733237346), and that sometimes sensors were added to the group, perhaps causing quite a few of the group0 issues.

minke890, 2020-12-27:
Thanks for your efforts so far! I am not in expert in de debug logs nor in the deCONZ api. @manup, I live in the netherlands, if you also live there I am happy to supply you the ZBmini to speed things up.

eklex, 2020-12-28:
I am seeing the same problem.
However I noticed that if I open the On/Off Cluster in Deconz, and execute manually the "On", "Off" or "Toggle" commands, the device behaves properly. It is only when I use Phoscon or HA that the device doesn't work as expected.

raddi, 2020-12-28:
@manup I live in Germany and have 3 Sonoff ZBMinis laying around. I would send you one over if this helps.

minke890, 2020-12-28:
> I am seeing the same problem.
> However I noticed that if I open the On/Off Cluster in Deconz, and execute manually the "On", "Off" or "Toggle" commands, the device behaves properly. It is only when I use Phoscon or HA that the device doesn't work as expected.

That is exactly what I noticed as well.

manup, 2020-12-30:
> @manup I live in Germany and have 3 Sonoff ZBMinis laying around. I would send you one over if this helps.

Thanks, I've already put them on the order list, should arrive in next two weeks.

If the switch works in the deCONZ Cluster Info panel it really seems to be more an issue in the plugin handling of it.

JCSCatena, 2021-01-01:
Hi,

Same bug in my HA installation.
My environment:
- RPi 4 4Gb
- HAOS 5.9 (same problem with 5.8)
- Conbee 2 Firmware: 26660700, Version: 2.07.01 / 12/8/2020

I have 4 ZBMINIs and I have exactly the same problem with all of them. If more information is needed, please tell me what you need.

Greetings and thanks to the team for being attentive to the problem and hopefully it will be resolved soon.

ads1230, 2021-01-05:
Had some issues setting these up but finally got them to work.
I found after pairing, the device shows in HA but has issues with control (immediately toggles off).
I found the following worked to get the devices set up and working
1. Restart HA, 
2. Stop and start deconz 
3. Reset the device whilst paired to deconz (holding down the physical button on the sonoff for 10s) 
4. Then repair with deconz.

Issue now is the weird flickering of states
https://user-images.githubusercontent.com/53495346/103662757-3a67b400-4f68-11eb-9ce1-4ad68891dcb2.mov

JCSCatena, 2021-01-05:
Hi;

I don't know how, but today I have (physically) connected two of the ZBMINI devices and reconnected them, and now they work perfectly for me. I don't know what could have happened to make them work for me now. :-(

But hey, it was just that, commenting that with the same software versions the devices are working correctly.

I will monitor and comment again if they stop working.

A greeting.

andrejem, 2021-01-05:
I am using 2.09.0 and I am able to control the ZBMINI. But it is recognized as light/bulb and I am not able to see the status. It switches to off in the UI and json answer. But the power is still on. 

@JCSCatena Do you see the correct status of the ZBMINI?

JCSCatena, 2021-01-05:
> I am using 2.09.0 and I am able to control the ZBMINI. But it is recognized as light/bulb and I am not able to see the status. It switches to off in the UI and json answer. But the power is still on.
> 
> @JCSCatena Do you see the correct status of the ZBMINI?

In phoscon appears like this.....

As Light
[image]

But in HA appears like this....
[image]

yes, I see the correct status.

A greetings.

thundergreen, 2021-01-06:
I switched to ZHA with flashed sonoff ZigBee bridge and never got problems
again. Also my light bulbs and ithe ZigBee stuff without issues. I still
use deconz for some sensors but replaced hue finally complete with sonoff

Jan. 2021, 22:55:

jonnycastaway, 2021-01-09:
Ok, here's another one with same problem.
ZBmini is connected and can be switched on and off. But the state is switching immediately to off afer switching to on.
So the status of device is always off in Phoscon. And as consequence also in my home automation system.

retoziegler, 2021-01-13:
Have the same issue with both my new ZBMINIs.
I discovered that an update in HA can be forced by force-reading the attributes of cluster On/Off in deCONZ.
So it seems that the attribute OnOff (0x0000) is not always auto-updated in deCONZ when changed in the device (sometimes it works in my case).
[image]
[image]

Whatsek, 2021-01-17:
> Had some issues setting these up but finally got them to work.
> I found after pairing, the device shows in HA but has issues with control (immediately toggles off).
> I found the following worked to get the devices set up and working
> 
> 1. Restart HA and deconz
> 2. Reset the device whilst paired to deconz (holding down the physical button on the sonoff for 10s)
> 3. Then repair with deconz.
> 
> conbee II
> 2.07.01
> Firmware: 26660700
> Latest HA on Rasp Pi 3b+
> 
> Issue now is the weird flickering of states
> https://user-images.githubusercontent.com/53495346/103662757-3a67b400-4f68-11eb-9ce1-4ad68891dcb2.mov

How do you do step 3? I cant find repair in deconz.

jmpaiva, 2021-01-17:
> How do you do step 3? I cant find repair in deconz.
i believe he meant "re-pair" and not "repair".

I'm having the same issue and will try this

Whatsek, 2021-01-17:
> > How do you do step 3? I cant find repair in deconz.
> > i believe he meant "re-pair" and not "repair".

Ah, that makes sense. I tried this, but unfortunately it didn't work for me. I can switch them on, but the status is then off again, so I cant turn them really off. Turning off with a switch works fin. Then I can switch them on again with home assistant. (but never off without the physical button)

jmpaiva, 2021-01-17:
> Ah, that makes sense. I tried this, but unfortunately it didn't work for me. I can switch them on, but the status is then off again, so I cant turn them really off. Turning off with a switch works [...]

well, I tried but I screwed something up, now conbee is having some issue, asking for a reboot :/

On step 3, do you re-pair in deconz or on phoscon? If you do it on deconz (from inside HA) how do you do it?

ads1230, 2021-01-17:
> well, I tried but I screwed something up, now conbee is having some issue, asking for a reboot :/
>
> On step 3, do you re-pair in deconz or on phoscon? If you do it on deconz (from inside HA) how do [...]

I used phoscon.
The reset of the sonoff should break the pairing, and will add it as a new device.
Any on/off issues I was having was corrected by a HA reboot.
Not exactly an exact science on my behalf but I did get 3 devices to work.

jmpaiva, 2021-01-17:
I 'm without luck.

ZBmini works great within deCONZ but either HA or Phoscon just won't work. it never changes state properly

I'm running latest HA on a pi4, deCONZ 2.07.01, fw 26660700

spiff42, 2021-01-22:
Sorry, let me put this in the OPEN issue.
I have the same issue. I just wanted to note that in Phoscon my ZBMINI is reporting:
Vendor Heiman
Model 01MINIZB
Version 20200927
I bought it on AliExpress, so I don't know if the device is genuine or a knock-off.

JCSCatena, 2021-01-22:
I have two ZBMINI configured correctly, but now I have bought 5 more and I have the problem that is described in the BUG, ​​that is, if the device is off, Phoscon or HA manages to turn it on, but then it is impossible to turn it off.

From deConz if I can turn it off and on every time without problems.

Hopefully it will be solved soon.

Another thing that happens is that I have added the 5 devices to HA through Phoscon and now I have deleted them from Phoscon but in HA I cannot remove them, does anyone know how I can remove them from HA?

Thank you.

marithpl, 2021-01-23:
How you pair ZBmini? Phoscone app doesn't found it as light or switch.

JCSCatena, 2021-01-24:
Hi,

I believe that I confirm that the problem is due to the fact that the ZBMINI device, once we pair it with phoscon / deconz / HA, does not refresh correctly in phoscon / HA.

If we follow the steps indicated below, we can link them without problems. I had 2 ZBMINIs linked correctly, but I bought another 5 and there was no way they would work correctly in Phoscon / HA, but following the steps below all 5 have been linked without problems:

1. Reset the ZBMINI by pressing the only button that has at least 10 seconds.
2. Restart HA.
3. Delete the ZBMINI devices that we do not have properly configured in HA. In my case, two of them had been able to link them for a long time, but another five that I had bought could NOT link them and I had them registered in Phoscon, HA and deconz, but they only worked well in deconz.
4. Call the service "deconz.remove_orphaned_entries" to clean up any devices that exist in HA but no longer exist in Phoscon because we have removed them from Phoscon in the previous step but they had remained in HA. What is done from the menu -> Development tools -> Services -> select "deconz.remove_orphaned_entries" in combo service and push the "call service" button
5. Add a ZBMINI device to phoscon, searching for the device in the section.
6. Optionally, but surely you will need it since the names that Phoscon offers when we install the device are not usually the definitive ones, update the name of the device in Phoscon and also optionally include the ZBMINI in the group that you see convenient also in Phoscon.
7. Restart Phoscon. What is done from the menu -> Supervisor -> Phoscon -> RESTART
8. We recharge the devices installed in Phoscon. Menu -> Configuration -> Phoscon Integration -> Click on the three dots -> Reload

I hope these instructions help you.

A greeting.

marithpl, 2021-01-24:
It doesn't work for me

> Delete the ZBMINI devices that we do not have properly configured in HA. In my case, two of them had been able to link them for a long time, but another five that I had bought could NOT link them and I had them registered in Phoscon, HA and deconz, but they only worked well in deconz.
Where should delete ZBMINI?

> Restart Phoscon. What is done from the menu -> Supervisor -> Phoscon -> RESTART
If I don't have supervisor should I restart Deconz container?

jmpaiva, 2021-01-30:
I followed all the steps and still it didn't work.

I even removed/re-added the phoscon integration instead of steps 3 to 5 and the result is always the same.

Is anyone working on this? Any change of being fixed?

borsa93, 2021-01-30:
Hi all, same issue here! Also for me no one of the solution proposed is working. Can someone help us? thanks.

Mimiix, 2021-01-30:
@borsa93 Manup already replied that until he got the device, not a lot can be done. It's a waiting game I guess. 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4016#issuecomment-751447670

spiff42, 2021-01-30:
A couple of days ago I managed to get one ZBMINI working correctly. I am not sure what I did, but it involved deleting the device, resetting it, restarting deCONZ. After the restart, it worked perfectly.

So today I spent several hours experimenting with a few other ZBMINIs, to try and get those working as well, but without success. I tried resetting the ZBMINI by holding the button for 10 sec, deleting it in phoscon, restarting deCONZ, only plugging in the ZBMINI while searching for new devices, reading the node and cluster info while in discovery mode, in a lot of different sequences, but nothing helped.

As previously stated, I do not use HA, but since phoscon uses the same API, I think we can leave HA out of the equation.

My findings are:
- When ZBMINI is connected, but not working, it is possible to use the on, off and toggle commands from the on/off cluster in deCONZ, and the device reacts correctly. However, the OnOff attribute is not updated after turning on/off the device. If a manual read of the attributes in the on/off cluster is performed, it shows correct state, and the correct state is also updated in phoscon.
- For the one that works, after executing the on, off or toggle command, the OnOff attribute is updated immediately, and this updates the state in Phoscon.
- The non-working devices can occasionally be operated once from Phoscon: If the device is off and shows off in phoscon, it can be turned on, but phoscon still shows it is off (and nothing happens for subsequent presses). Conversely, if the device is on, and manually reading the attributes in deCONZ, phoscon shows the state as on, and then pressing the bulb in phoscon turns off the device (but status is still not updated, so subsequent pushes do nothing). If once again a manual read of the attributes in the on/off cluster is performed, the state shows in phoscon and can be toggled once more.
- Every time I reset a ZBMINI and try to search for new lights in Phoscon, the ZBMINI shows up with a vendor of Heiman. If I go into deCONZ and perform a read of the basic cluster, it changes to SONOFF, and this also changes in Phoscon.
- I tried comparing the node and cluster info of the working and non-working devices, but found no differences. And let me add that the one I got working was not initially working (behaved the same as the others), but then suddenly did. I will NOT reset the one that works :-)

deebeo-blip, 2021-02-03:
Probably having the same problem: 
ZBMini can be paired and turned on once. 

What works:
Turn on first time after booting gateway. 
Group off/Group On: Actions are executed, Status doesn't change.
All off: Light is turned off, Status doesn't change.

What doesn't work:
Turn "Light/Device" off/on - status is not updated, actions are not executed.

Would be very nice if this could be fixed as these sonoff zbminis are quite affordable... and I now have 4 :)

manup, 2021-02-03:
Small update: We had ordered one which has now arrived, I'll test the device the coming days to see what's going on.

nukleuz80, 2021-02-03:
The same issue here.

jonnycastaway, 2021-02-07:
Same here,
i updated to latest beta today but issue stays.
Switching the device is possible with action command but status update is not right. Status switches to true an immediately back to false. So it's not possible to get actual state.

darkdevil373, 2021-02-07:
Hi all,
Have you restart your deconz Server? In my case phoscon Runs in a vm in Proxmox. Today i have restart the hole VM and now ist Works for me

jonnycastaway, 2021-02-08:
> Hi all,
> Have you restart your deconz Server? In my case phoscon Runs in a vm in Proxmox. Today i have restart the hole VM and now ist Works for me

In my situation, yes.
I'm not on a VM or Docker, i'm on bare metal raspberrypi 4. I updated deconz, started new, started teh complete pi, nothing works. The on status is always flipping to false.
BUT:
If i restart deconz while ZBmini is on the situations itself flips.
The on status flipps always back to true.

I'm searching fo a firmware update for the ZBmini but found nothing yet.

slash133, 2021-02-09:
Same issue here on raspberry pi 3 and conbee II

gang-himself, 2021-02-09:
Same issue also here on raspberry pi 3 and conbee II

gang-himself, 2021-02-09:
Just updated Deconz-Addon from 6.6.1 to 6.6.5 and now the already paired zbMini ist working perfectly!

marithpl, 2021-02-09:
> Just updated Deconz-Addon from 6.6.1 to 6.6.5 and now the already paired zbMini ist working perfectly!

I've installed HA 2021.2.1 and status seems to work.

slash133, 2021-02-09:
> Just updated Deconz-Addon from 6.6.1 to 6.6.5 and now the already paired zbMini ist working perfectly!

I have update too but no change for me
HA 2021.2.1
Deconz 6.6.5

maciej-szalaj, 2021-02-09:
Deconz updated to 6.6.5
HA updated to 2021.2.2

ZBmini seems to work as intended. Status in HA reflecting actual device status.

Mimiix, 2021-02-09:
@kane610 Did you change anything 😂 ? AFAIK, deCONZ didn't.

Kane610, 2021-02-09:
> @Kane610 Did you change anything 😂 ? AFAIK, deCONZ didn't.

Nope, I'm innocent of any improvement :)

maciej-szalaj, 2021-02-09:
> > @Kane610 Did you change anything 😂 ? AFAIK, deCONZ didn't.
> 
> Nope, I'm innocent of any improvement :)

It’s some kind of magic ;)

Kane610, 2021-02-09:
> > > @Kane610 Did you change anything 😂 ? AFAIK, deCONZ didn't.
> > 
> > 
> > Nope, I'm innocent of any improvement :)
> 
> It’s some kind of magic ;)

I don't believe in magic, must be the underpants gnomes 🤔

Mimiix, 2021-02-09:
So this issue can be closed? @minke890

ads1230, 2021-02-09:
The flickering of states when turning off the switch still occurs (off -> on -> off) for me . 

Deconz updated to 6.6.5
HA updated to 2021.2.2

Mimiix, 2021-02-09:
Then i suggest waiting for Manup still to make sure about it. 

**Please remain of adding "#Metoo". Otherwise i'll be forced to lock the issue.**

jmpaiva, 2021-02-09:
I have updated HA and deCONZ and now it's working for me! Thank you guys

minke890, 2021-02-09:
I also have updated HA and deCONZ but still have the flickering of states issue (off -> on -> off) for at least one ZBmini.

LeoeLeoeL, 2021-02-09:
But... the reason why ZBMINI goes alone to PAIRING MODE is a Deconz or a Sonoff fault?
If it's a Sonoff fault, I don't have too much time to give back this device because I don't think there will be the possibility to update the firmware without its bridge.

gang-himself, 2021-02-10:
Today I added two more zbMinis. Both worked the wrong way (whereas the yesterday mentioned old one still works correct).
I tried several things. Now all three devices work the correct way!
What helped (don't aks me why...):
1. Stopping the Deconz-Addon, while the devices are online.
2. Starting the Deconz-Addon again and before it has completed to load, reload the Deconz-Integration!
After that, every connected zbMini works. 
If there are not connected zbMinis (I had only one of the new ones on power), they don't switch to "normal" behaviour. So i did the procedure again with this one on power, and voilà...

I hope, this helps a bit

LeoeLeoeL, 2021-02-10:
I have a bare metal installation. Anyway, my device continues to go in Pairing mode".

Mimiix, 2021-02-10:
**@manup got the device, so let's see what happens on his side. I'll unlock whenever there is news.**

manup, 2021-02-22:
Fixed now in deCONZ core after some testing with the switch.

The problem is that the device uses wrong ZCL datatype 8-bit signed integer (0x28) for reporting instead boolean (0x10) datatype.
Surprisingly when reading the on/off attribute the correct boolean datatype is used.

The fix just updates internally used datatype to the one from the report, this was already done for ZCL read attributes responses but not for reports.

Mimiix, 2021-02-22:
**Thanks @manup!. For my administration: can you please link the issue to the pr? I'll unlock and re-open when it's merged and the beta is out.**

[38 more replies]

closed 2021-03-01 by SwoopX