dresden-elektronik/deconz-rest-plugin #4018: Deconz 2.8.0 Effekte Lidl Weihnachtsbaum Lichterkette benutzen

Sn4k3E4t3r, 2020-12-24:
## Describe the question or issue you are having

Hi zusammen,

ich habe jetzt schon länger Deconz im Einsatz. RaspBee2 wird verwendet auf eine Raspberry 3B+

Ich habe zu dem Thema schon so einiges hier gelesen. Aber ich verstehe noch nicht ob ich vielleicht noch etwas vergessen habe zu installieren. Also ich habe wie in eurer Anleitung den RaspBee2 https://phoscon.de/de/raspbee2/install durchgeführt.

Ich habe tatsächlich auf meinem Pi ein GUI nachinstalliert da ich mal die App sehen wollte. Die Frage ist hier wird immer von Plugins geredet. Ich sehe das es auch eine xmas.cpp ? LIDL manufacturer names gibt? Ich kann die Farben soweit einstellen. Noch nicht perfekt aber nunja aber wie komme ich denn mit der WEB App auf die Effekte?

## Screenshots

## Environment

 - Host system: (Raspberry Pi 3B+
 - Running method: (Raspbian / Home Assistent deCONZ Add-on
 - Firmware version: (26570700)
 - deCONZ version: (2.08.00)
 - Device: (RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II no
 - Is there any other USB or serial devices connected to the host system? If so: Which? no

## deCONZ Logs

## Additional context

popy2k14, 2020-12-24:
I have the same question.
I can set ex. Twinkle in fhem with 'set HUEDevice53 effect twinkle'

How can I set them ex. With HUE Essentials? My preferred hue App.

Thx a merry Christmas

Mimiix, 2020-12-24:
Hi, Can you translate to english?

popy2k14, 2020-12-24:
He wants to set an effect in phoscon web app.
I want to set an effect with hue essentials. 

Can you please save Christmas?

🙂

Mimiix, 2020-12-24:
Either question can't be answered by me. 

Phoscon questions need to be answered in their Repo.

Hue essentials by @Thomas-Vos

Thomas-Vos, 2020-12-24:
> He wants to set an effect in phoscon web app.

Phoscon would need to add support for the LIDL effects. I think it is best to open an issue in their repository.

> I want to set an effect with hue essentials.

In Hue Essentials, first open a room/group/zone, and then go to the third tab (lightning icon) at the top. There you will see a list of your current effects and the effect packages. The Christmas effects for the LIDL String Lights are at the bottom under "Zigbee". You can customise the speed and colours for every LIDL Zigbee effect.

You can also start the effects using the REST API, see: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3716#issuecomment-735467996

popy2k14, 2020-12-24:
Thomas, you saved Christmas 😉😉😉😉😉👌🏻👌🏻👌🏻
Didn't scrolled down to the end. Was thinking that's just effects for multiple light bulbs. 

Another question:
Can I'll add such an effect in an scene? 

Thx again and merry Christmas

Thomas-Vos, 2020-12-24:
At the moment you cannot add these effects in scenes in Hue Essentials. In a future update I will add support for starting these effects from scenes, schedules, and smart controls. That will be some time after Christmas.

Please share your Christmas lights in the #🎄xmas-showoff channel in Discord.

popy2k14, 2020-12-24:
No problem, take your time and keep up the good work. 

Keep save and merry Christmas.

closed 2021-02-08 by Mimiix