dresden-elektronik/deconz-rest-plugin #4020: Dimming a light even more

moonloopsun, 2020-12-25:
## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

moonloopsun, 2020-12-25:
Hi guys. I am new here and have a little question. Is it possible to dim a hue light even lower than 1% somehow?
I have a hue filament light and it’s strong even when its dimmed to 1% so i try to understand how i can go lower. 
thanks a lot!!

Mimiix, 2020-12-25:
This is not how the template work.

Closing.

closed 2020-12-25 by Mimiix

moonloopsun, 2020-12-25:
Thanks for the super unking and unsupportive answer!

moonloopsun, 2020-12-25:
*unkind