dresden-elektronik/deconz-rest-plugin #4021: Cannot set light state for light when `state.reachable` is false

ebaauw, 2020-12-25:
## Describe the bug

The API returns api error 3, "resource, /lights/xx, not available" when trying to set a light state of a light that's unreachable (for which `state.reachable` is false).

## Steps to reproduce the behavior

```
$ ph get /lights/12
{
  "colorcapabilities": 0,
  "ctmax": 65279,
  "ctmin": 0,
  "etag": "bb0b792390b8e37858d2304b7cb2859e",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2020-12-21T15:55Z",
  "manufacturername": "LIDL Livarno Lux",
  "modelid": "HG06106C",
  "name": "Extended color light 12",
  "state": {
    "alert": "none",
    "bri": 255,
    "colormode": "ct",
    "ct": 153,
    "effect": null,
    "hue": 0,
    "on": false,
    "reachable": false,
    "sat": 254,
    "xy": [
      0.7,
      0.3
    ]
  },
  "swversion": "0x00000050",
  "type": "Extended color light",
  "uniqueid": "84:2e:14:ff:fe:22:29:1a-01"
}
$ ph put -v /lights/12/state '{"on": true}'
ph put: warning: api error 3: resource, /lights/12, not available
[
  {
    "error": {
      "address": "/lights/12",
      "description": "resource, /lights/12, not available",
      "type": 3
    }
  }
]
```

## Expected behavior

This behaviour doesn't seem correct to me:
- The resource is available, so the error message is highly confusing.  At least it should read "not reachable".
- I think this should be a warning, not an error; i.e. the API plugin should still send the Zigbee commands to change the state, even though the light is marked unreachable.
## Screenshots

## Environment

 - Host system: All
 - Running method: All
 - Firmware version: All
 - deCONZ version: 2.9.0, but probably all.
 - Device: all

## deCONZ Logs

n/a

## Additional context

n/a

ebaauw, 2020-12-25:
I think is't best to remove the check for reachability altogether.  The Hue bridge (`swversion` 1941132080, API version 1.41.0) accepts changing the state without any warning when the light is unreachable.

@manup, do you agree?

```
$ ph -H hue-3 get /lights/27
{
  "state": {
    "on": false,
    "bri": 77,
    "hue": 6144,
    "sat": 254,
    "effect": "none",
    "xy": [
      0.561,
      0.4042
    ],
    "ct": 499,
    "alert": "select",
    "colormode": "ct",
    "mode": "homeautomation",
    "reachable": false
  },
  "swupdate": {
    "state": "notupdatable",
    "lastinstall": "2020-11-18T12:28:42"
  },
  "type": "Extended color light",
  "name": "GLEDOPTO",
  "modelid": "GL-C-008",
  "manufacturername": "GLEDOPTO",
  "productname": "Extended color light",
  "capabilities": {
    "certified": false,
    "control": {
      "colorgamuttype": "other",
      "ct": {
        "min": 0,
        "max": 65535
      }
    },
    "streaming": {
      "renderer": false,
      "proxy": false
    }
  },
  "config": {
    "archetype": "classicbulb",
    "function": "mixed",
    "direction": "omnidirectional"
  },
  "uniqueid": "00:12:4b:00:1b:91:ec:83-0b",
  "swversion": "1.0.6"
}
$ ph -H hue-3 put -v /lights/27/state '{"on": true}'
[
  {
    "success": {
      "/lights/27/state/on": true
    }
  }
]
```

manup, 2021-01-12:
Hmm this is very tricky, sending out many commands to potentially unreachable nodes can easily block the queue and lead to a "nothing works" experience.

Note sending a command to an unreachable device might occupy that APS Request slot for up to 30 seconds with all the APS ACK, re transmit and routing happening in lower layers.

The check basically prevents this.

So the question on my side is, is `state.reachable` not properly set, at least a while after deCONZ is started?

ebaauw, 2021-01-12:
Is there no timeout/max retry count?

At least, can we change the error message (and probably code as well)?

manup, 2021-01-12:
Yes that's what the reachable is based on, if the core can't reach a device with N errors or hasn't received anything back for time X it's marked as zombie which leads to `reachable` turning false.

The firmware has no knowledge of this, as it would just try to reach out to the device and if no working route is known a new route discovery takes place. This can take some seconds and multiple requests can run in parallel (doesn't apply for source routing). Therefor for normal operation the plugin should really only send when reachable is true. 

The core tries to reach out to all routers periodically and only one at the time, to detect if the device is available. Incoming ZCL reports and Device Announce messages further trigger the device to be reachable rather quickly.

>At least, can we change the error message (and probably code as well)?

I always thought reachable and available as the same :) but I'm fine replacing the wording to be more clear what's happening.

ebaauw, 2021-01-12:
> I always thought reachable and available as the same

The first condition is temporary, the second is permanent:
```
$ ph get /lights/inValidLight
ph get: request 1: GET /lights/inValidLight
ph get: error: request 1: api error 3: resource, /lights/inValidLight, not available
```

manup, 2021-01-13:
Ah I see now what you mean, not available in terms of 404 the resource doesn't exist I guess.
Agree this is misleading, not reachable is better for `state.reachable`.

ebaauw, 2021-01-13:
I think we need to introduce `ERR_DEVICE_NOT_REACHABLE` For this.  Any preference for the associated API error code?

manup, 2021-01-13:
>Any preference for the associated API error code?

Only that it needs to be unique :)

ebaauw, 2021-01-13:
```
$ ph put /lights/12/state '{"on": true}'
ph put: request 1: PUT /lights/12/state {"on":true}
ph put: error: request 1: api error 202: resource, /lights/12/state, is not modifiable. Device is not reachable.
```

There doesn't seem to be a similar check when changing the `config` of a non-CLIP `/sensors` resource.

closed 2021-01-17 by manup