dresden-elektronik/deconz-rest-plugin #4025: Phoscon Error Syncing with Jeedom in RPI4

srtxema, 2020-12-26:
## Describe the question or issue you are having
Hi, I have a problem with Conbee2 with deconz in RPI4 for jeedom.
deconz is installed ok, but when I click on the phoscon web button, sometimes it will open the ip localhost: 8080 and other times phoscon.de/app, with different devices on each one. Why is this happening? which one is correct to work with? with phoscon.de or with localhost? thanks!
both images are of the same installation and i can´t solve problem of sync with Jeedom

## Screenshots
[image]

[image]

## Environment
RPI 4
Deconz 2.07.01
firmware 0x26580700.bin.GCF
Raspbian Buster
no extension cable
usb for rfid keyboard

## deCONZ Logs
04:03:16:801 APS-DATA.request id: 13, addrmode: 0x03, addr: 0x00212effff066630, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
04:03:16:820 APS-DATA.confirm id: 13, status: 0x00 SUCCESS
04:03:16:820 APS-DATA.confirm request id: 13 -> confirmed, timeout 15735832
04:03:16:838 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
04:03:16:838 	asdu: cc00000000
04:03:16:839 APS-DATA.indication request id: 13 -> finished
04:03:16:839 APS-DATA.request id: 13 erase from queue
04:03:20:647 APS-DATA.request id: 31, addrmode: 0x03, addr: 0x00212effff066630, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
04:03:20:669 APS-DATA.confirm id: 31, status: 0x00 SUCCESS
04:03:20:670 APS-DATA.confirm request id: 31 -> confirmed, timeout -1308599456
04:03:20:680 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
04:03:20:680 	asdu: cd00000000
04:03:20:681 APS-DATA.indication request id: 31 -> finished
04:03:20:681 APS-DATA.request id: 31 erase from queue
04:03:24:481 APS-DATA.request id: 48, addrmode: 0x03, addr: 0x00212effff066630, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
04:03:24:494 APS-DATA.confirm id: 48, status: 0x00 SUCCESS
04:03:24:496 APS-DATA.confirm request id: 48 -> confirmed, timeout 15730296
04:03:24:508 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
04:03:24:510 	asdu: ce00000000
04:03:24:511 APS-DATA.indication request id: 48 -> finished
04:03:24:511 APS-DATA.request id: 48 erase from queue
04:03:28:321 APS-DATA.request id: 65, addrmode: 0x03, addr: 0x00212effff066630, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
04:03:28:336 APS-DATA.confirm id: 65, status: 0x00 SUCCESS
04:03:28:337 APS-DATA.confirm request id: 65 -> confirmed, timeout -1308578760
04:03:28:340 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
04:03:28:341 	asdu: cf00000000
04:03:28:342 APS-DATA.indication request id: 65 -> finished
04:03:28:343 APS-DATA.request id: 65 erase from queue
04:03:31:769 APS-DATA.request id: 81, addrmode: 0x03, addr: 0x00212effff066630, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
04:03:31:782 APS-DATA.confirm id: 81, status: 0x00 SUCCESS
04:03:31:783 APS-DATA.confirm request id: 81 -> confirmed, timeout -1308571424
04:03:31:786 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
04:03:31:787 	asdu: d000000000
04:03:31:787 APS-DATA.indication request id: 81 -> finished
04:03:31:787 APS-DATA.request id: 81 erase from queue
04:03:35:609 APS-DATA.request id: 99, addrmode: 0x03, addr: 0x00212effff066630, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
04:03:35:614 APS-DATA.confirm id: 99, status: 0x00 SUCCESS
04:03:35:614 APS-DATA.confirm request id: 99 -> confirmed, timeout 15247288
04:03:35:616 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
04:03:35:617 	asdu: d100000000
04:03:35:618 APS-DATA.indication request id: 99 -> finished
04:03:35:618 APS-DATA.request id: 99 erase from queue
04:03:39:449 APS-DATA.request id: 116, addrmode: 0x03, addr: 0x00212effff066630, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
04:03:39:460 APS-DATA.confirm id: 116, status: 0x00 SUCCESS
04:03:39:460 APS-DATA.confirm request id: 116 -> confirmed, timeout -1308571776
04:03:39:465 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
04:03:39:466 	asdu: d200000000
04:03:39:466 APS-DATA.indication request id: 116 -> finished
04:03:39:467 APS-DATA.request id: 116 erase from queue
04:03:43:065 APS-DATA.request id: 132, addrmode: 0x03, addr: 0x00212effff066630, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
04:03:43:149 APS-DATA.confirm id: 132, status: 0x00 SUCCESS
04:03:43:150 APS-DATA.confirm request id: 132 -> confirmed, timeout 15463360
04:03:43:192 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
04:03:43:193 	asdu: d300000000
04:03:43:194 APS-DATA.indication request id: 132 -> finished
04:03:43:195 APS-DATA.request id: 132 erase from queue
04:03:46:905 APS-DATA.request id: 150, addrmode: 0x03, addr: 0x00212effff066630, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
04:03:46:919 APS-DATA.confirm id: 150, status: 0x00 SUCCESS
04:03:46:920 APS-DATA.confirm request id: 150 -> confirmed, timeout -1415489248
04:03:46:925 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
04:03:46:927 	asdu: d400000000
04:03:46:928 APS-DATA.indication request id: 150 -> finished
04:03:46:929 APS-DATA.request id: 150 erase from queue

## Additional context
its not posible link with jeedom

Smanar, 2020-12-26:
You need to use the ip ans port from the second capture for jeedom.
Just by curiosity what do you have at this url ? https://phoscon.de/discover

But I don't understand why you haven't the same devices, you are sure not haven't 2 deconz instances on 2 differents machine ?

srtxema, 2020-12-26:
I don’t understand neither! I only have 1 conbee2 usb in usb 2.0 with extension cable. Only this!
This url return it 
[{"id":"00212EFFFF066630","internalipaddress":"192.168.1.20","macaddress":"00212EFFFF066630","internalport":8484,"name":"Phoscon-GW","publicipaddress":"83.35.223.___"}]

SwoopX, 2020-12-26:
Which screenshot depicts your devices now?

srtxema, 2020-12-26:
so now first image is phoscon.de/app and second is localhost:8484. In localhost i write correct password 2 times but always show "Login failed". i dont know now what happened.
I show another imagen. when i start deconz i is disconnected and show Conbee2 and Raspbee, but i havent raspbee

[image]

[image]

[image]

Smanar, 2020-12-26:
The second screenshoot is normal, deconz always ask for both.

srtxema, 2020-12-26:
ah, ok! i didnt know. but i have the problem that deconz normally is not connected and with phoscon that i explained before

Smanar, 2020-12-27:
But do you still have the password problem ?
Can you show the configuration setting from jeedom ? (on the first screenshoot you was using 8080 instead of 8084)

srtxema, 2020-12-27:
I have reinstalled and i have follow this steps. https://jeedomiser.fr/article/vos-equipements-zigbee-avec-la-cle-conbee-ii-et-jeedom
i havent installed Deconz, only jeedom and conbee by jeedon and its ok!

SwoopX, 2020-12-27:
Besides not having answered my question, have you checked if jeedom already installs any version of deconz? Your symptoms of being "not in network" indicate 2 instances of deconz are started simultaneously...

Smanar, 2020-12-27:
@SwoopX 
Jeedom assist a lot users (good or bad idea ....) and if they choose the "classic" installation, it s jeedom itself that manage deconz, they just clic on a button and have no clue what happen after.

A frequent issue is users that install themself deconz on their machine without disabling this feature.

srtxema, 2020-12-27:
> Besides not having answered my question, have you checked if jeedom already installs any version of deconz? Your symptoms of being "not in network" indicate 2 instances of deconz are started simultaneously...

I supuse that probably I installed by deconz installation and Jeedom installation. I tried a lot of and probably it occur

srtxema, 2020-12-27:
> Jeedom assist a lot users (good or bad idea ....) and if they choose the "classic" installation, it s jeedom itself that manage deconz, they just clic on a button and have no clue what happen after. And from my memory it use command line to run deconz and not the service.
> 
> A frequent issue is users that install themself deconz on their machine (for exemple to make a second instal with the GUI, even it 's the same) without disabling this feature.

Yes. I supuse it occurred. For the moment it’s ok and I think I won’t need more features. If in the future I needed deconz and phoscon, can I disable Jeedom installation and installed by deconz service?

Smanar, 2020-12-27:
Yes on jeedom it s called "’installation en deporté", with this mode it s you that manage deconz, the emplacement, the machine where to install it, headless or desktop, the update, ect ...

srtxema, 2020-12-27:
> Yes on jeedom documentation, it s called "’installation en deporté", with this mode it s you that manage deconz, the emplacement, the machine where to install it, headless or desktop, the update, ect ...
> 
> > Le plugin support plusieurs gateway deconz (1 clef conbee par gateway). Une gateway conbee peut etre installer sur :
> > ```
> > jeedom lui meme (smart, rpi et x64_86 supporté)
> > un rpi deporté
> > ou tout autre carte compatible avec deconz.
> > ```
> > 
> > 
> > Pour l’installation en deporté voici la documentation
> > Pour l’installation sur jeedom vous avez juste à cliquer sur le bouton pour installer deconz en local.
> > ```
> > IMPORTANT
> > 
> > Il ne faut SURTOUT PAS faire les 2 methodes d’installation sur le meme système c’est l’un ou l’autre
> > ```

Thanks a lot!! Very good solution

closed 2020-12-27 by srtxema

Smanar, 2020-12-27:
Oups, sorry was the terms in french, here for spanish https://doc.jeedom.com/es_ES/plugins/automation%20protocol/deconz/

srtxema, 2020-12-27:
No problem :) I have seen in English