dresden-elektronik/deconz-rest-plugin #4029: Conbee II constantly disconnects from USB or can't initialize

coaxial, 2020-12-26:
## Describe the question or issue you are having

The ConBee II constantly disconnects or fails to initialize when plugged in. As a result, operation is very unreliable (either using ZHA, Deconz, or Zigbee2mqtt)

## Screenshots

```
[220714.617887] usb 3-10: new full-speed USB device number 19 using xhci_hcd
[220714.767510] usb 3-10: New USB device found, idVendor=1cf1, idProduct=0030
[220714.767515] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[220714.767518] usb 3-10: Product: ConBee II
[220714.767521] usb 3-10: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[220714.767523] usb 3-10: SerialNumber: DE2258405
[220714.768618] cdc_acm 3-10:1.0: ttyACM0: USB ACM device
[220903.879737] usb 3-10: USB disconnect, device number 19
[220904.185283] usb 3-10: new full-speed USB device number 20 using xhci_hcd
[220904.334721] usb 3-10: New USB device found, idVendor=1cf1, idProduct=0030
[220904.334726] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[220904.334730] usb 3-10: Product: ConBee II
[220904.334732] usb 3-10: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[220904.334735] usb 3-10: SerialNumber: DE2258405
[220904.335788] cdc_acm 3-10:1.0: ttyACM0: USB ACM device
[220908.104942] usb 3-10: USB disconnect, device number 20
[220908.105076] cdc_acm 3-10:1.0: failed to set dtr/rts
[220908.453173] usb 3-10: new full-speed USB device number 21 using xhci_hcd
[220913.649074] usb 3-10: unable to read config index 0 descriptor/all
[220913.651756] usb 3-10: can't read configurations, error -110
[220913.781037] usb 3-10: new full-speed USB device number 22 using xhci_hcd
[220919.024956] usb 3-10: unable to read config index 0 descriptor/all
[220919.027642] usb 3-10: can't read configurations, error -110
[220919.030371] usb usb3-port10: attempt power cycle
[220919.680894] usb 3-10: new full-speed USB device number 23 using xhci_hcd
[220929.520837] usb 3-10: unable to read config index 0 descriptor/all
[220929.523496] usb 3-10: can't read configurations, error -110
[220929.860644] usb 3-10: new full-speed USB device number 24 using xhci_hcd
[220934.896643] usb 3-10: unable to read config index 0 descriptor/all
[220934.899306] usb 3-10: can't read configurations, error -110
[220934.902051] usb usb3-port10: unable to enumerate USB device
```

## Environment

 - Host system: PC (HASS OS, Linux)
 - Running method: HASS OS
 - Firmware version: deCONZ_ConBeeII_0x26680700.bin.GCF

 - deCONZ version: using zigbee2mqtt right now but it doesn't matter because the issue is with the USB layer, Linux can't initialize the device at all.
 - Device: ConBee II
 - Do you use an USB extension cable: tried with and without, no difference
 - Is there any other USB or serial devices connected to the host system? no. Tried every USB3 port I have (6 on that machine) and USB2 (2 ports), same result.

## deCONZ Logs

```
[220714.617887] usb 3-10: new full-speed USB device number 19 using xhci_hcd
[220714.767510] usb 3-10: New USB device found, idVendor=1cf1, idProduct=0030
[220714.767515] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[220714.767518] usb 3-10: Product: ConBee II
[220714.767521] usb 3-10: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[220714.767523] usb 3-10: SerialNumber: DE2258405
[220714.768618] cdc_acm 3-10:1.0: ttyACM0: USB ACM device
[220903.879737] usb 3-10: USB disconnect, device number 19
[220904.185283] usb 3-10: new full-speed USB device number 20 using xhci_hcd
[220904.334721] usb 3-10: New USB device found, idVendor=1cf1, idProduct=0030
[220904.334726] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[220904.334730] usb 3-10: Product: ConBee II
[220904.334732] usb 3-10: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[220904.334735] usb 3-10: SerialNumber: DE2258405
[220904.335788] cdc_acm 3-10:1.0: ttyACM0: USB ACM device
[220908.104942] usb 3-10: USB disconnect, device number 20
[220908.105076] cdc_acm 3-10:1.0: failed to set dtr/rts
[220908.453173] usb 3-10: new full-speed USB device number 21 using xhci_hcd
[220913.649074] usb 3-10: unable to read config index 0 descriptor/all
[220913.651756] usb 3-10: can't read configurations, error -110
[220913.781037] usb 3-10: new full-speed USB device number 22 using xhci_hcd
[220919.024956] usb 3-10: unable to read config index 0 descriptor/all
[220919.027642] usb 3-10: can't read configurations, error -110
[220919.030371] usb usb3-port10: attempt power cycle
[220919.680894] usb 3-10: new full-speed USB device number 23 using xhci_hcd
[220929.520837] usb 3-10: unable to read config index 0 descriptor/all
[220929.523496] usb 3-10: can't read configurations, error -110
[220929.860644] usb 3-10: new full-speed USB device number 24 using xhci_hcd
[220934.896643] usb 3-10: unable to read config index 0 descriptor/all
[220934.899306] usb 3-10: can't read configurations, error -110
[220934.902051] usb usb3-port10: unable to enumerate USB device
```

## Additional context

Smanar, 2020-12-26:
Hello, have you tried that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3262 ?

coaxial, 2020-12-26:
Hello, where do I get the `GCFFlasher` binary from?

Smanar, 2020-12-26:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-ubuntu not working ?

coaxial, 2020-12-26:
I am not running the update on the machine that has deconz installed since I'm running zigbee2mqtt on HASS OS. Is there no compiled binary I could download and run?

coaxial, 2020-12-27:
Well I flashed it once more using the docker container (since the `GCFFLash_internal` binary can't be had anywhere) and manually editing the firmware-flash.sh script to add the `-t 60` option. It flashed it fine, just like the 10 previous attempts (without the -t option), and it still can't be recognized properly by the host computer most of the time:

```
[  +0.149324] usb 1-2: New USB device found, idVendor=1cf1, idProduct=0030
[  +0.000004] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 1-2: Product: ConBee II
[  +0.000002] usb 1-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  +0.000002] usb 1-2: SerialNumber: DE2258405
[  +0.000629] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[  +2.691628] usb 1-2: USB disconnect, device number 27
[  +0.000134] cdc_acm 1-2:1.0: failed to set dtr/rts
[  +0.390199] usb 1-2: new full-speed USB device number 28 using xhci_hcd
[  +5.387851] usb 1-2: unable to read config index 0 descriptor/all
[  +0.002078] usb 1-2: can't read configurations, error -110
[  +0.133867] usb 1-2: new full-speed USB device number 29 using xhci_hcd
[  +5.239903] usb 1-2: unable to read config index 0 descriptor/all
[  +0.000933] usb 1-2: can't read configurations, error -110
[  +0.000955] usb usb1-port2: attempt power cycle
[  +0.650068] usb 1-2: new full-speed USB device number 30 using xhci_hcd
[  +9.587807] usb 1-2: unable to read config index 0 descriptor/all
[  +0.000472] usb 1-2: can't read configurations, error -110
[  +0.339521] usb 1-2: new full-speed USB device number 31 using xhci_hcd
[Dec27 09:39] usb 1-2: unable to read config index 0 descriptor/all
[  +0.002072] usb 1-2: can't read configurations, error -110
[  +0.002223] usb usb1-port2: unable to enumerate USB device
[ +10.087373] usb 1-2: new full-speed USB device number 32 using xhci_hcd
[  +0.149329] usb 1-2: New USB device found, idVendor=1cf1, idProduct=0030
[  +0.000001] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000001] usb 1-2: Product: ConBee II
[  +0.000001] usb 1-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  +0.000000] usb 1-2: SerialNumber: DE2258405
[  +0.001211] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[  +3.766662] usb 1-2: USB disconnect, device number 32
[  +0.000198] cdc_acm 1-2:1.0: failed to set dtr/rts
[  +0.382501] usb 1-2: new full-speed USB device number 33 using xhci_hcd
[  +5.319922] usb 1-2: unable to read config index 0 descriptor/all
[  +0.001654] usb 1-2: can't read configurations, error -110
[  +0.126301] usb 1-2: new full-speed USB device number 34 using xhci_hcd
[  +5.247883] usb 1-2: unable to read config index 0 descriptor/all
[  +0.002104] usb 1-2: can't read configurations, error -110
[  +0.002188] usb usb1-port2: attempt power cycle
[  +0.651702] usb 1-2: new full-speed USB device number 35 using xhci_hcd
[  +4.597390] usb 1-2: unable to read config index 0 descriptor/all
[  +0.000501] usb 1-2: can't read configurations, error -71
[  +0.333978] usb 1-2: new full-speed USB device number 36 using xhci_hcd
[  +0.021629] usb 1-2: New USB device found, idVendor=1cf1, idProduct=0030
[  +0.000005] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000003] usb 1-2: Product: ConBee II
[  +0.000002] usb 1-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  +0.000003] usb 1-2: SerialNumber: DE2258405
[  +0.001833] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[  +3.893163] usb 1-2: USB disconnect, device number 36
[  +0.000140] cdc_acm 1-2:1.0: failed to set dtr/rts
[  +0.367109] usb 1-2: new full-speed USB device number 37 using xhci_hcd
[  +5.231927] usb 1-2: unable to read config index 0 descriptor/all
[  +0.002253] usb 1-2: can't read configurations, error -110
[  +0.129722] usb 1-2: new full-speed USB device number 38 using xhci_hcd
[  +5.244073] usb 1-2: unable to read config index 0 descriptor/all
[  +0.000552] usb 1-2: can't read configurations, error -110
[  +0.000565] usb usb1-port2: attempt power cycle
[  +0.658623] usb 1-2: new full-speed USB device number 39 using xhci_hcd
[  +0.025954] usb 1-2: New USB device found, idVendor=1cf1, idProduct=0030
[  +0.000005] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000003] usb 1-2: Product: ConBee II
[  +0.000003] usb 1-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  +0.000003] usb 1-2: SerialNumber: DE2258405
[  +0.000889] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
```

Smanar, 2020-12-27:
It can't work on docker, it can broke the firmware more it can update it.
Here you have some version for GCFFlasher  for windows http://deconz.dresden-elektronik.de/win/

coaxial, 2020-12-28:
> It can't work on docker, it can broke the firmware more it can update it.

I don't get that. The tool flashes the firmware and then verifies it. How does running in docker or in a VM or on a "real" system change that?

If the flashing tool really is so finicky, it would be immensely helpful to have the `GCFFlash_internal` binary available for Linux too. I don't have a native Windows install, I'd have to run it in a VM which is apparently also the wrong way of doing it.

Smanar, 2020-12-28:
>How does running in docker or in a VM or on a "real" system change that?

Simple, it s virtual, you can transform something material just using software, for exemple an USB 2.0 in USB 3.0 just with a checkbox, it s like compare a real playstation and it s emulator.

>If the flashing tool really is so finicky, it would be immensely helpful to have the GCFFlash_internal binary available for Linux too. I don't have a native Windows install, I'd have to run it in a VM which is apparently also the wrong way of doing it.

Right, but was not able to find it, sorry, you can extract it from package (the package is easy to find) but idk for depencies.
http://deconz.dresden-elektronik.de/ubuntu/

coaxial, 2021-01-02:
> Simple, it s virtual, you can transform something material just using software, for exemple an USB 2.0 in USB 3.0 just with a checkbox, it s like compare a real playstation and it s emulator.

Yes and no. Flashing the firmware and verifying it's correctly written to the device should work reliably, no matter what the flasher is running on. Either it was flashed properly and it matches the file bit for bit or it doesn't. Virtual, docker, or metal has nothing to do with that.

Smanar, 2021-01-02:
It 's not so easy ^^. There is lot of layer, hardware and software, it s not so direct. And you have an exemple, you can send exaclty the same bit, but just a setting modification in a layer, you have as result the one for usb 2.0 or the one for USB 3.0.

coaxial, 2021-01-04:
Right. That still leaves any non-Windows user in a bind as one of the official methods for flashing the thing is to use a Docker container as the GCFFLasher_internal binary is not available on its own.

coaxial, 2021-01-06:
Well I'm giving up on mine. I don't believe I will ever get an answer from Dresden Electronics and I can't make the device work properly. What a waste of money that was.

Mimiix, 2021-01-06:
@coaxial feel free to email them.

Smanar, 2021-01-06:
You don't have real machine at all ? no Raspbian or Ubuntu ?

SwoopX, 2021-01-06:
@coaxial 
> ... as one of the official methods for flashing the thing is to use a Docker container ...

This is not an official running method. It's clearly stated that this is based on community support/efforts. There still might be some challenges to overcome...

coaxial, 2021-01-13:
> You don't have real machine at all ? no Raspbian or Ubuntu ?

I do but I don't want to have to install Deconz every time I need to flash.

This is way too much hassle for a *commercial* product (which cost twice as much as the Sonoff Zigbee ethernet gateway by the way). I'm giving up on the Conbee II, I've wasted too much time on it and still waiting for any answer from Dresden's tech support that I contacted by email about this almost a month ago now. I'm returning my Conbee II, it's been a waste of time and money.

Thanks to those who answered this thread, unfortunately it didn't fix the issue.

pawelka, 2021-01-25:
Reconnecting firmwares:
* deCONZ_ConBeeII_0x26680700.bin.GCF
* deCONZ_ConBeeII_0x26670700.bin.GCF

Looks stable (no reconnects): 
* deCONZ_ConBeeII_0x26580700.bin.GCF

I didn't tested firmwares between, maybe next after 0x265... works too.

Saviq, 2021-02-24:
This is randomly happening for me, too, on a NUC. It will work for a week at a time and suddenly start the reboot loop for no apparent reason. Connecting through a USB 2.0 hub seems to stabilize it.

Smanar, 2021-02-24:
Power usage ?

Saviq, 2021-02-24:
@Smanar no, the hub is powered from the same port.

Smanar, 2021-02-24:
Lol, yep, so not the power, the port is usb 3.0 ?

Saviq, 2021-02-24:
Yes, it's a 3.0 port, the hub is 2.0.

Fmstrat, 2021-03-15:
I am also having this issue in Ubuntu 20.04. As a note, the device worked perfectly in Ubuntu 18.04. Basically, this message repeated:
```
[35846.415591] usb 1-9: USB disconnect, device number 22
[35846.731177] usb 1-9: new full-speed USB device number 23 using xhci_hcd
[35847.070531] usb 1-9: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[35847.070533] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[35847.070535] usb 1-9: Product: ConBee II
[35847.070536] usb 1-9: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[35847.070537] usb 1-9: SerialNumber: DE2216078
[35847.085296] cdc_acm 1-9:1.0: ttyACM0: USB ACM device
[35850.297402] usb 1-9: USB disconnect, device number 23
[35850.647178] usb 1-9: new full-speed USB device number 24 using xhci_hcd
[35850.982553] usb 1-9: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[35850.982556] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[35850.982557] usb 1-9: Product: ConBee II
[35850.982558] usb 1-9: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[35850.982559] usb 1-9: SerialNumber: DE2216078
[35850.997327] cdc_acm 1-9:1.0: ttyACM0: USB ACM device
```

I have tried:

- Updating the firmware with `-t 60` (completed in Windows, worked great and validated)
- `sudo systemctl stop ModemManager`
- Plugged directly into a USB2 port
- Plugged directly into a USB3 port
- On a powered USB2 HUB on a USB2 port
- On a powered USB2 HUB on a USB3 port
- On a laptop running Ubuntu 20.04 doing the same things
- On another desktop running Ubuntu 20.04 doing the same things

This is completely independent of Home Assistant as this happens on a fresh boot with no non-system service running. I'm not really sure where to go from here. The fact that I can repeat this on 3 different installs makes me think this is a kernel/driver/firmware incompatibility with newer linux versions.

deviantintegral, 2021-03-15:
I am also seeing this on Ubuntu 20.04. I've even seen it in the initramfs shell after a clean boot, so that rules out anything like Hass or even deCONZ itself.

I was able to replicate on both USB 3.0 ports (`Intel Corporation 7 Series/C210 Series Chipset Family USB xHCI Host Controller (rev 04) (prog-if 30 [XHCI])` according to lspci) and the USB 2.0 ports (`00:1a.0 USB controller: Intel Corporation 7 Series/C216 Chipset Family USB Enhanced Host Controller #2 (rev 04) (prog-if 20 [EHCI])`). However, this is an old motherboard, so I picked up a cheap USB 2.0 PCI card with a VIA 6112 chipset to absolutely rule out the motherboard. It's been fine for 2 days now, but I need to wait longer as I could sometimes go a week or more without the disconnects.

deviantintegral, 2021-03-16:
I just had to reboot for a kernel update and the ConBee disconnects started again, so I've ruled out an issue with the kernel USB driver or my hardware.

AlexMarent, 2021-03-20:
Same issue here, running on rpi. Really annoying.

Fmstrat, 2021-03-22:
@deviantintegral I thought I was out of it too, but no, same kernel and problem came back. So no idea.

Mimiix, 2021-03-22:
@AlexMarent On RPI it could be different. Did you update firmware lately?

deviantintegral, 2021-03-22:
For one last try, I migrated deconz out of the KVM machine and into a docker container directly on the host. It lasted for a few days without issue, but just popped up again.

It seems like a good next step is to contact Dresden support for any possible debugging tips?

AlexMarent, 2021-03-23:
> @AlexMarent On RPI it could be different. Did you update firmware lately?

Firmware is updated to the newest version (flashed with a windows pc). The reconnecting looks very similar. 

```
[Fri Mar 19 12:10:13 2021] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[Fri Mar 19 12:32:01 2021] usb 1-1.2: USB disconnect, device number 31
[Fri Mar 19 12:32:02 2021] usb 1-1.2: new full-speed USB device number 32 using xhci_hcd
[Fri Mar 19 12:32:02 2021] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Fri Mar 19 12:32:02 2021] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Mar 19 12:32:02 2021] usb 1-1.2: Product: ConBee II
[Fri Mar 19 12:32:02 2021] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Fri Mar 19 12:32:02 2021] usb 1-1.2: SerialNumber: DE2140693
[Fri Mar 19 12:32:02 2021] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[Fri Mar 19 12:32:05 2021] usb 1-1.2: USB disconnect, device number 32
[Fri Mar 19 12:32:05 2021] usb 1-1.2: new full-speed USB device number 33 using xhci_hcd
[Fri Mar 19 12:32:06 2021] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Fri Mar 19 12:32:06 2021] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Mar 19 12:32:06 2021] usb 1-1.2: Product: ConBee II
[Fri Mar 19 12:32:06 2021] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Fri Mar 19 12:32:06 2021] usb 1-1.2: SerialNumber: DE2140693
```

Mimiix, 2021-03-23:
@AlexMarent Can you open a seperate issue and add logs?

cubertt, 2021-04-09:
> Firmware is updated to the newest version (flashed with a windows pc). The reconnecting looks very similar.
>
> ```
> [Fri Mar 19 12:10:13 2021] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
> [Fri Mar 19 [...]

Have the same logs (and problem), firmware updated via win 10 usb 3, running on hass.io 64 bit/rpi4-4gb (and also one time with rasbian)...
Try to get logs on a fresh system

crancker, 2021-05-06:
This is not just annoying, but rather irritating. I went through all the post and git issues related to this I could find and kinda the only solution is to upgrade or downgrade the FW or play with the usb ports mixed with or withous usb hubs. For like 2 month my conbee 2 worked like a charm but in the last couple of week the usb disconnection occured more and more frequently. Now I'm at the point that I cant even manage anything (including my heating systems) because when I instantly plug the device in, it start the disconnection loop immediately. I know my comment is not helping anyone, just wanted to make a notice that this issue can still happen to even. I'm on the current latest 29-Apr-2021 version but as I wrote, I tried every version available with every physical connection possibilities.

closed 2021-06-05 by github-actions[bot]

ReviloEgros, 2021-07-22:
I have exactly the same Problem on Ubuntu 21.04 and kernel 5.11.0-25-generic

SeanHama, 2021-08-10:
Did anyone find a fix for this? So annoying this bug

Saviq, 2021-08-10:
> Did anyone find a fix for this? So annoying this bug

I ended up getting a different dongle...

deviantintegral, 2021-08-10:
Same - I switched to a zzh! and have had zero issues with USB disconnects or resets, including inside of KVM. Actually, I've since added a z-wave stick and two rtl_433 radios, and they all work cleanly without resets as well, so I'm very confident the issue was with the ConBee II itself.

SeanHama, 2021-08-10:
I found the issue plaguing my setup (only if the Phoscon web app had an alert would have saved me two day).

My issue with the firmware “Not connected” was relating to Zigbee2mqtt and Mosquitto being both active running services on my Raspberry PI and listening on the same ports which Conbee2 requires port 8080 and port 80. 

I disabled both those services and restarted the deconz service and it’s back online and connecting to the firmware. 

Fingers crossed it holds. 

I’m going to add in a wiki page to this GitHub repo to help others and save them the back and forth.

Mimiix, 2021-08-11:
For those who still ahve this issue, try updating to the new firmware for Conbee II. IT has a lot of improvements on this subject.

martinsheldon, 2021-08-17:
> 
> 
> For those who still ahve this issue, try updating to the new firmware for Conbee II. IT has a lot of improvements on this subject.

Updated the firmware a few days ago. Still the same issue here. Endless connect/disconnects on a Windows 10 system and VirtualBox

Mimiix, 2021-08-17:
@martinsheldon that's probably due to VirtualBox. It's not reliable at all with usb devices.

martinsheldon, 2021-08-18:
Maybe. I'll try to go back to RPI4B within the next few days for testing. 

Would it make sense that the VirtualBox can make the Conbee II disconnect/connect in a endless loop at the Windows computer? When it happens the Windows USB disconnect sound plays with continuously with a interval of a few seconds.

I love everything about the stick except from the fact that its crashing way to often :(

Mimiix, 2021-08-18:
It is getting disconnected from Windows to the VirtualBox. And apparently that ejects it or so.

Wondering if you can get it working on windows. If that has the same behavior, i am wondering if you clicked "Firmware update"

martinsheldon, 2021-08-18:
Okay. I have not tried to run it directly in Windows but it's worth a try. 

Regarding the FW upgrade I downloaded the latest FW from the webside and upgraded by following this guide:
https://flemmingss.com/how-to-update-conbee-conbee-ii-firmware-in-windows-10/

Mimiix, 2021-08-18:
Did you use the VM?

martinsheldon, 2021-08-18:
No, I updated it on my laptop (Different computer than my "server")

Mimiix, 2021-08-18:
Should be fine. 
Check if it works on windows if you can:)

martinsheldon, 2021-08-18:
I'll do that! 

I will keep you updated on the result and think 4-5 days without crashing is needed to say the issue fixed. If it crashes before that I'll let you know, if not you'll hear from me during the weekend

Mimiix, 2021-08-18:
> I'll do that!
> 
> I will keep you updated on the result and think 4-5 days without crashing is needed to say the issue fixed. If it crashes before that I'll let you know, if not you'll hear from me during the weekend

Also, check if it works with VMWare Player. VirtualBox oftenly causes issues with usb devices. Had a lot of users with that on discord.

martinsheldon, 2021-08-18:
Okay. Which method has, in you opinion, the best chance of success? 

I don't really mind what solution I go for as long as its working and stable

Mimiix, 2021-08-18:
I'd say:

Update to the beta firmware (http://deconz.dresden-elektronik.de/deconz-firmware/beta/deCONZ_ConBeeII_0x26700700.bin.GCF)

See if it works with Windows _native_ not virtualized. Native OS (instead of VM) is always prefered.

If you want to run a VM: use VMware Player.

martinsheldon, 2021-08-18:
Thanks again. 

Beta firmware is already up to date. 
I'll try native Windows first then

martinsheldon, 2021-08-22:
Hi again,

I quick update on the status. 

As usual things never go exactly as planned...
First of all I decided to go back to the PI instead of using a Windows computer as I know this works pretty fine for me. I started this progress on thursdag afternoon and was done (mostly at least) the same day. The plan was to let it run without doing any major changes or reboots over until the end of the weekend. This went well until friday afternoon when there was like 3 or 4 shorter or longer power outages in my area...

Anyway, it has been stable since the last outage on friday afternoon. No disconnects or crashes. Still want to wait another few days before I can confirm the issue is resolved.

sroebert, 2022-01-12:
I have exactly this problem on the 21.10 version of Ubuntu on a Raspberry Pi (3 or 4). I thought at first it had to do with the connected SSD to USB 3 even though I was using an extension cord for the Conbee.

After more trial and error however, it seems to be that the disconnect of the Conbee only happens on Ubuntu 21.10. It seems to work fine on Ubuntu 20.04. Could it be issues with the latest kernel?

Mimiix, 2022-01-12:
@sroebert Seems to be a new issuie in that case. logs would help :)

sroebert, 2022-01-12:
What kind of logs can I provide? The dmesg logs look a lot like https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4029#issuecomment-816961641

Mimiix, 2022-01-12:
@sroebert Deconz logs. APS, APS L2, ERROR, ERROR L2, INFO, INFO L2.

sroebert, 2022-01-12:
Sure, I'll have a look later when I have access to the devices again.

tedpelas, 2023-10-23:
> Sure, I'll have a look later when I have access to the devices again.

Got access to the devices again? :)