dresden-elektronik/deconz-rest-plugin #4031: SonOff SNZB-03 no connection

wemach, 2020-12-26:
## Describe the question or issue you are having
I wanted to add a SonOff SNZB-03 motion sensor, I see it in the deConz App, but I get no connection to it and no device/entity in HA. I wanted to report this in the [linked Issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4030) from [deconz wiki supported devices](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices), but this issue is closed.

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi 4B 4GB
 - Running method:  Home Assistent deCONZ Add-on
 - Firmware version: 26610700
 - deCONZ version: 2.07.01
 - Device: RaspBee II
 - Do you use an USB extension cable:  no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
20:38:26:824 dev /dev/ttyAMA0
20:38:26:825 GW firmware version: 0x26610700
20:38:36:821 dev /dev/ttyAMA0
20:38:36:822 GW firmware version: 0x26610700
20:38:46:822 dev /dev/ttyAMA0
20:38:46:823 GW firmware version: 0x26610700
20:38:47:473 sensor 38 (lumi.sensor_motion): disable presence
20:38:48:073 Set sensor check interval to 1000 milliseconds
20:38:51:817 Current channel 15
20:38:51:839 Device TTL 6180 s flags: 0x7
20:38:53:796 ZCL attribute report 0x00158D00058A63BE for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:38:53:797 0x00158D00058A63BE extract Xiaomi special attribute 0xFF01
20:38:53:797 	01 battery 3105 (0x0C21)
20:38:53:798 	04 unknown 5032 (0x13A8)
20:38:53:798 	05 RSSI dB (?) 6 (0x0006)
20:38:53:799 	06 LQI (?) 4294967296 (0x0100000000)
20:38:53:799 	64 temperature 2160
20:38:53:799 	65 humidity 4788
20:38:53:800 	66 pressure 96656 (967)
20:38:53:800 	0a Parent NWK 0 (0x0000)
20:38:56:821 dev /dev/ttyAMA0
20:38:56:822 GW firmware version: 0x26610700
20:38:59:579 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:00:273 Set sensor check interval to 100 milliseconds
20:39:05:112 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:06:828 dev /dev/ttyAMA0
20:39:06:829 GW firmware version: 0x26610700
20:39:10:615 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:16:826 dev /dev/ttyAMA0
20:39:16:827 GW firmware version: 0x26610700
20:39:18:459 ZCL attribute report 0x00158D000588933F for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:18:499 ZCL attribute report 0x00158D000588933F for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:18:521 ZCL attribute report 0x00158D000588933F for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:20:712 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:26:822 dev /dev/ttyAMA0
20:39:26:823 GW firmware version: 0x26610700
20:39:27:250 ZCL attribute report 0x00158D00058A618E for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:27:251 0x00158D00058A618E extract Xiaomi special attribute 0xFF01
20:39:27:252 	01 battery 3085 (0x0C0D)
20:39:27:252 	04 unknown 5032 (0x13A8)
20:39:27:253 	05 RSSI dB (?) 9 (0x0009)
20:39:27:253 	06 LQI (?) 4294967296 (0x0100000000)
20:39:27:253 	64 temperature 2000
20:39:27:254 	65 humidity 5271
20:39:27:254 	66 pressure 96606 (966)
20:39:27:255 	0a Parent NWK 0 (0x0000)
20:39:28:589 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:35:563 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:36:828 dev /dev/ttyAMA0
20:39:36:829 GW firmware version: 0x26610700
20:39:46:219 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:46:821 dev /dev/ttyAMA0
20:39:46:822 GW firmware version: 0x26610700
20:39:51:592 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:39:51:819 Current channel 15
20:39:51:841 Device TTL 6120 s flags: 0x7
20:39:56:824 dev /dev/ttyAMA0
20:39:56:824 GW firmware version: 0x26610700
20:40:06:824 dev /dev/ttyAMA0
20:40:06:825 GW firmware version: 0x26610700
20:40:16:824 dev /dev/ttyAMA0
20:40:16:824 GW firmware version: 0x26610700
20:40:22:450 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:40:26:824 dev /dev/ttyAMA0
20:40:26:825 GW firmware version: 0x26610700
20:40:30:308 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:40:35:416 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:40:36:823 dev /dev/ttyAMA0
20:40:36:824 GW firmware version: 0x26610700
20:40:41:857 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:40:46:825 dev /dev/ttyAMA0
20:40:46:825 GW firmware version: 0x26610700
20:40:51:153 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:40:51:818 Current channel 15
20:40:51:839 Device TTL 6060 s flags: 0x7
20:40:56:821 dev /dev/ttyAMA0
20:40:56:822 GW firmware version: 0x26610700
20:40:58:310 ZCL attribute report 0x00158D00058A618E for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:40:58:347 ZCL attribute report 0x00158D00058A618E for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:40:58:371 ZCL attribute report 0x00158D00058A618E for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:41:04:061 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:41:06:829 dev /dev/ttyAMA0
20:41:06:830 GW firmware version: 0x26610700
20:41:09:877 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:41:15:221 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:41:16:829 dev /dev/ttyAMA0
20:41:16:830 GW firmware version: 0x26610700
20:41:21:436 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:41:26:824 dev /dev/ttyAMA0
20:41:26:825 GW firmware version: 0x26610700
20:41:27:068 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:41:32:555 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:41:36:824 dev /dev/ttyAMA0
20:41:36:825 GW firmware version: 0x26610700
20:41:38:321 ZCL attribute report 0x00158D0004AB7358 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
localhost - - [26/Dec/2020 20:41:53] 127.0.0.1:5900: Client closed connection
localhost - - [26/Dec/2020 20:41:53] 127.0.0.1:5900: Closed target
20:41:44:739 ZCL attribute report 0x00158D0004AB7358 for clus2020/12/26 20:41:53 [info] 1516#1516: *865 recv() failed (104: Connection reset by peer) while proxying upgraded connection, client: 172.30.32.2, server: _, request: "GET /novnc/websockify HTTP/1.1", upstream: "http://127.0.0.1:5901//websockify", host: "homeassistant.local:8123"
172.30.32.2 - - [26/Dec/2020:20:41:53 +0100] "GET /novnc/websockify HTTP/1.1" 101 4235954 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36 Edg/87.0.664.66"
Ignoring interrupted syscall

## Additional context
I'm new on smart home and home-asstistant, so it I would be verry happy to get some help.

SwoopX, 2020-12-26:
I assume this is because it didn't pair correctly in the first place. You can see that on the unusual node name being its NWK address. Have you tried again?

There's devices out there which are rather sensitive having an IAS zone cluster which didn't got enrolled properly (yours has one). They more or less instantly leave the network again in those cases...

wemach, 2020-12-27:
Thank you, I've tried retried once again and now the node name is different:
[image]
[image]
[image]

In PhosconGui I got the success message, but no new sensor. Nevertheless I got a new entity in HomeAsssistant, but the binary sensor ist not working, no motion is detected :(

SwoopX, 2020-12-27:
Ok, not seeing it in Phoscon is expected (for now). You now need to manually enroll the device:

1. Open deCONZ.
2. Select the Coordinator (blue node) on the map and copy value of the IEEE field.
3. Select the Motionsensor, click the "ias zone" cluster. 
4. Scroll down to the attributes list to attribute 0x0010
5. double click the "value"
6. put the IEEE value you copied earlier in the textbox above. 
7. Make sure the Sensor is awake. This can be done by clicking the reset button once.
8. Click "Write"

wemach, 2020-12-27:
Thank you!
I've tried, but I don't know where to paste the value :(

> 3. Select the Motionsensor, click the "ias zone" cluster.
> 4. Scroll down to the attributes list to attribute 0x0010

I guess I should open the ias  zone cluster here, but I can't, nothing happens when I click on 0500 IAS Zone..
[image]

SwoopX, 2020-12-27:
Lower left corner should have a cluster info panel. Click on it and scroll to the bottom.

wemach, 2020-12-27:
Thank you, I think I got it know how to enroll manually, but unfortunately writing failed, although I pressed reset button once.
[image]
I will delet node an try again.

SwoopX, 2020-12-27:
Delete/re-pair shouldn't be necessary. Just keep trying a bit.

wemach, 2020-12-27:
I tried three times without success, then I deleted the node an wrote the IAS_CIE_Adresse just after reconnecting, and it was successfull :)

Now I get the motion messages also in the binary sensor in HA :)

@SwoopX Thank you very much for your help!!

SwoopX, 2020-12-27:
Closing this one as resolved.

closed 2020-12-27 by SwoopX

wemach, 2020-12-27:
Just a question to understand:
Some aqara motion sensors do also have a IAS Zone and in this there is not mentioned the IEEE adress of the configurator, but they work fine. When is the configurator adress necessary and when not?

[image]

SwoopX, 2020-12-28:
No, for the Aqara's, it's not necessary. Those are fake clusters. They work entirely differently.