dresden-elektronik/deconz-rest-plugin #4035: Xiaomi Aqara Motion sensor unavailable in Phoscon / seems to "react" in Deconz

tosion, 2020-12-27:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

SwoopX, 2020-12-27:
Can you provide some deconz logs? Around 5 mins should suffice. Please make sure that the timeframe encompasses at least 2 mins where ther's no motion guaranteed and then motion is triggered. The log levels must be dbg-info=2 and dbg-error=2.

When we'd see an attribute report of the sensor sending occupancy = true on ZCL level, that would confirm the sensors is operational and sending status updates.

tosion, 2020-12-27:
Thanks for a super fast response!

As mentioned I am far away from this sensor so I can not create a motion remotely.  With logs are you meaning deconz addon logs? But if I can not create a motion?

I also would need some help how to provide this REST API output?

BTW how can I label my question as "User Quoestion" insted of "Bug Report"

tosion, 2020-12-29:
More info for this issue. I have 6+7 of these Xiaomi Aqara/Mija sensors in two different location. I another location 7 sensors work very well, they react as supposed and in HA/Phoscon/DeconzVNC they seem to "work" OK.

From six sensors in another location 2 work OK in HA/Phoscon/VNC.

One is not available in HA and is greyed and not available in Phoscon, but seem to work OK in VNC.

Three are available in HA, they are not greyed in Phoscon but in sensor details in Phoscon they are not reachable (pic below). In VNC they seem to be OK?

One more observation which is not related to this is that in sensor detail last seen seem to be in German even the language selection is in English (is there a reason to open a new bug report?):

[image]

[image]

SwoopX, 2020-12-29:
Are you able to provide some screenshots of the troublesome sensors from deconz GUI (via VNC) with an expanded cluster list (right most black bullet on the node)?

The symptoms would fit in, check https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

tosion, 2020-12-29:
Thanks! You do not mean that picture I had in the first post? Or d you mean like basic cluster sheet?

[image]

SwoopX, 2020-12-29:
Sorry, overlooked your previous screenshot while writing.

The Aqara I got here looks identical from the clusters perspective. Interesting fact around the lastseen though...

Could you please provide the REST API output from your most troublesome Aqara sensor? I'd be interested in the order of the 2 corresponding sensors.

tosion, 2020-12-29:
OK thanks!

I think I found (the picture below that log you mentioned earlier (I do not know how to copy it though). But can you help me out where to get that RES API output?

[image]

Mimiix, 2020-12-29:
@tosion In case of the HA addon: this is a limitation by how they implemented the VNC in the browser. I recommend setting up VNC itself and use a VNC client to connect.

tosion, 2020-12-29:
I can do VNC/VNC - but is it still limited what I can do there. But yes please can you help how to get REST API output with VNC/VNC so I can try it. ( My best bet is I can not do it when running Deconz Addon in HA)

tosion, 2021-01-01:
Last update for this case. I took a long drive to the location of these sensors. 

Three sensor which were available in HA but greyed in Phoscon became active in Phoscon when motion was triggered - so they worked well HA/Doconz also (I don't know why they were not reachable in Phoscon - now they are)

The one which was not available in Phoscon and HA needed a repair -> worked OK after that. 

Even I do not quite understand this behavior this is pretty much case closed from my side.

Thanks for our help anyway! I do not know should I close this because this "bug" is not solved yet?

closed 2021-02-08 by Mimiix