dresden-elektronik/deconz-rest-plugin #4038: Lidl Melinera Smart LED string lights usage

Preto, 2020-12-27:
## Describe the question or issue you are having
This was awesome! Had a lot of Christmas fun thanks to you guys and Lidl :-)

I used a lot of different effects and used RGB colors in them, all flawlessly. I am only unable to use the _colour mode_ correctly. I'm on the latest version and removed and repaired the lights as stated in the support issue with the following steps:

 - Removed device in GUI view, then pressed delete in Phoscon
 - Put Phoscon in search light mode
 - Repaired by pressing F button > 5 seconds
 - Got recognized as Fairy Lights (model), only with a new ID.

**Reproducible steps**
1. Setting the _white mode_ according to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3716#issuecomment-735467996:
```bash
curl -d '{ "bri":254, "sat":0 }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state"
```
&emsp;&emsp;The API returns a success for `bri` and a separate success for the `sat`. The REST-API reflects all this too.
&emsp;&emsp;The lights however, show shortly the _white mode_ but quickly reverts to _colour mode_. In my case it flashes
&emsp;&emsp;white and returns to the color green.

2. Issuing two commands in quick succession does keep the _white mode_:
```bash
curl -d '{ "bri":254, "sat":0 }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state" ; curl -d '{ "effect":"none" }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state"
```
&emsp;&emsp;The API returns a success for `bri`, a separate success for `sat` and the second command returns also a success
&emsp;&emsp;on `effect`. The lights look the way they should and the REST-API reflect this too.

3. So lets try to combine those commands:
```bash
curl -d '{ "bri":254, "sat":0 , "effect":"none" }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state"
```
&emsp;&emsp;The API returns **only** a success on `effect` and ignoring the other two. Is this expected behavior?
&emsp;&emsp;Should I fire separate commands at the API?

4. Lets change the color to the mentioned yellow color (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3716#issuecomment-736123450):
```bash
curl -d '{ "hue": 11000 , "sat": 254 }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state"
```
&emsp;&emsp;The API returns a success on `hue` and `sat`. The color is bright yellow (as bri is still 254)

5. Throwing a `bri` in the mix with the above command:
```bash
curl -d '{ "hue": 11000 , "sat": 254, "bri":100 }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state"
```
&emsp;&emsp;The API returns a success on all 3, very bright (green/yellow) if you ask me. Pretty okey.

6. Pumping up the brightness to its max:
```bash
curl -d '{ "hue": 11000 , "sat": 254, "bri":254 }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state"
```
&emsp;&emsp;The API returns again a success on all 3, but the brightness just went down! Testing around it seems the
&emsp;&emsp;brightness went to ~50. But this isn't reflected anywhere, just figured by looking at it. So if you ask me, in _colour mode_
&emsp;&emsp;the `bri` can only be maxed to 100. Anything above 100 will lower the brightness of the actual lights. Is this
&emsp;&emsp;assumption correct?

7. So if I only want the _white mode_ and not so very bright, I am only able to do this with issuing two separate commands:
```bash
 curl -d '{ "bri":50, "sat":0 }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state" ; curl -d '{ "effect":"none" }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state"
```
&emsp;&emsp;This turns the lights to a low brightness. Having a `bri` of 100 is a bit brighter and a `bri` of 254 seems to be
&emsp;&emsp;correctly the maximum. All good here! But why not able to combine those two commands in one?

8. Playing a bit more shows me that _white mode_ can also be accomplished by issuing any `hue` number, but having `sat` at `0`.
```bash
curl -d '{ "hue": 44888, "sat":0 }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state"
```
&emsp;&emsp;I get a success on both and the lights are on that _white mode_ color.

9. Adding `bri` to the above command adds to the confusion on how to use `bri` the correct way:
```bash
curl -d '{ "hue": 44888, "sat":0, "bri":100 }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state"
```
&emsp;&emsp;Success on all 3, lights are very bright blue as I think they should be. But if you use the above command and
&emsp;&emsp;change `bri` to `254` the lights turn less bright and more purple-ish. So again here I think `bri` maxed to 100. All
&emsp;&emsp;above results in a decrease in brightness and therefor color change?

Is this all expected behavior or am I simply using it all wrong?
If so, how do I lower the brightness in _white mode_ and _color mode_?

## Screenshots
N/A

## Environment

 - Host system: Raspberry Pi 3B
 - Running method: Ubuntu 20.04
 - Firmware version: 26660700
 - deCONZ version: 2.09.00-raspbian-buster-alpha
 - Device: ConBee II
 - Do you use an USB extension cable: no, works flawlessly
 - Is there any other USB or serial devices connected to the host system? Yes, P1

## deCONZ Logs
N/A

## Additional context
_I wanted to update the closed issue #3716 with this user question, but was unable to. Therefor a new one._

ebaauw, 2020-12-27:
> deCONZ version: 2.09.00-raspbian-buster-alpha

Not sure when that was released; it's 2.09.00-raspbian-buster-beta on PiOS.

From the symptoms you're describing, it would seem that my _Merry Christmas 4_ PR is missing from your distribution.  Before that PR, the API would send a regular _Move to Level_ command for `bri`, after setting the colour (or white mode) using the Tuya-like command.  This was an attempt to keep the _Current Level_ attribute up to date, but the light strip screws up its state on _Move to Level_, so _Merry Christmas 4_ removes that command.

1. The API should send one response, which is a JSON array, with an object per attribute that was set successfully.
2. Setting `effect` to "none"` shouldn't be needed: when you see `bri`, `hue`, and/or `sat`, the strip switches mode automatically.
3. That's probably a bug in the API.  `effect` trumps `bri`, `hue`, and `sat`, and I probably don't check for `"none"`.  I think I did for regular lights - need to check.
4. So that's OK?
5. So that's OK?
6. Cannot reproduce that.  Note that the API uses `hue`, `sat`, and `bri` to create the Tuya command to set the light.  If any of these is missing, the current value is substituted.  So setting `bri` to 254 should be no different from setting `bri` to 254 and `hue` and `sat` to the current values.
7. The single command  works for me.
8. Correct.  With `sat` 0 the `hue` value doesn't matter.  That's how HSV works (and polar coordinates in general).
9. Again, I cannot reproduce this.

Preto, 2020-12-28:
Thanks for your quick reply.

**tldr**; This is solved by a restart. Although its a _pebcak_ on my end, it might be for DE interesting why deCONZ isn't triggering a restart upon (manual) install (`dpkg -i`) and for you why `effect` is stopping further processing of commands on this light.

----

In regards to the version, when I installed Ubuntu 20.04 the deCONZ releases wheren't stable on a 64 bits OS on the pi. So I reinstalled Ubuntu to the 32 bits release and was only able to install the deCONZ version in the [Alpha](http://deconz.dresden-elektronik.de/raspbian/alpha/) directory due to the architecture (No armv7l repo on 32 bits). The alpha file date (24-12-2020 @ 11:27) matches the latest release.

I just tried the one in the beta directory and it seems I can install it now (file [deconz-2.09.00-qt5.deb](http://deconz.dresden-elektronik.de/raspbian/beta/deconz-2.09.00-qt5.deb)). deCONZ **wasn't restarted** upon this manual install, and it seems it didn't restart upon the previous install too. Grrr, your right that the last update didn't made it then. Thought I'd checked everything before bothering anyone... sorry 'bout that.

dpkg is showing `2.09.00-raspbian-stretch-beta` now. No Buster it seems, but I'm fine with that. As 64 bits is available now, I will reinstall to 64 bits so I can use the repositories again. Would make things easier.

Repaired the lights, just to make sure. For future reference, I'll give it another pass:
1. Correct
2. Works flawless without the use of effect indeed
3. That bug is still there, API only reacts on `effect` and ignores others
4. Good
5. Good
6. Works now flawlessly!
7. Single commands works good now!
8. Thanks, wasn't aware of this. For anyone in the need of hue colors, I used [this](http://www.workwithcolor.com/blue-color-hue-range-01.htm) site. Blue is between 221° to 240°. If we pick 240° that would be (`240 * 65535 / 360`) a hue value of `43690` for feeding the API.
9. Works flawless now

I just bought a few more, they sell for as low as EUR 39,- now!

I'll leave the issue open for you / DE to decide what to do with the packaging and effect bug and close it on your convenience.

ebaauw, 2020-12-28:
I cannot help with the packaging, but typically, you need to restart deCONZ after upgrading, also when using `apt`.

Above commit should fix the bug.

Preto, 2020-12-29:
Thanks for the fix!
Does that also fix other things slipping through the parameter parser? For instance, you will get both an error and a success when sending commands while the light is set to off. I would expect only an error. See below:

Turn the lights off:
```bash
curl -d '{ "on":false }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state"
```

Now modify any parameter:
```bash
curl -d '{ "bri":100 }' -X PUT "http://192.168.1.1:8080/api/KEY/lights/1/state"
```
You will get an error, which is the case for other lights as well. The light stays off. I expect this to be normal behavior. But the API is also returning a success, although doing nothing of course:
```json
[
    {
        "error": {
            "address": "/lights/1/state",
            "description": "parameter, bri, is not modifiable. Device is set to off.",
            "type": 201
        }
    },
    {
        "success": {
            "/lights/11/state/bri": 100
        }
    }
]
```

If not, than that success message might be a bug as well?

ebaauw, 2020-12-29:
Error handling and reporting by the deCONZ REST API is a mess.  It's inconsistent and differs from the Hue bridge API.  Not that the Hue bridge API one is more consistent in this area...

It's perfectly valid for a response to a PUT to contain both "success" and "error" entries in the response, e.g. when specifying a valid and an invalid attribute in the body.  It is a bit weird for the same attribute to appear in a "success" and an "error" response entry.

I found out by accident (while sniffing) that, when setting a light state on the Hue bridge for a light that's off (at least according to the Hue bridge cached state), the Hue bridge will send the Zigbee command anyways, while issuing the "error" response entry.  I tried to mimic that for the Xmas light, but that causes an additional "success" response entry, since the command is sent alright.  For other lights, deCONZ won't send the command, and you won't see a "success" response.  I still want to bring that more in line with the Hue bridge.

I also found that deCONZ refuses to send the Zigbee command when the light is marked unreachable, see #4021.  The Hue bridge doesn't even seem to take this attribute into consideration.

ma-ca, 2020-12-29:
The Xmas light has two buttons, C and F. Pressing the C button multiple times iterates the lights over seven different colors, then white and finally the light string has multi-color.

Is this last setting with multi-color also available from the REST API?

Preto, 2020-12-29:
Nice job! Its actually pretty cool that you can change the light state while it's off, basically queuing for whenever you turn it on again. The API response is indeed a bit confusing though, but yeah... there is always room for improvement.

Thanks for your efforts again, these lights pretty much stole the (christmas) show.

---

@ma-ca: I saw that _multicolor_ question before in a [comment](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3716#issuecomment-738777143). Unfortunately I don't think there was a solution though. I believe I found that _mode_ by accident while playing with the `"effect":"none"` as mentioned above and going / coming from `carnival` or `carnaval` (_i had a typo before, but didn't noticed it as I think something happened_) and not sending any colors along to get the default ones. I cannot reproduce it anymore on the latest release. If you reallllyyy want it, I suggest reverting to version `2.07.01` (with broken brightness) and play around with that as that was my version during Christmas where I had it stable on random colors. If you find it also, please do share your findings.

---

@manup: Can you share DE's thoughts about Debian/Ubuntu packaging? Is it on purpose that a restart isn't triggered upon package install/update?

Secondly, in the past I had stability issues with deCONZ on the rpi 3b with Ubuntu 20.04 on 64 bits. I am a bit confused on support also on the rpi 4.

The [Phoscon](https://phoscon.de/en/conbee2/install#ubuntu) website shows that 20.04 is supported with a 64 bits architecture. But the [Wiki](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-on-Arm64-aarch64-Ubuntu-Focal,-Debian-Stretch.) here on Github states its unsupported, with big fat letters. I go with supported as that's more _official_ right?

closed 2021-01-12 by manup