dresden-elektronik/deconz-rest-plugin #4042: deconz devices respond with a delay

vantino, 2020-12-28:
## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: (Raspberry Pi  3+ Model B)
 - Running method: (Raspbian headless)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.07.01)
 - Device: (RaspBee II)
 - Do you use an USB extension cable: ( no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mimiix, 2020-12-28:
What I've seen in cases like these, is that it mostly due to a slow sd card or a overloaded pi.

Or, of course, network issues. That would show in the logs with routes being unavailable.

vantino, 2020-12-28:
Hm, thanks for reply.

CPU load is constantly 8 -13% and 18% sporadically,
no network issues found
I use SanDisk Ultra 64GB Class 10 
no power issues, like "low voltage" ...

However, will check cpu load next time

vantino, 2020-12-29:
Thanks for your support!

With more than 12h runtime w/o any issues it looks like to be a network problem, as you suguested initially.  I found massive msg's in kern.log and broadcast msg's in syslog with massive increasing size of both files.

However, the route course is the "Phillips Hue Smart Button", which is the only change in the new system.
I removed its battery and my fhem (13 lights, 3 sensors, 2 switches) works as it did the last 18 month.

Finally - 

Phillips Hue Smart Button interferes Raspi 3+ Mod B "brcmf_cfg80211" module.

CHeers
Roland

closed 2020-12-29 by vantino