dresden-elektronik/deconz-rest-plugin #4044: Latest firmware for Conbee 1?

oscar-b, 2020-12-28:
## Describe the question or issue you are having

What's the latest firmware that should be used for the Conbee 1? Is it the "RPi" firmware? `deCONZ_Rpi_0x26390500.bin.GCF`?

Does this include the last routing/counter fixes which were made for Conbee 2, or is the 1 not supported any more?

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

benjaml4, 2021-02-08:
Would like to know the same...!

Mimiix, 2021-02-08:
http://deconz.dresden-elektronik.de/deconz-firmware/

deCONZ_Rpi_0x26390500.bin.GCF   

Yes, it does include those.

closed 2021-02-08 by Mimiix