dresden-elektronik/deconz-rest-plugin #4045: How to Turn off Heiman Siren.

AbDhops, 2020-12-28:
## Describe the question or issue you are having

I am using Heiman WarningDevice (Siren) with Deconz. It is paired with Home Assistant and Node Red. 
For some of other reason (water leakage), Node red automation triggered an alarm. So far so good. However, even after I switch off the alarm, device didnt stopped ringing. Even though the device was ringing (loud !!), status of Switch in Deconz was off and siren as sensor was also off.

## Screenshots

## Environment

 - Host system: NAS
 - Running method: Home Assistent deCONZ Add-on / Marthoc Docker container on Synology
 - Firmware version: 26660700
 - deCONZ version: 2.05.81 / 9/14/2020
 - Device: ConBee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? : No

## deCONZ Logs

## Additional context

RealKanashii, 2020-12-28:
Can you post the node red switch? Maybe you don't used the passthrough payload?

AbDhops, 2020-12-28:
I am directly using "turn_off" service... 
[image]

AbDhops, 2020-12-28:
So, I debugged further. I deleted the Alarm from every where (Home Assistant, Deconz and Node red). Also reseted the device physically. And added again the device as different name i.e. Siren. 
Now, this is visible in Home Assistant as Siren. But multiple devices. See below.

[image]

Now, if I turn on switch.siren, siren goes On. However, switch status (which I had turned On), goes back to off. Siren continues to ring. To switch it off, I must "Turn On" the switch and make it off again while it is in "On" state. Once it goes back to off I can't turn off the siren.

light.siren and sensor.siren are basically useless sensors. They never change the status. 

From PhosconGUI, every thing remains off. No matter if siren is ringing...

Smanar, 2020-12-28:
Wich one decone filed are you changing with your automation ? for siren it's not state/on and state/off.

AbDhops, 2020-12-30:
I am not using Deconz field. I am using Home Assistant Deconz integration. In it, device is visible as siren. See image from my previous [post ](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4045#issuecomment-751791861).

If I turn on switch.siren (from Home Assistant), siren starts ringing. However, switch status (which I had turned On in home assistant), goes back to off state automatically. Siren continues to ring. To switch it off, I must "Turn On" the switch (again in Home Assistant) and make it off again while it is in "On" state. Once it goes back to off (it goes off automatically in Home Assistant) I can't turn off the siren.

light.siren and sensor.siren are basically useless sensors in (Home Assistant). They never change the status.

From PhosconGUI, everything remains off. No matter if siren is ringing or not (by using switch in Home Assistant). Also, I am not able to turn on Siren from Phoscon GUI...

SwoopX, 2020-12-30:
... because this is not how it works.

AbDhops, 2020-12-30:
Can you please elaborate... How it works... ?

Smanar, 2020-12-30:
You don't know the command used by HA ?

AbDhops, 2020-12-30:
No....

SwoopX, 2020-12-30:
All sirens are exposed as lights, e.g. like this via REST API

```
"1": {
    "etag": "f57956703c228334a640334d6b7505c3",
    "hascolor": false,
    "lastannounced": "2020-12-23T14:54:28Z",
    "lastseen": "2020-12-28T21:20Z",
    "manufacturername": "Develco Products A/S",
    "modelid": "SMSZB-120",
    "name": "fl.sensor.rauchmelder",
    "state": {
      "alert": "none",
      "reachable": true
    },
    "swversion": "2018-11-27 00:40",
    "type": "Warning device",
    "uniqueid": "00:15:bc:00:31:00:42:94-23"
  },
```

They can be triggered and silenced by sending a PUT request through the API and modify the alert parameter

[image]

So,
`{"alert": "select"}` lets it sound for 1 sec, `lselect` for maximum duration
`{"alert": "none"}` silences it

AbDhops, 2021-01-04:
Thanks @SwoopX,
Things started to make sense now. But how can I do that using node red ? Obviously as "alert" is not standard property for "light", `{"alert": "select"}` triggers an `API error`....

og9gjz567, 2021-01-06:
Hi,
my solution was to do it in node-red with a http-call. First I define the params in a function node:
`msg.method = "PUT"
msg.url = "http://172.30.33.2:40850/api/xxxxxxxxxxxxx/lights/xx:xx:xx:xx:xx:xx:xx:xx-01/state" ;

msg.payload = '{ "alert": "blink" }'; //only blinks
//msg.payload = '{ "alert": "select" }'; //siren for ever
//msg.payload = '{ "alert": "lselect" }'; //sire for 3 seconds
//msg.payload = '{ "alert": "none" }'; //disable alarm
msg.headers = {};
msg.headers['content-type'] = 'application/json';
return msg;`

ans then I use the http request node to put the request. You can get the IP of the REST API by clicking "plugins" and then "REST API Plugin" in the "IP Addresses" Section. Hope this helps to:

[image]

AbDhops, 2021-01-06:
Can you please export your flow and paste it here...

og9gjz567, 2021-01-06:
yes:

`[{"id":"87a51622.21d23","type":"tab","label":"export","disabled":false,"info":""},{"id":"182092e5.fba0dd","type":"inject","z":"87a51622.21d23","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":180,"y":160,"wires":[["b5c915dd.645e08"]]},{"id":"3e3062c4.72a4c6","type":"inject","z":"87a51622.21d23","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":180,"y":240,"wires":[["5a88c2c4.cc51bc"]]},{"id":"b5c915dd.645e08","type":"function","z":"87a51622.21d23","name":"Sirene: aus","func":"msg.method = \"PUT\"\nmsg.url = \"http://192.168.178.100:40850/api/XXXXXXXXX/lights/xx:xx:xx:xx:xx:xx:xx:xx-01/state\" ;\n\n//msg.payload = '{ \"alert\": \"blink\" }';\n//msg.payload = '{ \"alert\": \"select\" }';\n//msg.payload = '{ \"alert\": \"lselect\" }';\nmsg.payload = '{ \"alert\": \"none\" }';\nmsg.headers = {};\nmsg.headers['content-type'] = 'application/json';\nreturn msg;\n\n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":410,"y":160,"wires":[["77494fde.91d75"]]},{"id":"da8b454d.fd5fb","type":"function","z":"87a51622.21d23","name":"Sirene: Alarm unendlich","func":"msg.method = \"PUT\"\nmsg.url = \"http://192.168.178.100:40850/api/XXXXXXXXX/lights/xx:xx:xx:xx:xx:xx:xx:xx-01/state\" ;\n\n//msg.payload = '{ \"alert\": \"blink\" }';\n//msg.payload = '{ \"alert\": \"select\" }';\nmsg.payload = '{ \"alert\": \"lselect\" }';\n//msg.payload = '{ \"alert\": \"none\" }';\nmsg.headers = {};\nmsg.headers['content-type'] = 'application/json';\nreturn msg;\n\n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":450,"y":200,"wires":[["77494fde.91d75"]]},{"id":"5a88c2c4.cc51bc","type":"function","z":"87a51622.21d23","name":"Sirene: Alarm","func":"msg.method = \"PUT\"\nmsg.url = \"http://192.168.178.100:40850/api/XXXXXXXXX/lights/xx:xx:xx:xx:xx:xx:xx:xx-01/state\" ;\n\n//msg.payload = '{ \"alert\": \"blink\" }';\nmsg.payload = '{ \"alert\": \"select\" }';\n//msg.payload = '{ \"alert\": \"lselect\" }';\n//msg.payload = '{ \"alert\": \"none\" }';\nmsg.headers = {};\nmsg.headers['content-type'] = 'application/json';\nreturn msg;\n\n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":420,"y":240,"wires":[["77494fde.91d75"]]},{"id":"6e8a40bd.d0003","type":"function","z":"87a51622.21d23","name":"Sirene: blinken","func":"msg.method = \"PUT\"\nmsg.url = \"http://192.168.178.100:40850/api/XXXXXXXXX/lights/xx:xx:xx:xx:xx:xx:xx:xx-01/state\" ;\n\nmsg.payload = '{ \"alert\": \"blink\" }';\n//msg.payload = '{ \"alert\": \"select\" }';\n//msg.payload = '{ \"alert\": \"lselect\" }';\n//msg.payload = '{ \"alert\": \"none\" }';\nmsg.headers = {};\nmsg.headers['content-type'] = 'application/json';\nreturn msg;\n\n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":420,"y":280,"wires":[["77494fde.91d75"]]},{"id":"77494fde.91d75","type":"http request","z":"87a51622.21d23","name":"","method":"use","ret":"txt","paytoqs":"ignore","url":"","tls":"","persist":false,"proxy":"","authType":"","x":710,"y":220,"wires":[[]]}]`

AbDhops, 2021-01-06:
Awesome... Works like charm...

closed 2021-01-06 by AbDhops

AbDhops, 2021-01-07:
If it helps : 
Max duration which can be triggered by `lselect` is possible to change by calibrating "Max Duration" using VNC in "IAS_WD" cluster for the device.  This max duration is applicable for maximum blink and maximum siren... 
[image]
[image]

Pedder007, 2021-02-22:
Hi all, thanks for the valuable info here.
I've got also here the Heiman siren HS2WD-E.
All commands work fine by 'curl ...', but not the 'lselect'. It doesn't switch off after a second.
Are there any experiences on that?

AbDhops, 2021-02-22:
I guess default value for "max_duration" is 45 seconds. To switch off
before, you have to send "none" to alarm or else fine tune the value (of
max_duration)

Pedder007, 2021-02-22:
Thanks for the quick reply.
I understood above, that "max_duration" is only for the lselect, but not for select?
I already checked what currently is in there and it is a 240 (by default, as I didn't change it).
So I would assume, that by sending 'lselect', after 4 minutes the siren would switch of hereself.
With regard to select, I thought to use it e.g. when setting the sensors to 'sharp', then having a kind of short acknowledgement by a 1 sec tone ...

Pedder007, 2021-02-22:
ok, have no clue why, but suddenly 'select' works propperly, keeping the siren 'on' only for a second
Thanks again!

Obelixje, 2021-06-25:
> Hi,
> my solution was to do it in node-red with a http-call. First I define the params in a function node:
> 
> ```
> msg.method = "PUT"
> msg.url = "http://172.30.33.2:40850/api/xxxxxxxxxxxxx/lights/xx:xx:xx:xx:xx:xx:xx:xx-01/state" ;
> 
> msg.payload = '{ "alert": "blink" }'; //only blinks
> //msg.payload = '{ "alert": "select" }'; //siren for 1 sec
> //msg.payload = '{ "alert": "lselect" }'; //siren forever
> //msg.payload = '{ "alert": "none" }'; //disable alarm
> msg.headers = {};
> msg.headers['content-type'] = 'application/json';
> return msg;
> ```
> 
> ans then I use the http request node to put the request. You can get the IP of the REST API by clicking "plugins" and then "REST API Plugin" in the "IP Addresses" Section. Hope this helps to:
> 
> [image]

Hi,

I hope you are able to help me.
I am working with your node red project.
I am unable to find the API key for my stick. Can you help me determine the key?
And /xx:xx:xx:xx:xx:xx:xx:xx-01/ in your call, is this the MAC of the sirene? Or the Sirene deconz nr?

I hope you can help me with this?

og9gjz567, 2021-06-25:
Hi. Take a look at: https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/configuration/#aquireapikey

og9gjz567, 2021-06-25:
You may have to expose the docker port to your lan 😅

AbDhops, 2021-06-25:
Or use MACVLAN and have dedicated IO to your docker...