dresden-elektronik/deconz-rest-plugin #4047: Neo Coolcam Siren Alarm ( Tuya compatible)

RealKanashii, 2020-12-28:
## Device

- Product name: Neo Coolcam Siren Alarm (tuya compatible)
- Manufacturer: _TZE200_d0yu2xgi
- Model identifier: TS0601
- Device type : P
  - Sensor
  - Siren

## Screenshots

[image]
[image]
[image]

RealKanashii, 2020-12-28:
I think the device is not really well identified as the Model Identifier is the same that a Tuya Thermostat has ( found in the issues ).

Smanar, 2020-12-28:
Classic ^^
Model id is useless for tuya, need to use the manufacture name only.

machielw, 2020-12-29:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3183

Mimiix, 2020-12-30:
Closing this as duplicate.

@RealKanashii Next time, please search the git isseus if a request is already open 😄

closed 2020-12-30 by Mimiix