dresden-elektronik/deconz-rest-plugin #4049: Stelpro STZB402+ Thermostat Attribute Missing for System Mode 0X001C

jveniez, 2020-12-28:
## Describe the bug
Within deCONZ VNC I can set the System Mode to Heat/Off by reading and changing the setting to off, but from HA I can not. When I check the Rest-API, it does not show that attribute. I see the following. Even when I change the state  "on": false, it does nothing.  When I an in HA and the climate.thermostat always show off even when the thermostat is on. I can adjust the temperature and setpoints, the only issues is the heat/off. The unit is in my garage and the lowest set point is 5c and our winters can drop to -20c, I don;t want to heat the garage when I am not home.

{
  "config": {
    "heatsetpoint": 1000,
    "offset": 0,
    "on": true,
    "reachable": true,
    "schedule": {},
    "schedule_on": null
  },
  "ep": 25,
  "etag": "755629c21a70cfa38e59b1883ba97cf6",
  "lastseen": "2020-12-28T11:10Z",
  "manufacturername": "Stelpro",
  "modelid": "STZB402+",
  "name": "Thermostat",
  "state": {
    "lastupdated": "2020-12-28T11:10:39.730",
    "on": false,
    "temperature": 650
  },
  "swversion": "20170411 00141",
  "type": "ZHAThermostat",
  "uniqueid": "f8:f0:05:ff:ff:f7:81:8a-19-0201"
}

## Steps to reproduce the behavior
In YARC! I try sending a put command for the System mode and receive an error back saying, "      "address": "/sensors/system mode",
      "description": "parameter, system mode, not available", 
-->

## Expected behavior
The Thermostat should be able to be set to off mod,. also when the unit is on it show that the unit is off all the time, even when the state is changed to on:true.
-->

## Screenshots
https://user-images.githubusercontent.com/76670160/103225771-aeff7a80-48f8-11eb-8446-5ce38916d780.png
https://user-images.githubusercontent.com/76670160/103225139-4bc11880-48f7-11eb-974b-50ca4206345a.png
https://user-images.githubusercontent.com/76670160/103225173-61ced900-48f7-11eb-89ef-af10b57f5399.png

-->

## Environment
Version 2.07.01 12/8/2020
Firmware 26660700 
- Host system: (Raspberry Pi)
 - Running method: (Home Assistent deCONZ Add-on)
 - Firmware version: (2660700)
 - deCONZ version: (2.07.01)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) 
 - Is there any other USB or serial devices connected to the host system? If so: No

## deCONZ Logs
When I turn off the Thermostat the logs do show the 0x1C in the logs. 

10:31:37:875 ZCL attribute report 0xF8F005FFFFF7818A for cluster: 0x0201, ep: 0x19, frame control: 0x18, mfcode: 0x0000 
10:31:39:599 ZCL attribute report 0xF8F005FFFFF7818A for cluster: 0x0201, ep: 0x19, frame control: 0x1C, mfcode: 0x1185 

-->

## Additional context

SwoopX, 2020-12-29:
Can you please provide some screenshots from thermostat cluster of all the attributes (while those have been read, important!) until the system mode?

Additionally, can you please double check attribute 0x0008 (Pi heating demand) gives you some values while the device is turned on and heating? Lastly, attribute 0x0001 in thermostat UI configuration cluster should be the cild lock, could you please also check and confirm? Thanks!

jveniez, 2020-12-29:
Thermostat Top End
[image]

Basic Cluster
[image]

Identify Cluster
[image]

Attributes
[image]
[image]
[image]
[image]
[image]
[image]
[image]

0x0008 (Pi heating demand) Heat Running
[image]

attribute 0x0001 Child Lock (Does not look like Lock)
[image]

SwoopX, 2020-12-29:
Thanks. You got the wrong attribute for the cild lock. It's 0x0001 in **thermostat UI configuration** cluster.

I'm afraid attribute 0x0008 (Pi heating demand) is useless without any heating being on...

jveniez, 2020-12-29:
Sorry about that, 

here is the Keypad lockout
[image]
[image]

jveniez, 2020-12-29:
This is attribute 0x0008 (Pi heating demand)

This is with the heat turned off
[image]

This is with the heat turned on
[image]

Thank you so much for helping on this..

SwoopX, 2020-12-29:
Ok, so Pi heating demand is usable as well. Now, as for the child lock, you would have to set `Level 1 lockout` to enable it. Putting that one in as well as im 99% sure it'll work out.

closed 2021-01-17 by manup

MattL0, 2021-02-16:
Just a question.I med to purchase 6* ‘120v’ thermostat. Not decided yet on zigbee/zwave version of sinope or stelpro . 

what are all the info the api is returned for this thermostat? 

And what control do you have ? 

Thank you. @jveniez

jveniez, 2021-02-17:
Hi Matt,  I have control in HA as well as with Alexa, The only issue I have and it looks to be coming in the next feature request is to turn the unit off completely with the API.  Below is the API pulled from HA, the unit is in my garage so the temperature is correct at -6. You can set the Min/Max temperature or adjust the temperature manually in HA or by voice through Alexa.

hvac_modes:
  - auto
  - heat
  - 'off'
min_temp: 5
max_temp: 22
current_temperature: -6
temperature: 20
valve: 0
friendly_name: Thermostat
supported_features: 1

Hope this help @MattL0