dresden-elektronik/deconz-rest-plugin #4051: Osram plug as a switch not as a light

tonibert, 2020-12-28:
## Describe the question or issue you are having

Is it possible to add a plug (Osram) as a switch and not as a light?
When I want to turn off all the lights, I often turn off my plugs.
That is not my intention.
A switch would possibly be more suitable

## Screenshots

## Environment

 - Host system: (Synology NAS)
 - Running method: (Marthoc Docker container )
 - Firmware version: (26660700)
 - deCONZ version: (2.07.01)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes )

## deCONZ Logs

## Additional context

Smanar, 2020-12-28:
You mean in phoscon ? if yes, it's not the good github.

SwoopX, 2020-12-28:
@tonibert No, this is not possible. Create a dedicated group for your lights. Using the "all" group is not the ideal solution.

closed 2020-12-28 by SwoopX