dresden-elektronik/deconz-rest-plugin #4052: etag collisions

hbkoenig, 2020-12-28:
## Describe the bug

etag values are not as uniqe as the 128 bit hash implies!

in only 5634 REST requests/answers for a sensor  I already got 4 clashes -- how can that be ???
(which broke my (bad?) filtering for "old" data;).

here is one example, I notice that the "lastupdated" without fraction seconds are identical (06:07:02),
but even the "airqualityppb" values are different.
how is the etag calculated, why can there be so many collisions ?
---8<------8<------8<------8<------8<------8<------8<------8<------8<------8<------8<------8<------8<---
{
  "config": {
    "on": true,
    "reachable": true
  },
  "ep": 2,
  "etag": "3b070cc041839c34ff4c5b36f4491258",
  "lastseen": "2020-12-21T06:07Z",
  "manufacturername": "BOSCH",
  "modelid": "AIR",
  "name": "ZA-37",
  "state": {
    "airquality": "moderate",
    "airqualityppb": 657,
    "lastupdated": "2020-12-21T06:07:02.464"
  },
  "swversion": "20200402",
  "type": "ZHAAirQuality",
  "uniqueid": "00:12:4b:00:14:4d:eb:bd-02-fdef"
}
---8<------8<------8<------8<------8<------8<------8<------8<------8<------8<------8<------8<------8<---
{
  "config": {
    "on": true,
    "reachable": true
  },
  "ep": 2,
  "etag": "3b070cc041839c34ff4c5b36f4491258",
  "lastseen": "2020-12-24T06:07Z",
  "manufacturername": "BOSCH",
  "modelid": "AIR",
  "name": "ZA-37",
  "state": {
    "airquality": "moderate",
    "airqualityppb": 578,
    "lastupdated": "2020-12-24T06:07:02.188"
  },
  "swversion": "20200402",
  "type": "ZHAAirQuality",
  "uniqueid": "00:12:4b:00:14:4d:eb:bd-02-fdef"
}
---8<------8<------8<------8<------8<------8<------8<------8<------8<------8<------8<------8<------8<---

## Steps to reproduce the behavior

check your etags and make some statistics.
I used this command line to check sensor "321" above in those 5634 REST answers for a specific short time interval:

$ cat A-1??????-* | jq '.sensors."321".etag'  | uniq | sort | uniq -c| sort -n
...
      2 "3b070cc041839c34ff4c5b36f4491258"
      2 "551b92f389913ead7be7469bc2adf450"
      2 "750899e035aba6513914cef6bf0f935c"
      2 "ded5763e8e7457cef225d429b40eb068"

## Expected behavior

no collisions [at least for a *LONG* time]

## Screenshots

## Environment

2.08.00 (2.08.00-raspbian-stretch-beta)

 - Host system: Raspberry Pi 4
 - Running method: Raspbian
 - Firmware version: (26680700)
 - deCONZ version: (2.08.00)
 - Device: ConBee II
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

SwoopX, 2021-01-11:
Apparently, time resolution taken for the etag is just in seconds, as no format has been specified.

From QT documentation:: `If format is empty, the default format "hh:mm:ss" is used.`

Done a commit to increase it.

hbkoenig, 2021-01-11:
> Apparently, time resolution taken for the etag is just in seconds, as no format has been specified.
> From QT documentation:: `If format is empty, the default format "hh:mm:ss" is used.

still etag with identical time from different dates will collide. this only makes it 1000 times more unlikely (10 bits) but does not solve the issue. it still will give collisions, now 1 of 86.4M instead of 1 of 86.4k.

date string *must* be included.

and why not include measurement values and maybe?
even with identical date/time, "airqualityppb": 657 and "airqualityppb": 578 should result in different etags, shouln't it?

closed 2021-01-17 by manup