dresden-elektronik/deconz-rest-plugin #4056: Daylight Sensor not reachable under Win10 Virtual Machine

pillemats, 2020-12-29:
Hey friends

i have big problems with my daylight sensor. it is not reachable and i need it :-)
I tried to set up the location and timezone in phoscon and hue essential on my iphone. both dont work.
i also tried older deconz versions but the same problem.
I dont have any knowledge about restAPI, so i cant create a new on for my own

I hope anybody can help to get the solution

## Screenshots
[image]

## Environment
2.08.00 / 8.12.2020
Firmware
Windows 10 Virtual machine on ESXI
 - Firmware version: (26680700)
 - deCONZ version: (8.12.2020)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes )

## deCONZ Logs
i dont know how to get

Smanar, 2020-12-29:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3908

>I tried to set up the location and timezone in phoscon and hue essential on my iphone. both dont work.

No error message like previous issue ?

Do you have the device JSON ? Do you have a connection to internet on this host ?

pillemats, 2020-12-29:
No errors 

what is device JSON?

The Host has Internet Connection

SwoopX, 2020-12-30:
Closing this one. See referenced issue for resolution.

closed 2020-12-30 by SwoopX