dresden-elektronik/deconz-rest-plugin #4057: Cannot change State by Jeedom

Mafiewww, 2020-12-29:
Hello everyone

## Describe the bug
Recently, all of my connected switches (Xiaomi Aqara) are blocked and no further change of state is possible.
Here is an error log : 
Error during request : 192.168.1.27:8484/api/931B85D49B/lights/2/state(PUT), data : {"on":true} error : 3 => resource, /lights/2, not available

## Steps to reproduce the behavior

1. Setup of Jeedom and last version of Deconz
2. Add aqara buttons by Phoscon
3. Try to change the states of the buttons 

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi 
 - Running method: Jeedom
 - Firmware version: 26680700
 - deCONZ version: 2.08.00 / 08/12/2020
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: USB Drive

## deCONZ Logs

  
## Additional context

Smanar, 2020-12-30:
Hello, they can have leave the network ?

Can you see/use them in phoscon ?
It s wired one, can you try to re-include one ?

srtxema, 2020-12-31:
i have the same problem. I received performing sync to create commands if needed

Smanar, 2020-12-31:
You have loose all your sensor in same time, and only this sensor type ?

srtxema, 2020-12-31:
I havent lost. there are in the deconz but the dont update the state.all of them

Smanar, 2020-12-31:
it s wired device or battery one ?

srtxema, 2020-12-31:
Battery. Door sensors and movement sensors

srtxema, 2020-12-31:
a friend said me one day that if something doesnt work with domotic, i wait for 24 hours thas probably it works again. It has occur now. without do nothing, everything is works again

Mafiewww, 2020-12-31:
> Hello, they can have leave the network ? But all in same time ..., you have a power off during a complete day ?
> 
> Can you see/use them in phoscon ?
> It s wired one, can you try to re-include one ?

Hi,

Yes,  the state.all don't change.
No i don't have power off during a complete day ^^
I see him on phoscom.
I can re include but after one night , they don't respond

Smanar, 2021-01-01:
OK so there is too much issue in one

>Error during request : 192.168.1.27:8484/api/931B85D49B/lights/2/state(PUT), data : {"on":true} error : 3 => resource, /lights/2, not available

This message is for wired device, you can't have it for battery one, it s a command for light not sensor.

Then you can check if they work in phoscon, if they work in phoscon and not third app, it s probably third app fault.

I don't know what is the "state.all" but if you have the GUI, take a look if the device is still here (else it will be grayed), you can just have connexion issue.

And I don't see why a sensor can work again after 24 h ....

Mafiewww, 2021-01-27:
Sorry for the delay.
my equipments are wired connected interpreters

Sometimes, they are gray on phoscom app and they are here after some times.

Smanar, 2021-01-27:
Connexions issues ?
I haven't see this bug yet for wired Xiaomi switch. But all in same time ...
And they are back themself ?

Mafiewww, 2021-01-28:
Maybe 
Not always. 
Today, my wired switch on my bedroom didn't work.
But yesterday, its working...
Erreur exécution de la commande [Chambre][Interrupteur Murale Chambre][On 02] : Erreur lors de la requete : 192.168.1.27:8484/api/931B85D49B/lights/19/state(PUT), data : {"on":true} erreur : 3 => resource, /lights/19, not available

Smanar, 2021-01-28:
Yep the error message is normal.
But another try we can do is checking if the error is from the api that have the device unreacheable, or if the device is realy unreachable.

When the api decide a device is unreacheable it memorise this state, and wait for something from device to decide if it is still alive, so sending request to the device will be blocked by the api, but not using group command (from my memory at least)

So something you can do for test, is putting all your switchs in a fake group (using phoscon) and the next time, you have an unreacheable switch, use this group.

And BTW what are you doing for your swiches be back ? just waiting ? I think using them manualy, make them working again no ?

Mafiewww, 2021-01-31:
I reset my conbee 2 and this week i will try with Deconz on Windows

I had tried to use phoscom to change state but they was no change...

Smanar, 2021-01-31:
To make a try in another machine, you need too migrate the database using the backup/restore feature from phosocon.

You have tested the "group" trick" ?

>what are you doing for your swiches be back ? just waiting ?

SwoopX, 2021-02-01:
> I had tried to use phoscom to change state but they was no change...

That could depend on how you tried to change it. The right way is not exactly intuitive.

closed 2021-03-05 by github-actions[bot]