dresden-elektronik/deconz-rest-plugin #4058: Can't update Conbee 2 trough Phoscon app

rutkoe, 2020-12-29:
## Conbee 2 Gateway display's up-to-date but it isn't. Also not showing new updates (it did previously)

I'm trying to update my Conbee 2 gateway trough the Phoscon app but is is showing me up-to-date all te time. I added a screenshot.

Could someone give me advice or tell me what i need to check to provide the latest stable update for my Conbee 2 stick?
Everything is running fine also in HomeBridge i experience no problems.

If any information is needed i will provide it when it is necessary.

Thanks in advance!

## Screenshots

## Environment

 - Host system: (Raspberry Pi 3B)
 - Running method: Raspbian GNU/Linux Buster (10)
 - Firmware version: 26580700
 - deCONZ version: HOW CAN I CHECK THIS?
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No other USB devices

## Additional context

rutkoe, 2020-12-29:
I just discovered through systemctl that "deconz-init.service     loaded failed     failed       deCONZ: ZigBee gateway -- Initialisation". Could this be the cause of the problem?

SwoopX, 2020-12-29:
[image]

rutkoe, 2020-12-30:
[image]

Mimiix, 2020-12-30:
You are on a failry old version of deCONZ. However, you should update the firmware manually.

Smanar, 2020-12-30:
Or if you want to update it using phoscon, just update deconz, it will propose you new firmware (if they are)

rutkoe, 2021-01-03:
I managed to update deconz manually, however, I'm still not sure if it is on the latest version.
[image]
for example: Version is still on 2.05.78 (it shows me a more recent date)
Firmware seems to be ok to me.

Any suggestions?

Mimiix, 2021-01-03:
@rutkoe read into how to update Linux packages.

Smanar, 2021-01-03:
If you have used package installation

```
sudo apt update
sudo apt install deconz
```

else you have it on the Github Homepage.

rutkoe, 2021-01-03:
Thanks a lot everyone for their responses and suggestions! I will close this one :)

closed 2021-01-03 by rutkoe