dresden-elektronik/deconz-rest-plugin #4060: Aqara Motion Sensor Long Inactive Time

marithpl, 2020-12-30:
## Describe the question or issue you are having
I'm a new user of Conbee II, I connected a few sensors mostly from Aqara/Xiaomi and Hue Motion Sensor.
Everything works fine except Aqara Motion Sensor. By default it show motion detected status for 90s.
It's quite long vs. 15s. for Hue Motion. I found an API request to change 'duration' parameter for this sensor. I changed fo 15 and only effect which I observed is that Sensor change status from detected to none after 15s. but is not possible to activate next motion for next 75s.

Is it possible to change any parameter to force sensor to detect motion after 15s?

My measures:
Philips Hue: Detected motion status is for about 15s. after that it's possible to detect move
Aqara Motion (duration 90): Detected motion status is for about 90s. after that it's possible to detect move
Aquara Motion (duration 15): Detected motion status is for about 15s. after that time it's not possible to detect move I suppose I have to wait about 90s. to detect next movement

## Screenshots
[image]
[image]

## Environment

 - Host system: NAS Synology
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.09.00 / 23.12.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: -----

## deCONZ Logs
`{
-"config": {
"battery": 100,
"duration": 90,
"on": true,
"reachable": true,
"temperature": 2900
},
"ep": 1,
"etag": "*****************",
"lastseen": "2020-12-30T10:23Z",
"manufacturername": "LUMI",
"modelid": "lumi.sensor_motion.aq2",
"name": "Motion Sensor",
-"state": {
"lastupdated": "2020-12-30T10:23:26.022",
"presence": false
},
"swversion": "20170627",
"type": "ZHAPresence",
"uniqueid": "*****************"
}`

## Additional context

duffbeer2000, 2020-12-30:
That's the normal behavior for the aqara motion sensor. If you want faster motion detection you need to hack the hardware:
https://www.bastelbunker.de/aqara-motion-sensor-hack-mod-und-fhem-bzw-home-assistant/
https://community.smartthings.com/t/making-xiaomi-motion-sensor-a-super-motion-sensor/139806

Mimiix, 2020-12-30:
@duffbeer2000's answer couldn't be better :)

duffbeer2000, 2020-12-30:
@Mimiix but the second question needs to be solved: "The second issue is it's not showing a brightness in lux."

marithpl, 2020-12-30:
Oh my god...
I thought it was going to be a software change, not a soldering.
Is it really enough to solder these two elements?

So what does changing the parameter 'duration' do? Only showing a occupancy status?

marithpl, 2020-12-30:
> @Mimiix but the second question needs to be solved: "The second issue is it's not showing a brightness in lux."

I don't know what happened, but after I wrote this post it's already change lux value :)

SwoopX, 2020-12-30:
Those "cool off" times are device specific. For some, it's possible to configure that to a certain extend, for others like the Xiaomi, there's nothing you can do about it.

closed 2020-12-30 by SwoopX

marithpl, 2020-12-30:
Just for other people, making line between this to points works. Even with normal pencil :)

Made just API change via:
`curl -X PUT -d '{"duration":5}' -s http://ipaddress:port/api/{APIKEY}/sensors/{SENSOR_NUMBER}/config`

[image]

SwoopX, 2020-12-30:
Nice and slick 😂

Smanar, 2020-12-30:
But on my side I don't see why so much people need 15 s coldown ?
Just increase the delay to 5 mn, and turn off the light when the presence go to false.