dresden-elektronik/deconz-rest-plugin #4061: Device name

arodriguezf, 2020-12-30:
## Device

- Product name: Curtain switch Tuya Smart Life ZigBee 3.0 SC400
- Manufacturer: LoraTap
- Model identifier: 
- Device type : 
  - Switch

## Screenshots

[image]
[image]
[image]

arodriguezf, 2020-12-30:
Sorry, now it seems to work, the only pending thing is the calibration of the curtain switch

Mimiix, 2020-12-30:
@Smanar this feels like a duplicate. Can you confirm?

Mimiix, 2020-12-30:
Closing and Continue in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3134#issuecomment-732957483

@arodriguezf please use search next time.

closed 2020-12-30 by Mimiix