dresden-elektronik/deconz-rest-plugin #4067: Group of lights without hue

Kristian8606, 2020-12-31:
## Describe the question or issue you are having

When I display all Deconz groups for the Homebridge-Hue plugin, all groups are displayed with the HUE characteristic, ie lights that have only a color temperature are also displayed with a color wheel.
Еeven lights that are only dimmable are displayed as colored, how can I fix this problem?
  Or an adjustment is needed on your part.

## Screenshots
No

## Environment

 - Host system: (Raspberry Pi )
 - Running method: (Raspbian)
 - Firmware version: (26680700)
 - deCONZ version: (2.9.0)
 - Device:  ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
No

## Additional context

ebaauw, 2020-12-31:
This is a well-known, long outstanding bug.  See #415, #342.  Don't know why the bloody bot closed these.

A `/groups` resource should only expose `action` attributes for which at least one member `/lights` resource has a corresponding `state` attribute.  So a group of _Dimmable Light_ devices should not expose `xy`, `ct`, `hue`, `sat`, nor `effect`.  The Hue bridge API works like this.

This should also work for `open`, `lift`, and `tilt` for Window Covering devices, so the group can send _Open_, _Close_, _Go to Lift Percentage_, and _Go to Tilt Percentage_ (but see also #3881, they should be sent to all devices, not just to _Receiver on When Idle_ devices).  For Warning Devices, we probably need a different attribute for `alert`, so the group can know when to send _Identify_ vs _Start Warning_.

I don't know what the state on door locks is, but if they're exposed as `/lights` (which would be valid if and only if they support _Groups_), they would need a different `state` attribute (`lock`?) so the group can send the _Lock Door_ and _Unlock Door_ commands.

ebaauw, 2021-03-20:
This is still an issue