dresden-elektronik/deconz-rest-plugin #4068: IKEA Tradfri Shortcut Button E1812

Zehir, 2020-12-31:
## Device

- Product name: Tradfri Shortcut Button E1812
- Manufacturer: IKEA of Sweden
- Model identifier: TRADFRI SHORTCUT Button
- Device type : 
  - Switch

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2020-12-31:
Closed. Duplicate of #3373

closed 2020-12-31 by Mimiix

Zehir, 2020-12-31:
Duplicate of #3495