dresden-elektronik/deconz-rest-plugin #4071: [Device support] Does deCONZ supports MoesHouse / Livolo Roller Blinds?

VivantSenior, 2021-01-01:
## Describe the question or issue you are having

Hi,

according to [this page](https://zigbee.blakadder.com/Zemismart_M515EGB.html) deCONZ supports Zemismart Chain Roller Shades Driver (Model M515EGB) but looking on the AliExpress I've found chipper versions:
- [MoesHouse](https://www.aliexpress.com/item/1005001698400295.html),
- [Livolo](https://www.aliexpress.com/item/1005001702490864.html).

By the looks of it, it looks the same as AM43 Blind motor with only Bluetooth. Does anyone knows if these devices are supported by deCONZ (with Conbee II)?

## Screenshots

[image]

## Environment

 - Host system: (Raspberry Pi 3B+)
 - Running method: (Home Assistant deCONZ Add-on 6.6.2)
 - Firmware version: (26580700) (I had to downgrade becase of [this bug](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3385#issuecomment-714294128) and [this](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3387#issuecomment-751625190))
 - deCONZ version: (2.07.01 / 8.12.2020)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? If so: external SSD via USB 3.0.

## deCONZ Logs

N/A

## Additional context

N/A

VivantSenior, 2021-01-03:
I was unable to find it in [Supported Devices](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices) so if anyone else is interested I will buy one and create ticket with device support request.

Smanar, 2021-01-03:
You can, I have already integrated some moe/tuya device, and covering are not the more complicated.

Smanar, 2021-01-05:
For information a similar device https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3348

eric-zimmer, 2021-01-05:
Hey, I'm also interessted in this device from Moes House. According to z2m the chip should also be from tuya TS0601_curtain, so the same as Zemismart Chain Roller Shades Driver M515EGB. So I think it should work.
I'm not sure to buy this one (zigbee is included i think) or the Zemismart one (extra zigbee converter needs a extra USB plug).

eric-zimmer, 2021-01-15:
Hey, I've bought the Moes House motor and it works manually but I am not able to connect it nor to the tuya app via Tuya gateway neither to conbee. I don't know how to set the device into pairing mode.

Smanar, 2021-01-15:
No informations on manual ? All in chinese ?

eric-zimmer, 2021-01-15:
Hey, no there was no information about this in the manual but at the z2m github someone could help me. 
So it shows in deconz but there is no green line to conbee. It is not in phsocon like the zemismart as a light.
Could you add the device? 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Do you need some more information?

eric-zimmer, 2021-01-15:
In phoscon I think it is recognized as a light sensor. This is not absolutley wrong. The motor comes with a solar panle which can be connected to it to load the battery and also to use it as a light sensor.

Smanar, 2021-01-15:
Can try.

Do you know how to compile the code to test it ?

```
sudo apt install deconz-dev
git clone --branch moehouse_rollerblind https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

eric-zimmer, 2021-01-15:
I realy don't know how to compile but I have a respberry from deconz, so I think it should be possible on this. I think i have time on sunday to try it but i don't know if it will work Out. Thank you for the fast support

Smanar, 2021-01-15:
Yep it s work fine on raspberry https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

You just need a real OS with Linux, no docker for exemple.

eric-zimmer, 2021-01-17:
Hi, i did the code you wrote and started deconz new.
I tried to connect the motor for a second time but nothing happend. So I deleted it and added it again. I see no difference. How am I amble to see it the new code is integrated? What should be different? How can I see if the device could be controlled?

Smanar, 2021-01-17:
For tuya, you need to make 2 inclusions.
The device can be visible on deconz in "light", but it will be visible in third app.
Else you can take a look in deconz/help/log with "info" and "info_l2".

eric-zimmer, 2021-01-17:
Thank you. How can I make a second inclusion? Do you mean add the deivce for a second time? I tried this several times but there is only one device visible in the map.

Smanar, 2021-01-17:
Yep, but why you are talking about the map ?
You need to use phoscon not deconz, to enable the API.

eric-zimmer, 2021-01-17:
HI, I don't know what I really did, but now in deconz the motor is named Smart Plug 16 and in Phsocon the motor is a light but nothing happens if I switch the light on or off in phoscon.
[image]

Smanar, 2021-01-17:
Try to make 2 inclusions using phoscon.
You don't use third app ? to check for device JSON ?

eric-zimmer, 2021-01-17:
I use hue essentials on my smartphone.

Smanar, 2021-01-18:
Ok so will be harder without third app ^^.

For IP and PORT you can use exactly the same you are using for phosocn.

First you need an API key:
`curl http://IP:PORT/api/ -X POST -d '{"devicetype": "TestKey"}'`

If you have an error message about autorisation or autentification, go in phoson/gateway/advanced/authenticate app and remake the command

Then to take a look on your light device
`http://IP:PORT/api/LEY/lights`

>When trying to add the motor a second time in phoscon via searching lights or switches notghing happens in phosocn or deconz.

This is strange, you need to have the same result at every try.

eric-zimmer, 2021-01-18:
Hi, thank you for support.

I got this for the motor:
```
{
    "etag": "e62b9e828a97ba080726cbdf1b3db335",
    "hascolor": false,
    "lastannounced": "2021-01-17T19:51:38Z",
    "lastseen": "2021-01-17T19:50Z",
    "manufacturername": "_TZE200_zah67ekd",
    "modelid": "TS0601",
    "name": "Smart plug 16",
    "state": {
        "alert": "none",
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:cb:ff:5b-01"
}
```

When sending open true or false the motor moves and also a percentage in lift works. It also works with Alexa. Unfortunately open and close are switched but this is because I needed to switch the directions of the buttons on the device itself I think.

Also I could add the device to alexa and it is possible to turn on, off and set a percentage for lift.
But everytime Alexa responds "Rollo scheint nicht richtig zu funktionieren" (Blind seems not to work correctly), but it does what was commanded. 

In Hue Essentials I cannont access the motor (shown as a light), I always got Client request invalid: 400 Bad request.

So I think it works somehow but not completly correct, but sufficient at the moment.

Smanar, 2021-01-18:
Yep, I don't see this model on the code, so you a re using the native working mode.

>Unfortunately open and close are switched but this is because I needed to switch the directions of the buttons on the device itself I think

It s realy hard to find the good direction, can be wiring, there is too a zigbee command to reverse it, so so much combinaisons possibles, it s for that I realy don't like force a direction in the code. The only way is making tries. You need to have at final

> **shutter closed**
> on= true
> bri = 255
> open = false
> lift = 100%

eric-zimmer, 2021-01-18:
Hello, so I am not using your code you made but the native deconz/phoscon code?

I made everything as you said (the code for compiling) but I am not able to make a second inculsion in phoscon or deconz, so only one device visible in deconz or phoscon and not two.
How to make this? I tired to pair a light or switch and set the motor in pairing mode? Tried this several times but it does not work, only one device.
At the moment the motor is in phoscon called smart plug but seems to be a light, that is why I think Alexa can work for it, light brightness in percent is the lift of the blind. Light on is blind open, light out is blind closed.

I have tried the reverse code from the Zemismart thread, if I use this, the dircetions of the buttons on the device are also reverted.

Smanar, 2021-01-18:
>Hello, so I am not using your code you made but the native deconz/phoscon code?

Oups, sorry ! have forget to submit the PR, so I don't find it ! Have just make it.
But your device is working ? The double inclusion is for device that not working.
If you have that in the json
```
    "state": {
        "alert": "none",
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
```

It mean it can work.

And IDK how to use it in phoscon, as you are forced to add it in a group. Try with using the pointer on the device alone, not the group. If I remember I have already see this problem.

eric-zimmer, 2021-01-18:
Yes this is in json and so the device works via alexa and API commands.
I think about buying a remote Switch maybe in future but alexa works so Not so immediately. 

So I need to compile your code again and restart deconz and add the device again?

I tried commands in phoscon in group and on device  itself but not everything worked. 

When I only See one decive how Do I know second inclusion worked?

Smanar, 2021-01-18:
>So I need to compile your code again and restart deconz and add the device again? Sorry for dummie question, what is PR?

But I don't understand what you are trying to do ? The device is working (at least direclty with API) ?

The PR is a code submission for it will be in final version (for all other users) but it s exactly the code you have tested, and validated ^^. If you still have problem with it better to tell it now, before an other devs validate the PR too.

>When I only See one decive how Do I know second inclusion worked?

On the JSON 
```
{
    "etag": "e62b9e828a97ba080726cbdf1b3db335",
    "hascolor": false,
    "lastannounced": "2021-01-17T19:51:38Z",
    "lastseen": "2021-01-17T19:50Z",
    "manufacturername": "_TZE200_zah67ekd",
    "modelid": "TS0601",
    "name": "Smart plug 16",
    "state": {
        "alert": "none",
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:cb:ff:5b-01"
}
```

Type = Window covering device and you have the state/lift state/open, not present on bad inclusion.
Or just if the device is working

On phoscon try here
[image]

eric-zimmer, 2021-01-18:
For me the motor is working at the moment, only problem is hue essentials can not access device but API and Alexa working.
I understand something was wrong in the code but that was a mistake of me. 
Thanks for your support. 
One last question: Should I wait with update of deconz till the support is in the beta. When I update to the latest beta your code will be lost?

Smanar, 2021-01-22:
Good question, it depend of version stable or beta, but this issue will be automatically closed when the PR will be in official code. And so disponible somes days afters.

closed 2021-02-15 by manup

aidbish, 2021-04-23:
i have the same controller witht ehsame manufacturer code, yet i can only see it in deconz, but not phoscon, i have tried pairing multiple times, (without deleting) and nothing shows up in phoscon, no light or anything. is there a trick to getting this setup

gio82c, 2021-06-09:
Hi, i try to include the device, but when i add i see on phoscon from "add sensor" = Ready, on deconz i see, but after add no sensor, light or switch on phoscon.

I also try to include 2 times.

On REST API i see only in sensors this:

  "32": {
    "config": {
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "94f374622926801a41d2b5a51fab9764",
    "lastseen": "2021-06-09T12:12Z",
    "manufacturername": "_TZE200_zah67ekd",
    "modelid": "TS0601",
    "name": "Battery 32",
    "state": {
      "battery": null,
      "lastupdated": "none"
    },
    "type": "ZHABattery",
    "uniqueid": "b4:e3:f9:ff:fe:07:d1:74-01-ef00"
  }
}

[image]

Smanar, 2021-06-09:
The device can be invisible on phoscon (it don't support covering).

On logic way, now if you try to include the device again "add new light", but without deleting the battery entry, it need to be created, but you probably need third app or take a look direclty in the api to be sure

But some users have succed with deleting the battery sensor (deleting the complete node on deconz) and remake inclusion https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4806#issuecomment-848910502

Smanar, 2023-01-11:
Yes, if all is fine, the battery sensor can help to include the device, so don't delete it, and retry to include the device.
Deconz will use the manufacture name of the device in the sensor list (the battery one) to reconize the device.

If it realy don't work delete the battery sensor, I know some users that have succed with deleting it, but the code is made to work with the sensor entry.

Smanar, 2023-08-09:
You see nothing in deconz/phoscon ?
No zigbee connexion at all, or it's just not visible in API ?

jscheurwater, 2023-08-09:
I don't see any movement in Deconz, so nothing is shown. My other device however is still connected.

Smanar, 2023-08-10:
No device at all ? even bad reconised ?
Not good, the pairing procedure is bad or there is something bad in zigbee configuration (or connection issue), can take a look in logs.