dresden-elektronik/deconz-rest-plugin #4073: Tradfri Driver unreachable after connecting Symfonisk Dimmer directly

pillemats, 2021-01-01:
## Describe the bug
Hey together,

i have a problem with the symfonisk driver:

i connect my 3 Tradfri driver with phoscon and it works great.
then i connect my symfonisk dimmer directly to the 3 lamps (I dont connect the dimmer over phoson)
after its done the tradfri driver are not reachable in phoscon.

i must delete it and reconnect them to get it work. - if i connect then the symfonisk dimmer after few minutes the same problem

## Steps to reproduce the behavior
i tried it many times - after connecting the dimmer to tradfri driver the triver will be unreachable

## Screenshots

[image]

[image]

## Environment

[image]

 - Host system: (PC / NAS)
 - Running method: (Windows / Virtual Machine)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes )

Mimiix, 2021-01-01:
Hi,

You can't do it that way as then they are connected to each other. I suggest you create the group in the old web ui. 

This is intended behavior, not a bug report. I'll change label.

pillemats, 2021-01-01:
After few Versions ist works Perfect!

what should i do now

pillemats, 2021-01-02:
i tested the same with the tradfri on off switch and this works great

can anybody help ?

pillemats, 2021-01-05:
any idea ?

SwoopX, 2021-01-05:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4073#issuecomment-753345285

pillemats, 2021-01-06:
in the old webif the switch will not shown.

pillemats, 2021-01-07:
I tried 4 different dimmer 2 black und 2 white versions.

In phoscon the 2 white shown a dfferent firmware version 2.1.024 and 022
022 works perfect without unreachable lamp

The blackversion shown no firmware versions - but why ?

closed 2021-03-05 by github-actions[bot]