dresden-elektronik/deconz-rest-plugin #4075: IKEA smart blinds disconnected after gateway restart

erd82, 2021-01-01:
## Describe the bug

After restarting deCONZ my IKEA smart blinds (IKEA FYRTUR) do not react anymore.

Neither the control from the Home Assistant nor the control with remote control for opening / closing work.
I connected the device to ConBee stick on the same way as shown here: https://www.youtube.com/watch?v=PVDNYLaAVIA
It perfectly worked for months to control the blinds from Home Assistant. 

After one of the recent updates they don't react after restart.

This happens for all blinds at the very beginning. If you are lucky, after several minutes the blind can be controlled from both HA + remote again. But some blinds never come back again. Then you have to restart deCONZ or pair the blind again with deCONZ. So it's sporadic.

## Steps to reproduce the behavior

Restart complete host (Raspberry Pi 4 Model B) including Home Assistant + deCONZ add-on OR only restarting deCONZ add-on leads to the described issue.

## Expected behavior

You should be able to control the blinds right after restarting deCONZ.

## Screenshots

interestingly, the blinds appear as if everything is ok.
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: 2.07.01 / 12/8/2020
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

Mimiix, 2021-01-04:
Are the blinds controllable from the deCONZ interface?

In deconz, open panels> Cluster  Info. 
On the map, select the Fyrtur > clikc the right dot. Then click the "window covering". 

then , click "up/open" or "down /close"

Do they move then?

SwoopX, 2021-01-05:
On a side not, you neither run the latest deconz version nor the latest firmware.

erd82, 2021-01-05:
> Are the blinds controllable from the deCONZ interface?
> 
> In deconz, open panels> Cluster Info.
> On the map, select the Fyrtur > clikc the right dot. Then click the "window covering".
> 
> then , click "up/open" or "down /close"
> 
> Do they move then?

@Mimiix: I'm a little embarrassed. Last week a constantly struggled with this issue and could not get it running without issues. But after your post I restarted several times and all three blinds seem to run without any problems. Also thanks so much for your short instruction above! I case the issue comes up again, I will try this out and post it again. Shall we close this issue from your point of view? ... or leave it open?

> On a side note, you neither run the latest deconz version nor the latest firmware.

@SwoopX : I installed the latest updates that I received from Home Assistant. A newer version is not yet provided. Do you think I should install a newer one? And how is it recommended to do when running deCONZ as addon in Home Assistant?

PS: Thanks for your quick replies and for your great job !!

Mimiix, 2021-01-05:
> Shall we close this issue from your point of view? ... or leave it open?

We can close it :) If you experience it again, just open a new one :)

erd82, 2021-01-05:
All right. Thanks. I will close this one. Cheers

trevorbenyack, 2021-01-06:
Just to add, I’m having the same problem, using the deCONZ add-on in Home Assistant. Got everything set up today, “working” (no control through phoscon, but could control through HA), moved the Rasberry Pi and then the blinds stopped working. They’re showing as on and connected through phoscon and deCONZ. Tried using the up/open and down/close exec buttons and they didn’t work. — I’m just going to go back to the Tradfri hub, but I wanted to add that this person’s experience is not isolated.

Mimiix, 2021-01-06:
@trevorbenyack I suggest opening a seperate issue.  Make sure to add logs as thsi seems to be related to interference issues.

erd82, 2021-01-06:
I guess the reason why it's working for me now is that I have installed a ZigBee repeater closer to the IKEA blinds. My assumption is that there was a signal problem before which is now solved. 

[image]

erd82, 2021-01-23:
@Mimiix, I am having the same issue with one of my blinds these days. I am not able connect to the blind neither in HA nor using deCONZ interface (deconz > Cluster Info > right dot > "window covering" > "up/open" / "down /close"). For the other blinds this works.

I actually thought after several days it will reconnect (after restarting host or HA). But up to now it does not react. The battery of the blind must also be full, since I charged it beginning of this month. Any idea?

erd82, 2021-02-06:
Hello, unfortunately this issue could not be solved after several weeks now - even deCONZ updates did not help. Do you think that there is a general issue with ConBee 2 / deCONZ ? Or could there be a general configuration issue for deCONZ on my side?

I am also having issues with other devices (aqara motion sensor, [aqara temperature humidity sensor](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4081), tradfri bulb E14 470lm, etc.). In most cases re-pairing the devices with deCONZ helps as a workaround. But it's not good if you cannot rely on device connectivity . I am tending to look for alternatives to ConBee 2.

trevorbenyack, 2021-02-06:
Yeah, I had to take them off deCONZ and put them back on the Ikea gateway. I tried pairing the repeater with deCONZ and moving it close to the blinds but it was still dropping commands every day. The remotes were laggy as well. And when I used the HomeKit integration on HA, I couldn't group the two blinds together in the home app which was super annoying. They've been great since going back to the gateway. Sad I have to keep it hooked up, but it's a decent trade off for the convenience and the low price.

closed 2021-03-09 by github-actions[bot]

erd82, 2021-04-13:
> put them back on the Ikea gateway. I tried pairing the repeater with deCONZ and moving it close to the blinds but it was still dropping commands every day. The remotes were laggy as well. And when I [...]

Hi trevorbenyack, thanks for this recommendation. In the meantime I also bought an IKEA gateway (instead of using ConBee II
stick) and connected all IKEA devices on this way to home assistant which perfectly worked! I tested this setup for several weeks now and the IKEA blinds stay reliably connected. No issues anymore since I am using the IKEA gateway.