dresden-elektronik/deconz-rest-plugin #4077: Attempting to updating firmware yields, "2184: Error: uart reset failed, check retry"

Ruthalas, 2021-01-02:
## Describe the bug
I am trying to flash the firmware on my ConBee.
When attempting to flash the ConBee I receive this error:
`2184: Error: uart reset failed, check retry`

## Steps to reproduce the behavior
1. Downloaded and installed the current D2XX drivers
2. Downloaded and installed the current deCONZ software (deCONZ_Setup_Win32_V2_09_00)
3. Downloaded the most recent firmware and confirmed it via the MD5 (deCONZ_Rpi_0x26390500.bin.GCF)
4. Insert the ConBee, wait a few seconds, check COM port (`GCFFlasher -l`)
4. Run the following command to flash:
`GCFFlasher -d COM9 -f deCONZ_Rpi_0x26390500.bin.GCF`
5. I get the following error:
`2184: Error: uart reset failed, check retry`

## Expected behavior
Successful Flash

## Screenshots
[image]

## Environment

 - Host system: Windows (currently)
 - Running method: Windows (currently)
 - Firmware version: 260B0500
 - deCONZ version: 2.09.00
 - Device: ConBee I
 - Do you use an USB extension cable: Tried both with and without
 - Is there any other USB or serial devices connected to the host system? No, I don't think

## deCONZ Logs
N/A

## Additional context
N/A

Smanar, 2021-01-02:
Hello,

Try `systemctl stop ModemManager` before starting the Update.

Smanar, 2021-01-02:
Hello, wich one windows versions ?

Mimiix, 2021-01-02:
Is deconz stopped?
Are you using a vm?

Ruthalas, 2021-01-02:
This is on current Windows 10, not a VM.
I have tried it with DeConz running (it says it doesn't have access) and with it off (it throws the above error).

SwoopX, 2021-01-02:
Deconz must NOT run while flashing.

Ruthalas, 2021-01-02:
I understand.
I have tried it with DeConz closed, as described in the wiki.

Mimiix, 2021-01-04:
@Ruthalas I am wondering. You are using the filename directly. But the guide says /firmware/_filename_ as location. 

So instead of 
`GCFFlasher -d COM9 -f deCONZ_Rpi_0x26390500.bin.GCF` 

use

`GCFFlasher -d COM9 -f ../firmware/deCONZ_Rpi_0x26390500.bin.GCF`

Also, is CMD running as administrator?

Ruthalas, 2021-01-04:
I have tried with an elevated command prompt, it gives that same error.

I just used the full path to the file and got the same error:
[image]

Thank you for the suggestions.

SwoopX, 2021-01-05:
You could try by removing the stick, execute the command and re-insert it after like 10 secs.

Ruthalas, 2021-01-06:
The command will not run with the stick is removed.
I tried a couple combinations of unplugging the stick while it is trying to perform the reboot process and got the same error:
[image]
You can see where I did so marked above.

The final result is the same error.

Smanar, 2021-01-06:
Nothing to see with your issue, but there is another strange issue here, that was solved using an old updater, if you can make a try by curiosity https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4104#issuecomment-754878591 ?

sunseaandpalms, 2021-01-19:
I have the same issue. My environment:
* Windows 7 Pro. SP 1
* deConz v 2.09.01 (2021-01-18)
* ConBee I (0x260b0500 2016-09-28) 
* Tried flashing with `deCONZ_Rpi_0x26390500.bin.GCF` first (2020-11-07) - *FAILED* with `2184: Error: uart reset failed, check retry`
* Tried older version `deCONZ_Rpi_0x261c0500.bin.GCF` (2018-02-04) - *FAILED* with the same error
* I do remember to plug out -> Wait -> plug in -> Wait and update the COM port

Mimiix, 2021-01-19:
@sunseaandpalms Are you using a VM?

sunseaandpalms, 2021-01-19:
That is native windows installation

Mimiix, 2021-01-19:
@sunseaandpalms Do you have a extensioncable in place?

sunseaandpalms, 2021-01-19:
@Mimiix no extension cable in place. I plug the ConBee I into my laptop

Mimiix, 2021-01-19:
Added a issue on our internal issuetracker. Asking Manup to reply.

sunseaandpalms, 2021-01-19:
* Downgraded deConz to `deCONZ_Setup_Win32_V2_05_04.exe` (2018-02-04) and was able to install `deCONZ_Setup_Win32_V2_05_04.exe` (2018-02-04) -> `SUCCESS`
* Tried `deCONZ_Setup_Win32_V2_09_01.exe` to update from `deCONZ_Setup_Win32_V2_05_04.exe` to `deCONZ_Rpi_0x26390500.bin.GCF` -> `FAILED` (*2184: Error: uart reset failed, check retry*)
* Tried `deCONZ_Setup_Win32_V2_09_00.exe` to update from `deCONZ_Setup_Win32_V2_05_04.exe` to `deCONZ_Rpi_0x26390500.bin.GCF` -> `FAILED` (*2184: Error: uart reset failed, check retry*)
* Tried `deCONZ_Setup_Win32_V2_08_00.exe` to update from `deCONZ_Setup_Win32_V2_05_04.exe` to `deCONZ_Rpi_0x26390500.bin.GCF` -> `FAILED` (*2184: Error: uart reset failed, check retry*)
* Tried `deCONZ_Setup_Win32_V2_05_88.exe` to update from `deCONZ_Setup_Win32_V2_05_04.exe` to `deCONZ_Rpi_0x26390500.bin.GCF` -> `FAILED` (*2184: Error: uart reset failed, check retry*)
* Tried `deCONZ_Setup_Win32_V2_05_88.exe` to update from `deCONZ_Setup_Win32_V2_05_04.exe` to `deCONZ_Rpi_0x261e0500.bin.GCF` -> `FAILED` (*2184: Error: uart reset failed, check retry*)
* Tried `deCONZ_Setup_Win32_V2_05_04.exe` to update from  `deCONZ_Setup_Win32_V2_05_04.exe` to `deCONZ_Rpi_0x261e0500.bin.GCF` -> `SUCCESS`
* Tried `deCONZ_Setup_Win32_V2_05_04.exe` to update from  `deCONZ_Rpi_0x261e0500.bin.GCF` to `deCONZ_Rpi_0x262f0500.bin.GCF` -> `SUCCESS`
* Tried `deCONZ_Setup_Win32_V2_05_04.exe` to update from  `deCONZ_Rpi_0x262f0500.bin.GCF` to `deCONZ_Rpi_0x26390500.bin.GCF` -> `SUCCESS`

manup, 2021-01-19:
The UART reset failed message means that GCFFlasher send a command to reboot the MCU into bootloader.
Since the old 0x260B0500 firmware doesn't support this command it is expected that this fails.

However GCFFlasher should as next step try to reboot the MCU via reset wire, I'm not sure if it is tried here.
Can you make the same as above but with added `-x 3` parameter, that should bring some more messages.

I'll make some tests to recreate the issue but it might take a few days.

sunseaandpalms, 2021-01-20:
@manup , my issue was solved by downgrading to `deCONZ_Setup_Win32_V2_05_04.exe`. Thus I am on `0x26390500` now. 

You want me to downgrade back to `0x260b0500` and retry updating with `deCONZ_Setup_Win32_V2_09_01.exe` and `-x 3` flag, is it right?

Ruthalas, 2021-01-20:
@manup 
I believe this is what you were looking for:
[image]

M-Gregoire, 2021-02-26:
I encountered the same issue with a Windows 10 Native host, with a Conbee I.
I used the most up to date Windows Installer and firmware. Doing the update from Ubuntu worked like a charm, so this seems specific to Windows.

Ruthalas, 2021-03-20:
I will be trying this upgrade on a linux Mint install soon, and will report back.

illese, 2021-03-27:
@sunseaandpalms  had the solution. Downgrade solved the issue. Once the new firmware is flashing will work as expected. Thanks.

closed 2021-04-25 by github-actions[bot]

uamithril, 2021-09-25:
The same problem, revert to older versions helped me. But I want to use the latest GUI. The only diff I have - ConBee II.

So last work version for me is 2.05.77, on that version I can update any firmware, even the latest - 0x26700700

When I try to use any version from 2.5.81 and up, I can't update firmware, I have an error - uart reset failed.

My setup: I use hassio Image, with the virtual machine.