dresden-elektronik/deconz-rest-plugin #4079: Thermostat Essentials Premium - Mode missing

N1c093, 2021-01-02:
## Describe the question or issue you are having

The thermostat "Thermostat Essentials Premium" (https://www.wesmartify.de/heizung/heizkoerperradiator/281/essentials-smart-home-heizkoerperthermostat-premium/) which was requested in issue #3381 and added in PR #3457 doesn't return any "mode" at the rest api. This results in an problem that the operation-mode in Home Assistant is always "off" and you can't set it to "heat". Everything else works fine and you can control the temperature.

Is it possible to add modes to this device? I don't know if this is a bug or work as designed. Maybe this is also an bug in Home Assistant.

Do you need any additional information?

## Screenshots

As you can see in this Home Assistant screenshot the "operation" is "off", but the thermostat is heating:

[image]

## Environment

 - Host system:  Raspberry PI 3
 - Running method:  Home Assistent deCONZ Add-on
 - Firmware version:  26660700
 - deCONZ version:  2.07.01 
 - Device:  ConBee II
 - Do you use an USB extension cable:  No
 - Is there any other USB or serial devices connected to the host system? Yes, USB SSD

## deCONZ Logs

## Additional context

Here are the current values from the rest api:

`{"config":{"battery":100,"heatsetpoint":2350,"locked":false,"offset":-200,"on":true,"preset":"manual","reachable":true,"schedule":{},"schedule_on":null,"setvalve":true,"windowopen_set":true},"ep":1,"etag":"383732a536bf32546936bb52f3a7aaa4","lastseen":"2021-01-02T13:59Z","manufacturername":"_TYST11_jeaxp72v","modelid":"eaxp72v","name":"Thermostat-Schlafzimmer","state":{"lastupdated":"none","on":null,"temperature":2490},"swversion":"20180727","type":"ZHAThermostat","uniqueid":"08:6b:d7:ff:fe:c1:f2:cf-01-0201"}`

Smanar, 2021-01-02:
I don't remember all the story, but from my memory this field was removed, because this device use "preset" command instead of "mode"

N1c093, 2021-01-02:
I think you are right, but I think this results in this strange behavior in home assistant.

I think "operation" or sometimes called "hvac_mode" in Home Assitant is a representation of "mode" from the deconz rest api. See: https://github.com/home-assistant/core/blob/6c3362217efad87426f674e0a4b8e61bfab051d0/homeassistant/components/deconz/climate.py#L113

I also can be wrong. I'm not an expert with the Home Assistant integration of deconz

Smanar, 2021-01-02:
On my side IDK how work HA ^^. But if the field "state/mode" is not in the json, I think HA can do something on his side.
There is this situation for some thermostats too.

Wich one value you have for "operation" : on, off and auto ?

N1c093, 2021-01-02:
Okay, maybe it's a problem on HA and I have to open an issue there. Can somebody else help me with this problem and knows where the issue came from? Deconz-Rest Api or Home Assistant?

I have the folowing values avaiable: "heat" and "off"
If I turn it to "heat" the values goes to "off" again after an seconds. See example: 
[image]

So the "operation/hvac_mode" stays "off" all the time. But I can control the temperature manually or with an preset.

Smanar, 2021-01-02:
For me it s normal, you have all "mode" in preset for this device.
Working preset are  "holiday" (frost control) "auto" and "manual"

N1c093, 2021-01-02:
Yes, you are right. 

You also were right some comments ago.
> "operation" is not the State/on representation ? Null on your json.

The "operation/hvac_mode" is always "off", it the "state/on" is not returning any value
https://github.com/home-assistant/core/blob/dev/homeassistant/components/deconz/climate.py#L176

As you already mentioned my "state/on" is NULL. Is there some way I can fix it?
Sorry for this confusing issue.

Smanar, 2021-01-02:
From the code all seem fine.
You can take a look at the log, in deconz/help with "info" and "info_l2"
Tuya log start all by "tuya debug", to fill the value state/on I m using dp = 0x02 0x6D

N1c093, 2021-01-03:
Attached you will find my log file. I already filtered for "tuya debug" I changed the temperature physically at the thermostate.
[Tuya_log_03.01.2021.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5761036/Tuya_log_03.01.2021.txt)

I can't see any communication with with "0x02 0x6D". For example the "heatsetpoint" is transmitted successfully.

Btw. Should I create a new issue? Because the only remaining problem is, that the "state/on" is NULL.

Smanar, 2021-01-03:
Yes, you have only

>01:01:23:077 Tuya debug 5 : Status: 1 Transid: 211 Dp: 515 (0x02,0x03) Fn: 0 Data 222

temperature

>01:08:07:187 Tuya debug 5 : Status: 1 Transid: 216 Dp: 1028 (0x04,0x04) Fn: 0 Data 1

Preset > auto

>01:08:12:198 Tuya debug 5 : Status: 1 Transid: 217 Dp: 514 (0x02,0x02) Fn: 0 Data 110

Setpoint

And nothing more.
For information it will be a new correction for tuya code that will change lot of thing, some device miss temperature for exemple wirhout it, perhaps it can unlock some setting for you too https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3922

You can wait for this version to be sure.

Else I can remove too the "state/on" but it seem working for other devices (perhaps not exactly your, tuya use a lot of variant)

N1c093, 2021-01-04:
Thank you for your help and feedback. 👍  I think we can't change this behavior of my device.
I just wonder, why my device works different then the other "Essentials Premium Thermostat". It's exactly the same device and the same firmware.

I hope the linked issue will also solve my problems. I will close this issue now, because there doesn't seems to be any problem with the deconz software.

closed 2021-01-04 by N1c093

Smanar, 2021-01-04:
>I just wonder, why my device works different then the other "Essentials Premium Thermostat". It's exactly the same device and the same firmware.

It can come from the device, tuya stuff are realy similar but have some variant.
But can come from the plugin code too, if you have take a look on the issue, it works for some users, but other haven't some temperature, just because of the code. This bug is realy 'random"