dresden-elektronik/deconz-rest-plugin #4081: Aqara sensors sporadically disconnected after deCONZ restart

erd82, 2021-01-02:
## Describe the bug

After restarting deCONZ sporadically my Aqara sensors do not send updates anymore.

This bug happens after update to latest deCONZ version - similarly to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4075.

This example shows that the bathroom sensor did not update for three hours. Even if you warm up the sensor manually there is not reaction / update.
[image]

If you check the temperature over the last couple of hours, you can see that one sensor (living room) started updating again and another sensor (bathroom) stopped sending updates:
[image]

## Steps to reproduce the behavior

Restart complete host (Raspberry Pi 4 Model B) including Home Assistant + deCONZ add-on OR only restarting deCONZ add-on leads to the described issue.

## Expected behavior

Sensors should constantly deliver data - even after restart.

## Screenshots

see above.

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: 2.07.01 / 12/8/2020
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

Mimiix, 2021-01-02:
@SwoopX wasn't this a duplicate?

Phreet, 2021-01-02:
I can confirm this behaviour for AQARA sensors on RaspBee II, both on Raspbian and as a Home Assistant deCONZ Add-On. I also - after couple of issues for itself - found this behaviour for multiple FW versions of RaspBee II.

## Environment
 - Host system: Raspberry Pi 4B (4GB)
 - Running method: Raspbian + Home Assistent deCONZ Add-on
 - Firmware version: 26690700, 26680700, 26670700
 - deCONZ version: 2.07.01 
 - Device: RaspBee II

## Additional context
I found that especially Aqara Temperature and vibration sensors do update their status values quite irregularly - compared to my previous state where everything seemed to work. Temperature sensors may take up to multiple hours to update their sensor readings. OTHER than Aqara door sensors, they immediately push their status change to DeCONZ / Phoscon.

SwoopX, 2021-01-02:
> @SwoopX wasn't this a duplicate?

@Mimiix Don't think so.

I have a monitoring on all my Aqara sensors, which gives me a Telegram message if they do not report within 2 hrs. All of them are totally stable and it doesn't matter if I reboot. The only exception is my, well yes, bathroom sensor. It drops every 1 - 2 weeks for whatever reason. But when it drops, it is completely silent, meaning when you sniff the traffic, there's literally 0 comms. Cost vs benefit is maximally high in may case, so I live with the re-pairing. Raspbee 1 and 2 with various firmwares here.

erd82, 2021-02-06:
Hello, unfortunately this issue could not be solved after several weeks now - even deCONZ updates did not help. Do you think that there is a general issue with ConBee 2 / deCONZ ? Or could there be a general configuration issue for deCONZ on my side?

I am also having issues with other devices (aqara motion sensor, [IKEA tradfri smart blinds FYRTUR](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4075), IKEA tradfri bulb E14 470lm, etc.). In most cases re-pairing the devices with deCONZ helps as a workaround. But it's not good if you cannot rely on device connectivity . I am tending to look for alternatives to ConBee 2.

thecem, 2021-02-16:
A repairing with 26680700 is for me not possible. Only with 26670700 there is a repairing possible.

closed 2021-03-19 by github-actions[bot]