dresden-elektronik/deconz-rest-plugin #4082: TRÅDFRI Motion sensor disconnected from network

riston, 2021-01-02:
## Describe the question or issue you are having
Recently my TRÅDFRI Motion sensor lost connection to my Zigbee coordinator initially, I thought it was a problem because of the battery level dropped to 15%. But even after the battery change the issue remained, is it possible to recover/wake up the lost device so I don't have to delete it? 

## Screenshots
[image]

## Environment
 - Host system: (**Raspberry** Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / **Home Assistent deCONZ Add-on** / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00) - latest
 - deCONZ version: (2.xx.yy) - latest
 - Device: (ConBee I / **ConBee II** / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / **no**) -- only relevant for ConBee I/II

LeoeLeoeL, 2021-01-03:
It could be a problem for Tradfri devices.
With deconz 2.9 and the latest firmware my 3 extenders lost the connection at the same time. I had to pair them again.

Mimiix, 2021-01-04:
What happens if you re-pair it?

maxheadroom999, 2021-01-05:
I lost the connection to different lights / on-off switches.

Tradfri, Paulmann (MLI)

I can´t figure out when and why

I need to re-pair then, Groups are still the same, but the naming is gone sometimes

riston, 2021-01-06:
I wasn't able to repair the device until the node was deleted after that everything worked again.

closed 2021-03-05 by github-actions[bot]