dresden-elektronik/deconz-rest-plugin #4083: Aqare Sesnor WSDCGQ11LM won

hflocki, 2021-01-03:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Metus88, 2021-01-03:
Did you tried:
Press and hold the "reset button" for 5 seconds... until blu led blinking... 
Release the "Reset button"
Every 1 second press (singol press) the reset button to keep alive the switch
Continue until the sensor is connected.

I hope it helps you

hflocki, 2021-01-04:
Yes i have tried this.
But for those sensors you must press and hold the button for 10 seconds and than every 3 seconds. I have tried it many many times and for the timeperiod of 3 minutes.
Also i have tried another combinations of press and holding the button with different times. No way.

AURELB87, 2021-01-06:
I have exactly the same problem. I just received my sensors AQARA and I think I have the same configuration than you!
I'm using the 'add-on Deconz' with my supervised HA.

0rsa, 2021-02-22:
I have also this issue.

Mimiix, 2021-02-22:
Can you provide some logs?

0rsa, 2021-02-22:
I sucessfully linked my sensors using this method:
- Reset sensor, it will blink 3x
- Then it will turn on 1 time (I guess it sends a signal)
- Then it will turn on again one time when he receives an answer

At this moment, two different behaviours :
- The sensor sucessfully connects to Conbee, the sensor blinks again many times quickly
- The sensor does not blink, press again sensor button to send a new signal and receive a new answer and repeat this step to finally get a successfully connection (sometimes I need to press sensor button around 10x to etablish connection).

hflocki, 2021-03-16:
The solution doesn't work for me.
I can't provide logs because i can't connect them

Mimiix, 2021-03-16:
@hflocki You can provide logs from during pairing :)

closed 2021-04-18 by github-actions[bot]