dresden-elektronik/deconz-rest-plugin #4085: Gledopto GL-C-009 not pairing

m-kuutti, 2021-01-03:
## Describe the question or issue you are having

A factory-fresh Gledopto GL-C-009 can’t seem to pair with deconz. Not on first power-on, not after a large number of resets. The same unit connected with a Philips hue hub without a hitch on first try.

Based on a number of github threads, I assumed it should be supported although not on the supported devices list.

## Screenshots

## Environment
**Raspberry Pi 4
Raspbian**
 - Firmware version: (**26520700**)
 - deCONZ version: (**2.09.00** / 23/12/2020)
 - Device: (**Raspbee II**)
 - Do you use an USB extension cable: (**no**)
 - Is there any other USB or serial devices connected to the host system? **No**

## deCONZ Logs

Mimiix, 2021-01-04:
Hi! 

Do you have any  other devices connected to deCONZ?

Also: Are you using a SSD?

m-kuutti, 2021-01-04:
Hi, I’ve got a bunch of other devices from various vendors connected.
[image]

I’m running the pi with the Sd card only.

ebaauw, 2021-01-04:
Based on the mac address, I think the `0x8162` node is the GLEDOPTO?  Seems like it was connected but isn't now (it looks greyed out).  And deCONZ didn't read the descriptors (the right drop-down dot is missing).

Can you try and select _Read Node Descriptor_ from the left drop down dot?  If the node un-greys, that's hopeful; if you see the status "light" blinking red, that's not good.  Try resetting the dimmer once more, followed by power cycling it, while searching for new devices.  Once the right dot appears, select _Read Simple Descriptor(s)_ from the left dot to populate the endpoints and clusters on the right dot.  Once you see the _Basic_ cluster, try to read the attributes, while still searching for new devices.  If the REST API still doesn't create the resource, please post screenshots of the node (with the endpoints and clusters) and of the _Node Info_ panel.

m-kuutti, 2021-01-05:
Hi ebaauw,

thanks for looking into this. I had removed the Gledopto to test it on the Philips hub. I deleted the node but managed to only get it back partially as before. No amount of resetting and powercycling while searching or not lead me further. A connected light blomks and dims during the search but in the end, I always just get the blinking red status, and no right dot appearing. So this is the best I can do:

[image]
[image]

If the device were properly paired, would it be supported by Phoscon or just deconz?

ebaauw, 2021-01-05:
Should be by both, afaik, but I don't use Phoscon.

Can you try and pair the dimmer with the Hue bridge once more.  Then search for new devices from Phoscon and remove the dimmer from the Hue app.  The Hue bridge should send a factory reset request to the dimmer, but I don't really know if the GLEDOPTO will honour this.

When resetting the dimmer through power-on power-off sequence, make sure to disconnect the dimmer from the power adapter.  If you just unplug/plug the adapter, it might not work as the adapter still has some residual load in its capacitors.

m-kuutti, 2021-01-11:
I managed to re-pair the device with the Hue bridge. Then started search for new lights and deleted the Gledopto from Hue. Nothing found during Phoscon search. I have been powering off between the supply and the dimmer so that’s not the issue. But the darn thing is very temperamental with the resetting sequence. I wish all deveices had a reset button.

Are there any logs etc. that could be of use in trying to assess what is going on? Or shall I just give up?

Mimiix, 2021-01-12:
@m-kuutti 

To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

m-kuutti, 2021-01-12:
Ok, done that. Quite a lot seems to be going on all the time. Is there a way to grep releveant entries after trying to pair the gledopto?

Mimiix, 2021-01-12:
I'd like to have the full log.

Pastebin preffered.

m-kuutti, 2021-01-12:
Ok. Will try soon.

m-kuutti, 2021-01-12:
Done it: https://pastebin.com/Xgba01C3 has the log for the three minutes Phoscon was searching for new lights.

m-kuutti, 2021-01-16:
Hi Guys, 
Today I tried to pair a new LIDL bulb, and ran into the same exact behaviour as with the Gledopto dimmer. I hard rebooted the Pi, and lo and behold – now both paired at first try. So deconz was in a bad state, not the dimmer. 

Thank you for your support, and sorry for bothering you. I’ll try and remember to always freshly reboot before crying out for help in the future.

closed 2021-03-05 by github-actions[bot]