dresden-elektronik/deconz-rest-plugin #4091: IKEA Tradfri Remote E1810 "Hold/Release events" for < > are not recognized by deconz

tnaumann71, 2021-01-04:
## Describe the bug
IKEA Tradfri Remote E1810 "Hold/Release events" for < > are not recognized by deconz
When holding "<" or ">" the deconz_events
4001 → Hold <
4003 → Release <
5001 → Hold >
5003 → Release >
are not recognized.

## Steps to reproduce the behavior
The problem is reproducable. Steps:

  Click and Hold the "<" or ">" on the IKEA Tradfri remote

## Expected behavior
deCONZ should report the Events 4001/4003 or 5001/5003.

## Screenshots

## Environment
	• Hass OS 5.9
	• Home Assistant Core version: 2020.12.2
	• deCONZ 2.07.01 / 8.12.2020 Firmware 26660700
	• AppDaemon 4 version: 4.0.5
	• ControllerX version: 4.3.0
	• HACS version (if installed from there): 1.9.0
	• ConBeeII (connected via 1m USB extension cable)
	• IKEA Tradfri Remote E1810 mit Firmware 2.3.014
	* Hass OS running on a RPi 4B 2 GB
	* SSD Disk is connected via USB3 to SATA Adapter

## deCONZ Logs
08:35:59:283 dev /dev/ttyAMA0
08:35:59:289 GW firmware version: 0x26660700
08:35:59:290 GW firmware version is up to date: 0x26660700
08:36:00:233 APS-DATA.request id: 118, addrmode: 0x03, addr: 0xd85def11a10048da, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x12 queue: 0 len: 5 tx.options 0x00
08:36:00:258 APS-DATA.confirm id: 118, status: 0x00 SUCCESS
08:36:00:259 APS-DATA.confirm request id: 118 -> erase from queue
08:36:00:296 aps request id: 118 finished, erase from queue
08:36:00:303 APS-DATA.indication srcAddr: 0xcc20, srcEp: 0x12 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -46
08:36:00:305 	asdu: 18f1010000001000
08:36:00:307 ZCL got data for node=0xCC20, cl=0x0006, at=0x0000, status=0x00, type=0x10
08:36:00:687 APS-DATA.indication srcAddr: 0x05d1, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0001, lqi: 255, rssi: -58
08:36:00:689 	asdu: 087a0a2100204a
08:36:00:693 [INFO] - No button handler for: TRADFRI remote control endpoint: 0x01 cluster: 0x0001 command: 0x0A payload[0]: 021
08:36:00:695 ZCL attribute report 0x680AE2FFFEC60F56 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
08:36:00:697 APS-DATA.request id: 124, addrmode: 0x02, addr: 0x05d1, profile: 0x0104, cluster: 0x0001, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
08:36:00:732 APS-DATA.confirm id: 124, status: 0x00 SUCCESS
08:36:00:734 APS-DATA.confirm request id: 124 -> erase from queue
08:36:00:772 aps request id: 124 finished, erase from queue
08:36:01:104 APS-DATA.indication srcAddr: 0x05d1, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 255, rssi: -58
08:36:01:107 	asdu: 057c117b0700010d00
08:36:01:110 [INFO] - Button 5002 Step ct warmer TRADFRI remote control
08:36:02:396 APS-DATA.indication srcAddr: 0x05d1, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 255, rssi: -58
08:36:02:400 	asdu: 057c117c0700010d00
08:36:02:404 [INFO] - Button 5002 Step ct warmer TRADFRI remote control
08:36:03:233 APS-DATA.request id: 139, addrmode: 0x03, addr: 0xd85def11a10048da, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x12 queue: 0 len: 5 tx.options 0x00
08:36:03:260 APS-DATA.confirm id: 139, status: 0x00 SUCCESS
08:36:03:263 APS-DATA.confirm request id: 139 -> erase from queue
08:36:03:315 APS-DATA.indication srcAddr: 0xcc20, srcEp: 0x12 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -46
08:36:03:317 	asdu: 18f2010000001000
08:36:03:318 APS-DATA.request id: 139 erase from queue
08:36:03:319 ZCL got data for node=0xCC20, cl=0x0006, at=0x0000, status=0x00, type=0x10
08:36:05:236 APS-DATA.indication srcAddr: 0x05d1, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 255, rssi: -46
08:36:05:238 	asdu: 057c117d0701010d00
08:36:05:241 [INFO] - Button 4002 Step ct colder TRADFRI remote control
08:36:06:474 APS-DATA.indication srcAddr: 0x05d1, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 255, rssi: -58
08:36:06:475 	asdu: 057c117e0701010d00
08:36:06:478 [INFO] - Button 4002 Step ct colder TRADFRI remote control
08:36:07:333 APS-DATA.request id: 165, addrmode: 0x03, addr: 0xd85def11a10048da, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x12 queue: 0 len: 5 tx.options 0x00
08:36:07:356 APS-DATA.confirm id: 165, status: 0x00 SUCCESS
08:36:07:358 APS-DATA.confirm request id: 165 -> erase from queue
08:36:07:408 APS-DATA.indication srcAddr: 0xcc20, srcEp: 0x12 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -46
08:36:07:411 	asdu: 18f3010000001000
08:36:07:413 APS-DATA.request id: 165 erase from queue
08:36:07:415 ZCL got data for node=0xCC20, cl=0x0006, at=0x0000, status=0x00, type=0x10
08:36:07:772 APS-DATA.indication srcAddr: 0x05d1, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 255, rssi: -58
08:36:07:774 	asdu: 057c117f0701010d00
08:36:07:778 [INFO] - Button 4002 Step ct colder TRADFRI remote control
08:36:08:853 Mgmt_Lqi_req zdpSeq: 35 to 0x7CB03EAA0A0978A8 start index 3
08:36:08:854 APS-DATA.request id: 175, addrmode: 0x03, addr: 0x7cb03eaa0a0978a8, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
08:36:08:891 APS-DATA.confirm id: 175, status: 0x00 SUCCESS
08:36:08:894 APS-DATA.confirm request id: 175 -> confirmed, timeout 1609745768
08:36:08:940 APS-DATA.indication srcAddr: 0xafea, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -58
08:36:08:943 	asdu: 2300040301ff5106ffff2e2100a5a91804008d1500f5371201015c
08:36:08:945 APS-DATA.indication request id: 175 -> finished
08:36:08:947 APS-DATA.request id: 175 erase from queue
08:36:08:949 ZDP status = 0x00 -> SUCCESS
08:36:08:951 ZDP Mgmt_Lqi_rsp zdpSeq: 35 from 0x7CB03EAA0A0978A8 total: 4, startIndex: 3, listCount: 1
08:36:08:953     * neighbor: 0x00158D000418A9A5 (0x37F5), LQI: 92, relation: 0x01 rxOnWHenIdle: 0
08:36:10:232 APS-DATA.request id: 184, addrmode: 0x03, addr: 0xd85def11a10048da, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x12 queue: 0 len: 5 tx.options 0x00
08:36:10:259 APS-DATA.confirm id: 184, status: 0x00 SUCCESS
08:36:10:261 APS-DATA.confirm request id: 184 -> erase from queue
08:36:10:314 aps request id: 184 finished, erase from queue
08:36:10:321 APS-DATA.indication srcAddr: 0xcc20, srcEp: 0x12 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -46
08:36:10:324 	asdu: 18f4010000001000
08:36:17:233 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)

## Additional context
I checked with other users / developer that 4001/4003 and 5001/5003 are recognized via z2m Integartion with the IKEA Tradfri Remote E1810 FW 2.3.014.
Please find here: https://github.com/xaviml/controllerx/issues/215

SwoopX, 2021-01-04:
Just checked with mine, and it doesn't emit those events either. However, I've sniffed the traffic, and the device doesn't even send the "expected" commands, but just all the same. Therefore, it is related to the device (configuration) and not deconz itself.

tnaumann71, 2021-01-04:
One user reported the device with the fw FW 2.3.014 emits the events with the z2m (zigbee2mqtt) integration.
This sound s strange. Are there different 2.3.014 versions/builds out there? Because in deconz it shows version as 0x23014631 ... so there is "631" after 2.3.014.

Find here: https://github.com/xaviml/controllerx/issues/215#issuecomment-753690851

SwoopX, 2021-01-04:
That's what I got

[image]

> Because in deconz it shows version as 0x23014631 ... so there is "631" after 2.3.014.

Where?

tnaumann71, 2021-01-04:
I got the evidence concerning E1810 IKEA Tradfri remote fires hold/release events and zigbee2mqtt catch them:

Even with the same dateCode / Date Code (see the date "20190401")

Find the log here:

Quote: Feel free to cut and my z2m log below showing that Ikea E1810 HW v1, with fw 2.3.014 indeed sends hold and release commands for all four direction buttons.

```
{"dateCode":"20190401","description":"TRADFRI remote control","friendly_name":"Office Ikea 5 button","hardwareVersion":1,"ieeeAddr":"0xec1bbdfffe1b99e8","lastSeen":1609751218742,"manufacturerID":4476,"manufacture Name":"IKEA of Sweden","model":"E1524/E1810","modelID":"TRADFRI remote control","networkAddress":32904,"powerSource":"Battery","softwareBuildID":"2.3.014","type":"EndDevice","vendor":"IKEA"}

Zigbee2MQTT:info  2021-01-04 13:47:24: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"arrow_left_hold","battery":37,"linkquality":102,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:24: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"","battery":37,"linkquality":102,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:25: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"arrow_left_release","action_duration":0.859,"battery":37,"duration":0.859,"linkquality":99,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:25: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"","battery":37,"linkquality":99,"update":{"state":"idle"},"update_available":false}'

Zigbee2MQTT:info  2021-01-04 13:47:27: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"arrow_right_hold","battery":37,"linkquality":99,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:27: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"","battery":37,"linkquality":99,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:28: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"arrow_right_release","action_duration":0.823,"battery":37,"duration":0.823,"linkquality":102,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:28: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"","battery":37,"linkquality":102,"update":{"state":"idle"},"update_available":false}'

Zigbee2MQTT:info  2021-01-04 13:47:29: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"brightness_up_hold","battery":37,"linkquality":102,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:29: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"","battery":37,"linkquality":102,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:30: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"brightness_up_release","battery":37,"linkquality":102,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:30: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"","battery":37,"linkquality":102,"update":{"state":"idle"},"update_available":false}'

Zigbee2MQTT:info  2021-01-04 13:47:32: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"brightness_down_hold","battery":37,"linkquality":102,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:32: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"","battery":37,"linkquality":102,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:33: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"brightness_down_release","battery":37,"linkquality":99,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2021-01-04 13:47:33: MQTT publish: topic 'zigbee2mqtt/Office Ikea 5 button', payload '{"action":"","battery":37,"linkquality":99,"update":{"state":"idle"},"update_available":false}'
```

tnaumann71, 2021-01-04:
I got my version number from here:

[image]

Mimiix, 2021-01-04:
@tnaumann71 You should read from the basic cluster on the device.

tnaumann71, 2021-01-04:
I only got the DateCode but no SW Build ID. But this identical to yours.

[image]

Mimiix, 2021-01-04:
I just tested this, and it seem to work on my setup:

```
Event 4 fired 4:56 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "bedafstandsbediening",
        "unique_id": "68:0a:e2:ff:fe:ad:60:c0",
        "event": 5003
    },
    "origin": "LOCAL",
    "time_fired": "2021-01-04T15:56:05.995672+00:00",
    "context": {
        "id": "580da5d51b177be94324b43dd1125843",
        "parent_id": null,
        "user_id": null
    }
}
Event 3 fired 4:56 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "bedafstandsbediening",
        "unique_id": "68:0a:e2:ff:fe:ad:60:c0",
        "event": 5001
    },
    "origin": "LOCAL",
    "time_fired": "2021-01-04T15:56:04.825788+00:00",
    "context": {
        "id": "1f87c58e001c62f61509b69ba3e520ba",
        "parent_id": null,
        "user_id": null
    }
}
Event 2 fired 4:56 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "bedafstandsbediening",
        "unique_id": "68:0a:e2:ff:fe:ad:60:c0",
        "event": 4003
    },
    "origin": "LOCAL",
    "time_fired": "2021-01-04T15:56:03.033681+00:00",
    "context": {
        "id": "2811e4e3f599e114d9c78018fd78526a",
        "parent_id": null,
        "user_id": null
    }
}
Event 1 fired 4:56 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "bedafstandsbediening",
        "unique_id": "68:0a:e2:ff:fe:ad:60:c0",
        "event": 4001
    },
    "origin": "LOCAL",
    "time_fired": "2021-01-04T15:56:01.660774+00:00",
    "context": {
        "id": "421b3ca69527fb1f673c6f437301adc0",
        "parent_id": null,
        "user_id": null
    }
}
Event 0 fired 4:56 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "bedafstandsbediening",
        "unique_id": "68:0a:e2:ff:fe:ad:60:c0",
        "event": 4002
    },
    "origin": "LOCAL",
    "time_fired": "2021-01-04T15:56:00.124065+00:00",
    "context": {
        "id": "05485872ec6b2bf21dc3d092b001c6d7",
        "parent_id": null,
        "user_id": null
    }
}
```
I'm on 2.07.00.

tnaumann71, 2021-01-05:
After factory reset and new pairing and holding the remote direct beside the ConBee II nothing changed.

Found and solved the issue.
During my test the battery went from 80 to 60% low. I renewed the battery and all events are now emitted.

Strange thing is that this happens with a battery level higher than 50%. Anyway...

Thanks for your support. I really appreciate that and your work.
Best regards, Torsten

closed 2021-01-05 by tnaumann71

SwoopX, 2021-01-05:
@tnaumann71 Great to hear that it finally worked out for you.

Just for completeness sake, this is what the device sends at my end on ">" longpress

[image]

All the commands send are identical and do not differ as one would expect. However, I just noticed that the commands emitted are all broadcasts. So eventually, there should have been some other bindings involved, being the reason for the device to act differently 🤷