dresden-elektronik/deconz-rest-plugin #4096: Eurotronic Spirit Zigbee does not turn off

ChristophHornung, 2021-01-04:
## Describe the bug

Some Spirit Zigbee TVRs do not set the "off" flag correctly. When turning them off the value of 'Host Flags' remains at 0x01. This only seems to affect a set of new Spirits I bought, as the old one I purchased does turn off correctly. (They are also the ones described in #3911)

All report their version the same though: 2.22.5.35 - Eurotronic SPZB0001 - 20191014.

(Re-reported for #4095 because the bot closed the last one since I deleted the empty "Screenshots" section)

## Steps to reproduce the behavior

I see the following happening when things don't work

1. Turn Off the Spirit (in HA)
2. The command is triggered inside deCONZ, but with the Thermostat cluster open the Host Flags value remains at 0x01 the entire time.
3. Re-read the host flags value -> Returns 0x01 and the device is switched back to "on" in HA.

For the Spirit that behaves correctly things look different:

1. Turn Off the Spirit (in HA)
2. The command is triggered inside deCONZ, and the Host Flags in the cluster view is set to 0x11 immediately.
3. Re-read the host flags value -> Returns 0x11 and the device remains switch off in HA.

The temperatures are always set correctly, so the issue does not seem to be related to a communication problem.

## Expected behavior
The Host Flags should be set correctly to 0x11 indicating "off".

## Screenshots

## Environment
- Host system: Raspberry Pi 3
- Running method: Home Assistent deCONZ Add-on 6.6.0
- Firmware version: 26680700
- deCONZ version: (2.07.01)
- Device: ConBee II
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system? no
- HA Supervisor 2020.12.7
- Host System Home Assistant OS 5.9

## deCONZ Logs
This is a log that captures turning on the Spirit and then re-reading the Host Flags value. (On sensor 11 (0x00158D00053D2DCB))

[hostFlagsProblem.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5764264/hostFlagsProblem.txt)

## Additional context
Network consists of quite a few Spirits as well as some Ikea and Lidl lights as well as Lidl Power Plugs. I am trying to find a difference between the Spirit that works and the ones that don't but so far I don't see anything substantially different.

ChristophHornung, 2021-01-04:
Additional information:

When I try to write the value to the cluster in deCONZ manually I get an error:

[image]

The log during that time:
[manualWriteFailure.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5764414/manualWriteFailure.txt)

ChristophHornung, 2021-01-08:
Is there anything I can do to help find the issue? Are there any other commands I could send to the Spirit to check for problems?

magicbenny, 2021-01-08:
sounds like here [https://github.com/Koenkk/zigbee2mqtt/issues/4571](url)

rosevalleyDE, 2021-01-12:
I have the same issue. Older thermostats work, the newer ones cannot turned off with the off-button in Homeassistant. I need to manually decrease the temp to the minimum of 7°C instead which is not ideal. 
Thanks in advance for looking into it and hopefully fix it.

closed 2021-03-05 by github-actions[bot]

boehser-enkel, 2021-04-06:
Same for me

kanuni89, 2021-05-03:
Same for mee

Mimiix, 2021-05-04:
All issues are spoken of here:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1098

According to the devs its the firmware, so there's nothing to do about it.