dresden-elektronik/deconz-rest-plugin #4097: Please add USB Loop issue and solution to instrucions

dvbit, 2021-01-04:
## Feature request type

Include instructions on how to fix USB loop problem in readme of the product until it is not presenting itself / fixed 

## Description

It is not so infrequent, especially after a successfull update of firmware, to have the dongle entering a "loop" of negotiation with the USB Host.
The update is done via phoscon app and is reported as "succesfull" but the dongle starts exhibiting the following behaviour:
recognized then dropped, new id assigned and so on
Verified to be happening on following architectures:
ARM - Raspbian
Intel - Debian
Intel - MS Win 10

This is happening frequently enough to have the dignity of being reported as a known issue with a workaround.
For a confirmation do search on google on deconz github for USB Loop.
Posts on reddit are also present

It would be helpful for user to have such information in a single official place in order not to waste time.
This is the expected behavior for a Company supporting a product

## Considered alternatives

- Issue the command with first firmware; sudo GCFFlasher_internal -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x264a0700.bin.GCF 
- reset coordinator
- Issue the command with desired firmware; sudo GCFFlasher_internal -d /dev/ttyACM0 -f deCONZ_ConBeeII_0???.bin.GCF
## Additional context

test

Mimiix, 2021-01-04:
The real question here, is to see how you've updated the device. If you are using the HA Addon, the thing fails because their implementation is incorrect.

[This page](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually) clearly describes on how to update manually and there is some do and dont's in there. Feel free to edit the page, it's a wiki after all!

If you are missing info, you can add the wiki page yourself and contribute 😄

dvbit, 2021-01-04:
The stick has not yet been used in homeassistant.
I have followed the instructions of the page ad litteram.
I have first followed the instructions to update via phoscon app (I think this is your app correct ?)
It reported a success while it was a failure.
This is often called 'bug'.
In the instructions provided on your page nowere it is written what to do in case the issue appears .
This is often called 'known issue and workarounds'
And the issue, as you know, appears in some cases.
So if you want I can add it to the instructions of the phoscon app.

If I chose to buy a product and not use open hardware is for the support I can get in case of trouble and the updates.
If I do not have support part of that value is not there anymore .

The real matter is this. You may want the sales and marketing department read this comment.

closed 2021-01-04 by dvbit