dresden-elektronik/deconz-rest-plugin #4099: Migrating from RaspBee (I or II) and ConBee II, Backup/Restore ConBee II

lukicsl, 2021-01-04:
## Describe the bug
I am running deCONZ inside docker. 
I started with using RaspBee I and buildup a network of 80+ nodes.
I later bought a ConBee II hoping to migrate the installation to ConBee II.
I failed multiple times and put the idea byside.
2 months ago I ordered RaspBee II and the migration was as: Shutdown Pi, replace the RB I for RB II, boot up, start the container, and it was done. Works as expected.

Yesterday the RB II stopped working as was not visible as a device under /dev.
This incident brought back the idea of migrating the installation on the CB II.

## Steps to reproduce the behavior

I first installed back the RB I into the Pi and it picked up the installation without any issue: 

[image]

I made a fresh backup.
I shut down and removed RB I and install CB II. Booted up, started container, and loaded the backup, as a result, all nodes were visible but not connected to the coordinator:

[image]

I have the feeling that the backup restore process does not work on CB II.
I have made a test on a fresh installation with CB II: I added 2 lights from scratch. The installation worked as expected. After back-restore the lights do not connect anymore to the coordinator:

[image]

## Expected behavior

  Backup made fro RB I or RB II should be ok to be imported for CB I

## Screenshots

  in the description above

## Environment
 - Host system: (Raspberry Pi )
 - Running method: Marthoc Docker container
 - Firmware version: 264A0700
 - deCONZ version: 2.09.00 / 23.12.2020
 - Device: ConBee II / RaspBee I / RaspBee II
 - Do you use an USB extension cable: no

## deCONZ Logs

No logs
-->

## Additional context

no additional context

lukicsl, 2021-01-04:
I have downloaded the sd-card image:

[image]

And added my 2 test lights and created a test group. everything worked as expected.

Then I made a backup from this 2nd test configuration.

I then loaded the backup from the previous example created from the docker image and got the following result:

[image]

More or less as expected

I loaded then the 2nd test configuration backup (from scratch created group and 2 light on the sd-card image installation) and got this result:

I had to wait for connect button to be visible:
[image]

And after connect:

[image]

one light lost its name no connection to the coordinator and the group has gone

[image]

I do not have a spare RaspBee yet to state that it is a ConBee issue, but the backup restore process does not work. 

**This is really bad.**

SwoopX, 2021-01-05:
Restore your backup. Then, as in your 2nd screenshot of the initial post, set `APS Use Ext PAN ID`, copy in the MAC of the `TC address`. That should do the trick.

Had pretty much the same while switching from Raspbee 1 to Raspbee 2 a couple of days before.

Please, do not fiddle around with your network config unless you exactly know what you're doing. Also, there's no need to re-pair devices!

lukicsl, 2021-01-05:
I had a look at the logs and I see a segmentation fault at Step 9:

```
Segmentation fault (core dumped)
14:05:55:583 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 242, 4, x'f2e0a1610000012100012100', 1609855555 FROM devices WHERE mac = '14:b4:[marthoc/deconz] Starting deCONZ...

```

Mimiix, 2021-01-06:
That's a nice mac Adress.

Mimiix, 2021-01-06:
I'll ask Manuel to look at it.

lukicsl, 2021-01-06:
The mack address I suppose was cut off by the core dump, the "[marthoc/deconz] Starting deCONZ..." is the restarted process starting putting out log entries.

Mimiix, 2021-01-06:
Aah okay! Fair enough. 
Already found it strange 😂

closed 2021-03-05 by github-actions[bot]