dresden-elektronik/deconz-rest-plugin #4101: Sinope thermostat TH1300ZB paired to deCONZ but not recognized in REST-API and Phoscon app.

zguy2008, 2021-01-04:
## Describe the bug

Sinope thermostat TH1300ZB paired to deCONZ but not recognized in REST-API and Phoscon app.

[image]
[image]
[image]

I am not able to find the above attributes in deCONZ gui.

## Steps to reproduce the behavior

REST-API request
[image]

{ "config": { "UTC": "2021-01-04T19:51:04", "apiversion": "1.16.0", "backup": { "errorcode": 0, "status": "idle" }, "bridgeid": "00212EFFFF05BB1C", "datastoreversion": "93", "devicename": "RaspBee II", "dhcp": true, "disablePermitJoinAutoOff": false, "factorynew": false, "fwversion": "0x26690700", "gateway": "192.168.1.1", "internetservices": { "remoteaccess": "disconnected" }, "ipaddress": "192.168.1.71", "linkbutton": false, "localtime": "2021-01-04T14:51:04", "mac": "dc:a6:32:dc:ce:5e", "modelid": "deCONZ", "name": "ZguyPhoscon", "netmask": "255.255.255.0", "networkopenduration": 180, "ntp": "synced", "panid": 21495, "portalconnection": "disconnected", "portalservices": false, "portalstate": { "communication": "disconnected", "incoming": false, "outgoing": false, "signedon": false }, "proxyaddress": "none", "proxyport": 0, "replacesbridgeid": null, "rfconnected": true, "starterkitid": "", "swupdate": { "checkforupdate": false, "devicetypes": { "bridge": false, "lights": [], "sensors": [] }, "notify": false, "text": "", "updatestate": 0, "url": "" }, "swupdate2": { "autoinstall": { "on": false, "updatetime": "" }, "bridge": { "lastinstall": "2020-12-08T16:14:37", "state": "allreadytoinstall" }, "checkforupdate": false, "install": false, "lastchange": "", "lastinstall": "", "state": "allreadytoinstall" }, "swversion": "2.7.1", "timeformat": "12h", "timezone": "Etc/GMT+5", "uuid": "f653ea6b-5b7b-42e2-af5a-386d56468dde", "websocketnotifyall": true, "websocketport": 443, "whitelist": { "047DC91BCE": { "create date": "2020-12-28T01:20:43", "last use date": "2020-12-31T23:36:06", "name": "JowiHue" }, "17A96A69BD": { "create date": "2021-01-03T20:45:59", "last use date": "2021-01-04T19:51:04", "name": "zguyapp" }, "234FD02F55": { "create date": "2020-12-27T22:41:01", "last use date": "2020-12-27T22:42:17", "name": "Phoscon#B1647x810" }, "420A458F32": { "create date": "2021-01-01T00:58:24", "last use date": "2021-01-03T20:01:24", "name": "Phoscon#B1646x771" }, "4FF2A293C2": { "create date": "2021-01-03T20:15:17", "last use date": "2021-01-04T19:51:00", "name": "Phoscon#B1122x771" }, "507729CD13": { "create date": "2020-12-31T22:46:07", "last use date": "2020-12-31T23:36:44", "name": "Phoscon#B1381x855" }, "64E40FFC21": { "create date": "2020-12-31T20:39:58", "last use date": "2020-12-31T23:36:44", "name": "Phoscon#B1004x630" }, "BCD782C80C": { "create date": "2020-12-27T23:27:46", "last use date": "2020-12-28T02:08:25", "name": "Phoscon#B1843x810" } }, "zigbeechannel": 15 }, "groups": { "1": { "action": { "alert": "none", "bri": 127, "colormode": "hs", "ct": 0, "effect": "none", "hue": 0, "on": false, "sat": 127, "scene": null, "xy": [ 0, 0 ] }, "devicemembership": [], "etag": "b05acd997068ee3e5b3b55e555ed760f", "id": "1", "lights": [], "name": "Test group", "scenes": [], "state": { "all_on": false, "any_on": false }, "type": "LightGroup" } }, "lights": { "1": { "etag": "30b41cb9ddd2a6080f0233b04e7e0439", "hascolor": false, "lastannounced": null, "lastseen": "2021-01-04T19:50Z", "manufacturername": "dresden elektronik", "modelid": "RaspBee II", "name": "Configuration tool 1", "state": { "reachable": true }, "swversion": "0x26520700", "type": "Configuration tool", "uniqueid": "00:21:2e:ff:ff:05:bb:1c-01" } }, "resourcelinks": {}, "rules": {}, "scenes": {}, "schedules": {}, "sensors": { "1": { "config": { "configured": true, "on": true, "sunriseoffset": 30, "sunsetoffset": -30 }, "etag": "b05acd997068ee3e5b3b55e555ed760f", "manufacturername": "Philips", "modelid": "PHDL00", "name": "Daylight", "state": { "dark": false, "daylight": true, "lastupdated": "2021-01-04T17:01:56.644", "status": 170, "sunrise": "2021-01-04T12:37:03", "sunset": "2021-01-04T21:26:46" }, "swversion": "1.0", "type": "Daylight", "uniqueid": "00:21:2e:ff:ff:05:bb:1c-01" } } }

I was expecting to see something in there related to the thermostat.

## Expected behavior

## Screenshots

## Environment

Host system: Raspberry Pi
Running method: Raspbian
Firmware version: (26690700)
deCONZ version: (2.07.01)
Device: RaspBee II
Do you use an USB extension cable: N/A
Is there any other USB or serial devices connected to the host system? BT dongle for keyboard/mouse

## deCONZ Logs

## Additional context

Mimiix, 2021-01-04:
Okay, but why is this a bug? Seems to me it's just unsupported.

Mimiix, 2021-01-04:
After reading a git issue, it seems to should be. However, can you read the basic cluster and post a screenshot of it?

zguy2008, 2021-01-04:
That was my impression as well that it was supported.

Is this what you need?
[image]

Mimiix, 2021-01-04:
No the basic cluster. The top one.

zguy2008, 2021-01-04:
This is the only one that pulls.  How do I get that top one?

zguy2008, 2021-01-04:
And what does it look like?

Mimiix, 2021-01-04:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

zguy2008, 2021-01-04:
ok, well, hope that this is it you need

[image]

Mimiix, 2021-01-04:
Now you need to click the read button.

Make sure the device is awake.

zguy2008, 2021-01-04:
I did, didn't see a change.  It's a line powered thermostat, I'd assume this is online all the time.

Mimiix, 2021-01-04:
Is the light blinking red?

zguy2008, 2021-01-04:
no.  grey all the time.  the left bar is always yellow as well.

zguy2008, 2021-01-04:
Although paired, it looks like it's not talking.  Just changed the temperature at the device tried a rea again but no changes,

zguy2008, 2021-01-04:
at debug level ZCL, I get this when pressing the read button:
16:24:09:820 ZCL read cluster: 0x0000, attribute: 0x0000
16:24:09:821 ZCL read cluster: 0x0000, attribute: 0x0001
16:24:09:822 ZCL read cluster: 0x0000, attribute: 0x0002
16:24:09:822 ZCL read cluster: 0x0000, attribute: 0x0003
16:24:09:822 ZCL cmd-req nwk: 0x6ECF, profile: 0x0104, cluster: 0x0000 cmd: 0x00
16:24:09:823 ZCL read cluster: 0x0000, attribute: 0x0004
16:24:09:824 ZCL cmd-req nwk: 0x6ECF, profile: 0x0104, cluster: 0x0000 cmd: 0x00
16:24:09:824 ZCL read cluster: 0x0000, attribute: 0x0005
16:24:09:825 ZCL cmd-req nwk: 0x6ECF, profile: 0x0104, cluster: 0x0000 cmd: 0x00
16:24:09:825 ZCL read cluster: 0x0000, attribute: 0x0006
16:24:09:825 ZCL cmd-req nwk: 0x6ECF, profile: 0x0104, cluster: 0x0000 cmd: 0x00
16:24:09:826 ZCL read cluster: 0x0000, attribute: 0x0007
16:24:09:826 ZCL read cluster: 0x0000, attribute: 0x0008
16:24:09:826 ZCL read cluster: 0x0000, attribute: 0x0009
16:24:09:827 ZCL read cluster: 0x0000, attribute: 0x000A
16:24:09:827 ZCL cmd-req nwk: 0x6ECF, profile: 0x0104, cluster: 0x0000 cmd: 0x00
16:24:09:828 ZCL read cluster: 0x0000, attribute: 0x4000
16:24:09:828 ZCL cmd-req nwk: 0x6ECF, profile: 0x0104, cluster: 0x0000 cmd: 0x00
16:24:09:828 ZCL read cluster: 0x0000, attribute: 0x0010
16:24:09:829 ZCL cmd-req nwk: 0x6ECF, profile: 0x0104, cluster: 0x0000 cmd: 0x00
16:24:09:829 ZCL read cluster: 0x0000, attribute: 0x0011
16:24:09:829 ZCL read cluster: 0x0000, attribute: 0x0012
16:24:09:830 ZCL read cluster: 0x0000, attribute: 0x0013
16:24:09:830 ZCL read cluster: 0x0000, attribute: 0x0014
16:24:09:830 ZCL cmd-req nwk: 0x6ECF, profile: 0x0104, cluster: 0x0000 cmd: 0x00
16:24:09:831 ZCL read cluster: 0x0000, attribute: 0xFFFD
16:24:09:831 ZCL cmd-req nwk: 0x6ECF, profile: 0x0104, cluster: 0x0000 cmd: 0x00

SwoopX, 2021-01-05:
The node info panel screenshot you've taken above is from the coordinator, so this is not helpful in our case. Can you please provide one from the device to investigate further?

zguy2008, 2021-01-05:
I believe the screen shot above is good.  Here is the one from the RaspBee 2:
[image]
The RaspBee responds when pushing the read button.

Here is the thermostat one, again:
[image]
It show the same as the first one posted, and it doesn't respond when pushing the read button.

SwoopX, 2021-01-05:
> I believe the screen shot above is good.

I'm telling you it's not. The very first screenshot has the coordinators NWK address (0x0000), the manufacturer code from Dresden Elektronik and the MAC address as well.

Also, as it seems that the device is not paired (anymore) to your network, you need to reset and re-pair it. Btw, are you sure your network is working allright?

zguy2008, 2021-01-05:
> > I believe the screen shot above is good.
> 
> I'm telling you it's not. The very first screenshot has the coordinators NWK address (0x0000), the manufacturer code from Dresden Elektronik and the MAC address as well.

In the previous message I put 2 screen shots.  

First one is of the RaspBee, where it does show Dresden Elektronik.  I'm not seeing any NWK or MAC address in that screen shot though.  Pressing the read button on this one did update the fields.

Second screen shot is from the thermostat.  However, pressing the read button doesn't do anything.  Hence why I suggested it was not communicating properly after discovering how to access these so-called clusters.

> 
> Also, as it seems that the device is not paired (anymore) to your network, you need to reset and re-pair it. Btw, are you sure your network is working allright?

I'm hard pressed to say as not much has been working so far.  At this point what I know is deCONZ and Phoscon app and REST-API can talk to the Rasbian.  I suppose that means serial comm. to the radio works.  I'm kinda assuming since I managed to pair the thermostat that the ZigBee network is working somewhat.

> 
> [image]
> 
The bar in my deCONZ is the same as your screen shot.

> And how did you try to pair your device?

Tried to add the device via the Phoscon app.  I'm not exactly certain how.  Only lights, switches and sensors show in the app.  I tried several things with switches and sensors and at some point the thermostat appeared, at which point I thought I was good to go.  Obviously not.

Is there a way to add the device with deCONZ instead?

SwoopX, 2021-01-05:
> Tried to add the device via the Phoscon app. I'm not exactly certain how. Only lights, switches and sensors show in the app. I tried several things with switches and sensors and at some point the [...]

No, that's totally fine and must be done this way. Was just asking to make sure your took the right path here. All good.

> The very first screenshot has the coordinators NWK address (0x0000), the manufacturer code from Dresden Elektronik and the MAC address as well.

**The very first screenshot**. This here:
https://user-images.githubusercontent.com/76969146/103575875-50aa3d00-4ea0-11eb-8558-4c6d20dff6b7.png

This is not your device, but the coordinator. I need it from the device to double check the code covers it.

zguy2008, 2021-01-05:
Ha!  The node info.  Sorry about that.  here:
[image]
It's indeed Sinope mfg code.

zguy2008, 2021-01-05:
I have removed and rejoined the thermostat.

[image]

I can now read all its clusters.  Even adjusted temp on it.  Seems to be communicating properly now.

I'm still not seeing it in the REST-API though.

Documented endpoints are:
[image]

None for thermostats.  I tried a few obvious possibilities but they all return 'not available resource' error.

zguy2008, 2021-01-05:
Here is the basic cluster of the thermostat

[image]

SwoopX, 2021-01-06:
Ok, thanks. So it appears the integration was more than rudimentary back in the days...
Anyway, I can take care of that. In the meantime, you could try to make the sensors appear. For that, please enter sensor search in Phoscon. While it is searching, please select the thermostat cluster and press the read button. That should hopefully trigger sensor creation. Repeat the same for temperature measurement and electrical measurement on success. If yes, I can guide you on how you continuously get the data in and updated, so let me know.

Also, I'd appreciate a screenshot of the Sinope specific cluster, after you've read the attributes. Thanks!

zguy2008, 2021-01-06:
Here is the Sinope specific cluster:
[image]

I'll attempt the rest and get back to you.

zguy2008, 2021-01-06:
A sensor was added to Phoscan when reading basic cluster:
[image]
[image]

However, reading all the other clusters didn't add anything.
I also tried adding a switch reading all clusters.  No results there either.

SwoopX, 2021-01-06:
This look really good. You should check the REST API, it should contain 3 sensors for your thermostat, based on your screenshot.

Now, you need to be my lab rat to enhance device support here. I need additional screenshots of some clusters where the attributes have been read, namely:

- electrical measurement cluster
- thermostat cluster

The thermostat cluster is somehow special in that sense, that it doesn't read all attributes on a button press unfortunately, since its too many. It stops around attribute 0x001C. For all remaining, you must double click on it and read it manually, I'm afraid.

What you could also try upfront is, in thermostat user interface configuration cluster, attribute 0x0001 (keypad lockout), set it to `level 1 lockout`. This should activate the child lock.

zguy2008, 2021-01-06:
> This look really good. You should check the REST API, it should contain 3 sensors for your thermostat, based on your screenshot.

Correct.
type": "ZHATemperature
type": "ZHAPower
type": "ZHAThermostat

> 
> Now, you need to be my lab rat to enhance device support here. I need additional screenshots of some clusters where the attributes have been read, namely:

Anything I can help with you'll get.

> 
> * electrical measurement cluster

[image]

> * thermostat cluster

[image]
[image]
[image]
[image]
[image]
[image]
[image]

> 
> The thermostat cluster is somehow special in that sense, that it doesn't read all attributes on a button press unfortunately, since its too many. It stops around attribute 0x001C. For all remaining, you must double click on it and read it manually, I'm afraid.
> 

I'm not seeing 0x001C.  Double clicks doesn't work with dimmed items nor the read button.  However, it works on non-dimmed items and brings a Attribute editor window.

> What you could also try upfront is, in thermostat user interface configuration cluster, attribute 0x0001 (keypad lockout), set it to `level 1 lockout`. This should activate the child lock.

That works.  There are 5 levels available along with 'No Lockout'.  The lock icon appears on the thermostat when writing one of the levels.

SwoopX, 2021-01-06:
Thanks a lot. For the thermostat cluster, you might have missed the scrollbar where the attributes show up at the bottom. There like roughly 50 attributes listed when you scroll down (you can skip the 0x0040X - AC related attributes however).

I'll prepare everything so far based on the info given so far.

zguy2008, 2021-01-06:
@SwoopX here are the extra settings.  I had not noticed the second scroll bar.

[image]
[image]
[image]
[image]
[image]
[image]

closed 2021-01-17 by manup

MattL0, 2021-02-16:
Just a question. I need to purchase 6* ‘120v’ thermostat. Not decided yet on zigbee/zwave version of sinope or stelpro .
what are all the info the api is returning for this thermostat? Is it push or poll?

And what controls do you have ?

Thank you. @zguy2008

zguy2008, 2021-02-17:
> Just a question. I need to purchase 6* ‘120v’ thermostat. Not decided yet on zigbee/zwave version of sinope or stelpro .
> what are all the info the api is returning for this thermostat? Is it push or poll?
> 
> And what controls do you have ?
> 
> Thank you. @zguy2008
Hi @MattL0 

Not sure what info you have on hand but here is what I know.

Communication: Sinope offers Zigbee and Wi-Fi options.  Stelpro offers Z-Wave, Zigbee and Wi-Fi options.

Voltage: both Sinope and Stelpro can handle 120/240 volts.  I can't help wondering why you would run heating on 120v though.

Load: The Sinope TH13xxZB are for heated floor control.  They do have thermostats for baseboard heaters as well. I'm not aware that Stelpro supports heated floors.  If they did have heated floor Z-Wave thermostats, that's what I'd be using.

While I use several technologies, Homeseer is the hearth of my system, which is Z-Wave based.  The only reason I'm using Sinope thermostats is they are the only ones available for heated floors in North America.  Until recently, I could not control these thermostats with Homeseer.  But now, a Homeseer plugin named JowieHue supporting the deCONZ Zigbee gateway has allowed a partial control of the Sinope Zigbee thermostat within Homeseer.

However, that said, even with the last stable update of deCONZ recently released, the API response for the 3 thermostats I have differs,  I still have to analyse the differences but it's definitely inconsistent at this point.  I feel like deCONZ may need a couple more updates before everything works properly.  Not that I'm complaining.  I now have basic control of the thermostats where I wanted it.  It just would be nice if all thermostat functionalities would be available, notwithstanding the thermostat version.  I'll have to work with Dresden for that but the fact they updated the Sinope thermostat in the latest release means they are committed to improve.

Hope that helped.  

G.

MattL0, 2022-03-03:
Hey thanks for your complete answer. I am really sorry for the delay , I’ve just seen your reply.

I am also using jowihue/HomeSeer. 

Do you know if the latest deconz have better support now vs 1 year ago? 

I guess I could just get one from amazon , and if it does work I can easily chip it back. 

what are you 3 thermostats models?

MattL0, 2022-03-03:
They now have TH1500ZB model. I need to remove my thermostat to know if it is dual or single pole.

MattL0, 2022-03-03:
It looks like some improvement have been made in home assistant ( will be inserted later in zha component) .
[
](https://github.com/claudegel/sinope-zha)