dresden-elektronik/deconz-rest-plugin #4102: Last Working Version is 26570700 (27th April 2020)

jaruba, 2021-01-04:
## Describe the bug

The last Conbee II version that works for me is `26570700` (27th April 2020), all version after that (5 versions) refuse to work. By this I mean that the device fails to connect.

The only other previous version I know worked is `264a0700` (previous version from the one mentioned above)

I avoided updating for month because of this issue, decided to update now, tried all of them versions by flashing from Win 10, restarting the NUC device, waiting, checking the Deconz component. (I get the "Network was not found ..." error)

## Steps to reproduce the behavior

Every time I try to update above version `26570700` (from Win 10, or from the Deconz component UI), the device stops being able to connect.

## Expected behavior

I would expect it to continue connecting with a newer version too..

## Screenshots

[image]

## Environment

 - Host system: `NUC10i5FNH2`
 - Running method: Home Assistent deCONZ Add-on (version 6.6.2) / HA 2020.12.2 / HA Supervisor 2020.12.7 / Docker 20.10.1 (HA, HA Supervisor and deCONZ add-on running in Docker) / Ubuntu 20.04.1 LTS 
 - Firmware version: 26570700 (last working version)
 - deCONZ version: 6.6.2 (deCONZ Add-on version)
 - Device: ConBee II 
 - Do you use an USB extension cable: (yes / no) -- yes
 - Is there any other USB or serial devices connected to the host system? Yes. If so: Wireless Keyboard (with Touchpad)

## deCONZ Logs

```
20:21:30:810 , 08:48:53
20:21:32:804 COM check bootloader
20:21:32:815 
R21B18 Bootloader
20:21:32:816 Vers: 2.07
20:21:32:816 build: Jun 17 2019
20:21:32:816 , 08:48:53
20:21:32:818 device disconnected reason: 5, index: 0
20:21:33:741 wait reconnect 15 seconds
20:21:33:778 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_[censored]-if00 / serialno: [censored], ConBee II
20:21:34:741 wait reconnect 14 seconds
172.30.32.2 - - [04/Jan/2021:20:21:35 +0200] "GET /api/config?_=1609784496290 HTTP/1.1" 200 263 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [04/Jan/2021:20:21:35 +0200] "GET /api/hassio_ingress/[censored]/api/config?_=1609784496309 HTTP/1.1" 403 140 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
20:21:35:741 failed to reconnect to network try=2
20:21:35:742 wait reconnect 13 seconds
20:21:36:742 wait reconnect 12 seconds
20:21:37:741 wait reconnect 11 seconds
20:21:38:741 wait reconnect 10 seconds
20:21:39:741 wait reconnect 9 seconds
20:21:40:741 failed to reconnect to network try=3
20:21:40:742 wait reconnect 8 seconds
20:21:41:741 wait reconnect 7 seconds
172.30.32.2 - - [04/Jan/2021:20:21:42 +0200] "GET /api/config?_=1609784503388 HTTP/1.1" 200 263 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
20:21:42:742 wait reconnect 6 seconds
172.30.32.2 - - [04/Jan/2021:20:21:42 +0200] "GET /api/hassio_ingress/[censored]/api/config?_=1609784503410 HTTP/1.1" 403 140 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
20:21:43:741 wait reconnect 5 seconds
20:21:44:741 wait reconnect 4 seconds
20:21:45:742 failed to reconnect to network try=4
20:21:45:742 wait reconnect 3 seconds
20:21:46:742 wait reconnect 2 seconds
20:21:47:741 wait reconnect 1 seconds
20:21:47:777 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_[censored]-if00 / serialno: [censored], ConBee II
20:21:47:789 COM check bootloader
20:21:47:801 
R21B18 Bootloader
20:21:47:801 Vers: 2.07
20:21:47:801 build: Jun 17 2019
20:21:47:801 , 08:48:53
20:21:49:796 COM check bootloader
20:21:49:809 
R21B18 Bootloader
20:21:49:809 Vers: 2.07
20:21:49:809 build: Jun 17 2019
20:21:49:809 , 08:48:53
172.30.32.2 - - [04/Jan/2021:20:21:49 +0200] "GET /api/config?_=1609784510486 HTTP/1.1" 200 263 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [04/Jan/2021:20:21:49 +0200] "GET /api/hassio_ingress/[censored]/api/config?_=1609784510507 HTTP/1.1" 403 140 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
20:21:50:741 failed to reconnect to network try=5
20:21:51:798 COM check bootloader
20:21:51:809 
R21B18 Bootloader
20:21:51:809 Vers: 2.07
20:21:51:809 build: Jun 17 2019
20:21:51:809 , 08:48:53
20:21:53:803 COM check bootloader
20:21:53:815 
R21B18 Bootloader
20:21:53:815 Vers: 2.07
20:21:53:815 build: Jun 17 2019
20:21:53:815 , 08:48:53
20:21:55:742 failed to reconnect to network try=6
20:21:55:807 COM check bootloader
20:21:55:818 
R21B18 Bootloader
20:21:55:818 Vers: 2.07
20:21:55:818 build: Jun 17 2019
20:21:55:819 , 08:48:53
172.30.32.2 - - [04/Jan/2021:20:21:56 +0200] "GET /api/config?_=1609784517598 HTTP/1.1" 200 263 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [04/Jan/2021:20:21:56 +0200] "GET /api/hassio_ingress/[censored]/api/config?_=1609784517617 HTTP/1.1" 403 140 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
20:21:57:810 COM check bootloader
20:21:57:821 
R21B18 Bootloader
20:21:57:821 Vers: 2.07
20:21:57:821 build: Jun 17 2019
20:21:57:821 , 08:48:53
20:21:59:816 COM check bootloader
20:21:59:828 
R21B18 Bootloader
20:21:59:828 Vers: 2.07
20:21:59:828 build: Jun 17 2019
20:21:59:829 , 08:48:53
20:21:59:831 device disconnected reason: 5, index: 0
20:22:00:742 failed to reconnect to network try=7
20:22:00:743 wait reconnect 15 seconds
20:22:00:758 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_[censored]-if00 / serialno: [censored], ConBee II
20:22:01:741 wait reconnect 14 seconds
20:22:02:741 wait reconnect 13 seconds
20:22:03:741 wait reconnect 12 seconds
172.30.32.2 - - [04/Jan/2021:20:22:04 +0200] "GET /api/config?_=1609784524690 HTTP/1.1" 200 263 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [04/Jan/2021:20:22:04 +0200] "GET /api/hassio_ingress/[censored]/api/config?_=1609784524707 HTTP/1.1" 403 140 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
20:22:04:742 wait reconnect 11 seconds
20:22:05:742 failed to reconnect to network try=8
20:22:05:742 wait reconnect 10 seconds
20:22:06:741 wait reconnect 9 seconds
20:22:07:741 wait reconnect 8 seconds
20:22:08:741 wait reconnect 7 seconds
20:22:09:741 wait reconnect 6 seconds
20:22:10:741 failed to reconnect to network try=9
20:22:10:741 wait reconnect 5 seconds
172.30.32.2 - - [04/Jan/2021:20:22:11 +0200] "GET /api/config?_=1609784531771 HTTP/1.1" 200 263 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [04/Jan/2021:20:22:11 +0200] "GET /api/hassio_ingress/[censored]/api/config?_=1609784531780 HTTP/1.1" 403 140 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
```

## Additional context

The Deconz add-on config:
```
device: >-
  /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_[censored]-if00
```

Mimiix, 2021-01-04:
What guide did you follow to update firmware on windows?

Also, is that a vm?

jaruba, 2021-01-04:
@Mimiix I used this guide: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-on-a-system-where-deconz-is-already-installed

Every time I updated / downgraded it said `SUCCESS`.

The definition of a Docker container (according to [this official article](https://www.docker.com/resources/what-container)) is:
> Containers are an abstraction at the app layer that packages code and dependencies together. Multiple containers can run on the same machine and share the OS kernel with other containers, each running as isolated processes in user space. Containers take up less space than VMs (container images are typically tens of MBs in size), can handle more applications and require fewer VMs and Operating systems.

As far as I know, the HA Deconz Add-on can only run in a Docker container, there is no alternative to it.

Mimiix, 2021-01-04:
Asked Manuel to check up :)

jaruba, 2021-01-04:
Oh, and I'm using GCFFlasher v3_14 if that helps

manup, 2021-01-04:
Hmm strange the latest ConBee II firmware version 0x26680700 should be way more robust than these old versions.

From the logs it appears that the bootloader is cycling, like when the firmware isn't installed fully, can you please make a test on Windows by updating to latest version and test if it connects in deCONZ on Windows (just to be sure firmware boots up).

jaruba, 2021-01-04:
Alright, I flashed the latest version, I tried using `-x 3` too to get more logs, but it didn't seem to show more logs..

> $ GCFFlasher -d COM3 -t 60 -f ../firmware/deCONZ_ConBeeII_0x26680700.bin.GCF -x 3
GCFFlasher V3_14 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device COM3 (ConBee II)
deCONZ firmware version 26570700
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 164373 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts

I then checked it in the Deconz app on Windows:
[image]

Then plugged it into the NUC and started the NUC. **This time it worked!**

[image]

I tried 5 different version before this, and none of them worked, I tried this latest version 3 different times and it also didn't work. I also tried updating in the past and had the same issues. I'm lost as to what made it work this time..

 The only thing I can think of is using `-x 3` and updating from `26570700` to `26680700` with Windows. Initially I updated from `264a0700` through the "update" button in the UI. When that failed I started upgrading / downgrading from Windows to each version.
 
 I really don't know what fixed it.. I lost a day trying to make it work..
 
 Regardless, I won't loose more of your time.

closed 2021-01-04 by jaruba

manup, 2021-01-05:
Cool, glad that it works now. I'm not sure why it didn't work before, the -x 3 parameter is only for debug prints but shouldn't make any difference here.

Mimiix, 2021-01-05:
> Regardless, I won't loose more of your time.

No worries! Happy to help out.

bschenzle, 2021-01-09:
Hi, 

I have a really weird behaviour and probably after 3 years of a sucessfully ConBee Device i think it got broken somehow. Anyway I spent today at least the whole day and couldn solve the issue by myself. 
**My behaviour is the follow:**
1. As soon as I upgraded to deconz to deconz-2.09.00-qt5.deb the Phoscon App was not showing any information of the Vendor as well as Firmware is telling Not Connected. 
[image]

2. It is also not appearing withing the Deconz - GUI anymore
[image]
Till this point I read a couple of comments that some people went to a previous version deconz-2.07.01-qt5.deb for example and downgraded their firmware version. I follow this steps and downgraded my firmware Verion as well es went to the latest stable version (deconz-2.07.01-qt5.deb). As soon as I startet Phoscon I receive the error message network was not found.... and I am not able to do anything. 

3. I thought I got crashed something on my Pi and I completely reinstalled Raspian OS as well as Deconz but it is not changing anything in the behaviour

[image]
Tried a couple of different Firmare Versions. Flashing of the Stick works without error but it is not solving my issue so far.
[image]
[image]

4. I connected my stick to a Win 10 PC and the ConBee-Stick does not even get recognized anymore within Deconz as well as the GCFFlasher.
[image]
[image]
[image]
So I stopped on Win and did not investigate even further.

5. Last but not least I installed a preinstalled Phoscon Raspian OS on a other SD Card and just executed Phoscon.
On the preinstalled OS even the latest Firmare is runing on 2.07.00 and looks fine.
[image]
[image]
[image]

Any thoughts on that? To be honest I dont know why I can not run it on a pure Raspian OS. Would appreciate your comments.

Thank you