dresden-elektronik/deconz-rest-plugin #4104: GCFflasher does not upgrade conbee firmware,  could not sync with bootloader

uusul, 2021-01-04:
## Describe the bug
Hi Dresden-elektronik,

I am having an issue upgrading the Conbee firmware using the [manual method](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-windows). GCFflasher does not upgrade conbee firmware, it throws: **could not sync with bootloader**

I would appreciate if you could help me. The conbee is a great device and is a perfect way to automate my Zigbee network.

## Steps to reproduce the behavior
- I've tried with GCFflasher packed in the deconz software.
- I've also tried with GCFflasher 3.10 dowloaded from [here](http://deconz.dresden-elektronik.de/win/)

## Expected behavior
Conbee to be upgraded with new firmware.

## Screenshots

## Environment
 - Host system: windows
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: 0x260b0500
 - deCONZ version: 2.08.00
 - Device: ConBee 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
```
C:\_local\GCFFlasher_Win_V3_10>GCFFlasher -d \\.\COM3 -t 120 -f deCONZ_Rpi_0x26390500.bin.GCF -x 3
GCFFlasher V3_10 (c) dresden elektronik ingenieurtechnik gmbh
using firmware file: deCONZ_Rpi_0x26390500.bin.GCF
ls dev: \\.\COM3 (0x0403/0x6015) sn: DM01D08RA
Reboot device \\.\COM3 (ConBee)
serial port open device
deCONZ firmware version 260B0500
set watchdog ttl status: 0x04
close serial port
uart reset failed, try FTDI reset
action: reset device FTDI
reset FTDI DM01D08RA (dev: 0)
GCF_ResetDeviceDone
action: update firmware after 9 ms
serial port open device
STARTING APP
 (380 ms)
STARTING APP
?♫☺ (475 ms)
STARTING APP
?♫☺ (2621 ms)
could not sync with bootloader
close serial port
retry, elapsed 9 seconds
parse commands: GCFFlasher -d \\.\COM3 -t 120 -f deCONZ_Rpi_0x26390500.bin.GCF -x 3

using firmware file: deCONZ_Rpi_0x26390500.bin.GCF
ls dev: \\.\COM3 (0x0403/0x6015) sn: DM01D08RA
Reboot device \\.\COM3 (ConBee)
serial port open device
```
## Additional context

SwoopX, 2021-01-05:
You need to close deconz before flashing. Have you done that?

uusul, 2021-01-05:
> You need to close deconz before flashing. Have you done that?

Hi SwoopX,
Yes I have done that. (Trying to flash with deconz running resulted in "serial port open failed, status = access denied")

Mimiix, 2021-01-05:
Can you try with a extension cable?

Smanar, 2021-01-05:
During you test on windows, was a real machine ?
No other stuff on USB port ?

Mimiix, 2021-01-05:
Did you run the Command line as admin?

Smanar, 2021-01-05:
And where have you found this path ? Just `GCFFlasher -d COM3..... ` not working ?

SwoopX, 2021-01-05:
Version 3.10 is also a little bit outdated. You might want to try the latest Windows version (2.09.00).