dresden-elektronik/deconz-rest-plugin #4105: Network not found - Me too - can't control my home

prestonsmartuk, 2021-01-05:
## Describe the bug

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on 
 - deCONZ version: (2.xx.yy)
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes

## deCONZ Logs

prestonsmartuk, 2021-01-05:
I don't know how to work Github, I'm new here and new to all of this.  All my lights have stopped working and I need to get back up and running now.  

LOG FILE
172.30.32.2 - - [05/Jan/2021:20:04:12 +0000] "GET /api/8FE2467DF1/config?_=1609876770700 HTTP/1.1" 200 2906 "http://homeassistant:8123/api/hassio_ingress/ILVdpf1XuEvC-qqplaobgDBBcIu7_-mVtrNx-lM0qIE/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [05/Jan/2021:20:04:12 +0000] "GET /api/8FE2467DF1/lights?_=1609876770701 HTTP/1.1" 200 12 "http://homeassistant:8123/api/hassio_ingress/ILVdpf1XuEvC-qqplaobgDBBcIu7_-mVtrNx-lM0qIE/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [05/Jan/2021:20:04:12 +0000] "GET /api/8FE2467DF1/sensors?_=1609876770702 HTTP/1.1" 200 4410 "http://homeassistant:8123/api/hassio_ingress/ILVdpf1XuEvC-qqplaobgDBBcIu7_-mVtrNx-lM0qIE/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [05/Jan/2021:20:04:12 +0000] "GET /api/8FE2467DF1/config?_=1609876770703 HTTP/1.1" 200 2906 "http://homeassistant:8123/api/hassio_ingress/ILVdpf1XuEvC-qqplaobgDBBcIu7_-mVtrNx-lM0qIE/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [05/Jan/2021:20:04:17 +0000] "GET /api/8FE2467DF1/config?_=1609876770704 HTTP/1.1" 200 2906 "http://homeassistant:8123/api/hassio_ingress/ILVdpf1XuEvC-qqplaobgDBBcIu7_-mVtrNx-lM0qIE/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [05/Jan/2021:20:04:22 +0000] "GET /api/8FE2467DF1/config?_=1609876770705 HTTP/1.1" 200 2906 "http://homeassistant:8123/api/hassio_ingress/ILVdpf1XuEvC-qqplaobgDBBcIu7_-mVtrNx-lM0qIE/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
20:03:54:081 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
20:03:54:097 CTRL. 3.27.220:03:54:154 COM: use stable device path /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001949C994-if00
20:03:54:208 dev /dev/ttyAMA0
20:03:54:208 COM: /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001949C994-if00 / serialno: __0X00124B001949C994, ConBee II
20:03:54:209 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
20:03:54:349 parent process s6-supervise
20:03:54:349 gw run mode: docker/hassio
20:03:54:349 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
20:03:54:350 sd-card cid: 035344534333324780ffffffff014100
20:03:54:351 DB sqlite version 3.27.2
20:03:54:353 DB PRAGMA page_count: 33
20:03:54:353 DB PRAGMA page_size: 4096
20:03:54:353 DB PRAGMA freelist_count: 0
20:03:54:353 DB file size 135168 bytes, free pages 0
20:03:54:354 DB PRAGMA user_version: 7
20:03:54:354 DB cleanup
20:03:54:354 DB create temporary views
20:03:54:365 started websocket server at port 8081
20:03:54:368 [INFO] - Found file containing button maps. Parsing data...
20:03:54:372 [WARNING] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
20:03:54:379 [INFO] - Button maps loaded.
20:03:54:381 dlg action: Read binding table
20:03:54:381 found node plugin: libde_rest_plugin.so - REST API Plugin
20:03:54:384 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
20:03:56:881 OTAU: create 1189-006F-02056550.zigbee
20:03:57:068 OTAU: create 1189-0040-00103103.zigbee
20:03:57:102 OTAU: create 1189-003B-00103103.zigbee
20:03:57:133 OTAU: create 1189-0020-00102428.zigbee
20:03:57:163 OTAU: create 1189-005C-00103101.zigbee
20:03:57:190 OTAU: create 1189-0063-00103201.zigbee
20:03:57:335 OTAU: create 1189-008D-02056550.zigbee
20:03:57:563 OTAU: create 110C-0005-01020510.zigbee
20:03:57:583 OTAU: create 110C-0007-01020510.zigbee
20:03:57:605 OTAU: create 110C-005C-01020510.zigbee
20:03:57:628 OTAU: create 110C-005B-01020510.zigbee
20:03:57:648 OTAU: create 110C-0065-01020510.zigbee
20:03:57:671 OTAU: create 110C-0069-01020510.zigbee
20:03:57:693 OTAU: create 110C-006E-01020510.zigbee
20:03:57:715 OTAU: create 110C-0068-01020510.zigbee
20:03:57:738 OTAU: create 110C-006A-01020510.zigbee
20:03:57:743 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
20:03:57:795 dev /dev/ttyAMA0
20:03:57:795 COM: /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001949C994-if00 / serialno: __0X00124B001949C994, ConBee II
20:03:57:929 created username: 8FE2467DF1, devicetype: Home Assistant
20:03:58:863 dev /dev/ttyAMA0
20:03:58:864 COM: /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001949C994-if00 / serialno: __0X00124B001949C994, ConBee II
20:03:58:891 COM check bootloader
20:03:59:749 dev /dev/ttyAMA0
20:03:59:994 Announced to internet https://phoscon.de/discover
20:03:59:995 discovery server date: Tue, 05 Jan 2021 20:03:59 GMT
20:03:59:995 	 local time seems to be ok
20:03:59:996 discovery found version 2.04.35 for update channel stable
20:04:00:901 COM timout query bootloader, assume application
20:04:01:395 New websocket 127.0.0.1:45872 (state: 3) 
20:04:02:487 device state timeout ignored in state 2
[... cut ...]
172.30.32.2 - - [05/Jan/2021:20:05:27 +0000] "GET /api/8FE2467DF1/config?_=1609876770728 HTTP/1.1" 200 2906 "http://homeassistant:8123/api/hassio_ingress/ILVdpf1XuEvC-qqplaobgDBBcIu7_-mVtrNx-lM0qIE/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [05/Jan/2021:20:05:32 +0000] "GET /api/8FE2467DF1/config?_=1609876770729 HTTP/1.1" 200 2906 "http://homeassistant:8123/api/hassio_ingress/ILVdpf1XuEvC-qqplaobgDBBcIu7_-mVtrNx-lM0qIE/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"

Mimiix, 2021-01-05:
This is not how the template works.

Read and you shall find.

closed 2021-01-05 by Mimiix