dresden-elektronik/deconz-rest-plugin #4106: Time-triggered rules don't fire during database commit

ebaauw, 2021-01-05:
## Describe the bug

A commit of the database blocks deCONZ, causing time-triggered rules (condition on `/config/localtime`, or `ddx` condition) to fail.  There's a discontinuity in the `/config/localtime` events.  I've been suspecting this for weeks, but finally caught the bug red-handed:
```
Jan  5 22:59:57 pi2 deCONZ[26232]: 22:59:56:657 rule event /config/localtime: 22:59:55.651 -> 22:59:56.651 (1000ms)
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:657 rule event /config/localtime: 22:59:56.651 -> 22:59:57.651 (1000ms)
Jan  5 23:00:00 pi2 deCONZ[26232]: 23:00:00:058 rule event /config/localtime: 23:00:00.047 -> 23:00:00.057 (10ms)
Jan  5 23:00:00 pi2 deCONZ[26232]: 23:00:00:070 rule event /config/localtime: 23:00:00.047 -> 23:00:00.057 (10ms)
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:00:660 rule event /config/localtime: 23:00:00.057 -> 23:00:00.652 (595ms)
Jan  5 23:00:03 pi2 deCONZ[26232]: 23:00:01:660 rule event /config/localtime: 23:00:00.652 -> 23:00:01.656 (1004ms)
Jan  5 23:00:03 pi2 deCONZ[26232]: 23:00:02:671 rule event /config/localtime: 23:00:01.656 -> 23:00:02.651 (995ms)
```

## Steps to reproduce the behavior

That's the hard part.  Window for this to happen is a couple of seconds.

## Expected behavior

Rules should be executed.

## Screenshots

N/a

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: All
 - deCONZ version: 2.9.0, but quite some versions already, I suspect.
 - Device: All

## deCONZ Logs

```
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:657 rule event /config/localtime: 22:59:56.651 -> 22:59:57.651 (1000ms)
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:702 Poll APS confirm 186 status: 0x00
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:703 Erase task req-id: 186, type: 19 zcl seqno: 115 send time 0, profileId: 0x0104, clusterId: 0x0008
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:704 APS-DATA.confirm id: 186, status: 0x00 SUCCESS
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:706 APS-DATA.confirm request id: 186 -> erase from queue
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:708 read attributes of 0x0017880100F07118 cluster: 0x0300: [ 22:59:57:709 0x0001 22:59:57:710 0x0003 22:59:57:711 0x0004 22:59:57:711 0x0007 22:59:57:712 0x0008 22:59:57:713 0x4000 22:59:57:713 0x4001 22:59:57:715 0x4002 22:59:57:715 ]
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:716 add task 1748476 type 19 to 0x0017880100F07118 cluster 0x0300 req.id 188
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:717 Poll APS request 188 to 0x0017880100F07118 cluster: 0x0300
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:735 aps request id: 186 finished, erase from queue
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:745 APS-DATA.indication srcAddr: 0x37b2, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -46
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:749 #011asdu: 18730100000020fe
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:752 Node data 0x0017880100f07118 profileId: 0x0104, clusterId: 0x0008
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:756 0x0017880100F07118: update ZCL value 0x0B/0x0008/0x0000 after 0 s
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:760 APS-DATA.request id: 188, addrmode: 0x03, addr: 0x0017880100f07118, profile: 0x0104, cluster: 0x0300, ep: 0x01 -> 0x0B queue: 1 len: 19 tx.options 0x00
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:831 Poll APS confirm 188 status: 0x00
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:833 Erase task req-id: 188, type: 19 zcl seqno: 116 send time 0, profileId: 0x0104, clusterId: 0x0300
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:836 APS-DATA.confirm id: 188, status: 0x00 SUCCESS
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:838 APS-DATA.confirm request id: 188 -> erase from queue
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:895 aps request id: 188 finished, erase from queue
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:901 APS-DATA.indication srcAddr: 0x37b2, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 255, rssi: -46
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:904 #011asdu: 187401010000208f0300002171750400002110690700002171010800003002004000216b3a01400030020240002000
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:910 Node data 0x0017880100f07118 profileId: 0x0104, clusterId: 0x0300
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:914 0x0017880100F07118: update ZCL value 0x0B/0x0300/0x0001 after 0 s
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:917 0x0017880100F07118: update ZCL value 0x0B/0x0300/0x0003 after 0 s
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:919 0x0017880100F07118: update ZCL value 0x0B/0x0300/0x0004 after 0 s
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:925 0x0017880100F07118: update ZCL value 0x0B/0x0300/0x0007 after 0 s
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:926 0x0017880100F07118: update ZCL value 0x0B/0x0300/0x0008 after 0 s
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:927 0x0017880100F07118: update ZCL value 0x0B/0x0300/0x4000 after 0 s
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:928 0x0017880100F07118: update ZCL value 0x0B/0x0300/0x4001 after 0 s
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:57:929 0x0017880100F07118: update ZCL value 0x0B/0x0300/0x4002 after 0 s
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:58:375 APS-DATA.request id: 194, addrmode: 0x03, addr: 0x0017880100cb6379, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x04
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:58:452 APS-DATA.confirm id: 194, status: 0x00 SUCCESS
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:58:454 APS-DATA.confirm request id: 194 -> confirmed, timeout 38912920
Jan  5 22:59:58 pi2 deCONZ[26232]: 22:59:58:472 poll node 00:17:88:01:10:27:c9:3d-0b
[... 78 lines cut ...]
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:100 Erase task req-id: 210, type: 19 zcl seqno: 118 send time 0, profileId: 0x0104, clusterId: 0x0006
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:101 APS-DATA.confirm id: 210, status: 0x00 SUCCESS
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:102 APS-DATA.confirm request id: 210 -> erase from queue
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:172 binding for attribute reporting of ep: 0x02 cluster 0x0400 seems to be active
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:173 binding for attribute reporting of ep: 0x02 cluster 0x0000 seems to be active
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:174 binding for attribute reporting of ep: 0x02 cluster 0x0406 seems to be active
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:175 binding for attribute reporting of ep: 0x02 cluster 0x0402 seems to be active
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:177 aps request id: 210 finished, erase from queue
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:225 APS-DATA.indication srcAddr: 0x5391, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -44
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:226 #011asdu: 1876010000001000
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:228 Node data 0x001788011052adc9 profileId: 0x0104, clusterId: 0x0006
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:229 enqueue event attr/lastseen for /lights/201
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:230 0x001788011052ADC9: update ZCL value 0x0B/0x0006/0x0000 after 0 s
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:231 enqueue event event/checkgroupanyon for /groups/200
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:232 enqueue event event/checkgroupanyon for /groups/2
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:232 enqueue event event/checkgroupanyon for /groups/65520
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:234 Websocket 127.0.0.1:44712 send message: {"attr":{"lastseen":"2021-01-05T22:00Z"},"e":"changed","id":"201","r":"lights","t":"event","uniqueid":"00:17:88:01:10:52:ad:c9-0b"} (ret = 131)
Jan  5 23:00:01 pi2 deCONZ[26232]: 23:00:01:236 Websocket 127.0.0.1:44718 send message: {"attr":{"lastseen":"2021-01-05T22:00Z"},"e":"changed","id":"201","r":"lights","t":"event","uniqueid":"00:17:88:01:10:52:ad:c9-0b"} (ret = 131)
Jan  5 23:00:01 pi2 CRON[20187]: (pi) CMD (Applications/cputemp.command)
Jan  5 23:00:02 pi2 dc_eventlog[793]: /lights/271: {"lastseen":"2021-01-05T22:00Z"}
Jan  5 23:00:03 pi2 deCONZ[26232]: 23:00:01:651 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
Jan  5 23:00:03 pi2 deCONZ[26232]: 23:00:01:656 enqueue event config/localtime for /config/
Jan  5 23:00:03 pi2 deCONZ[26232]: 23:00:01:660 rule event /config/localtime: 23:00:00.652 -> 23:00:01.656 (1004ms)N
```
## Additional context

manup, 2021-01-06:
Since the `idleTimerFired()` is called by Qt timer in a fixed 1 second interval, I suspect that when main thread is blocked during database write, Qt schedules the timer callback immediately, even multiple times due the  < 1 sec executions.

Enqueing the `config/localtime` event is not guarded against this yet, I have a simple fix for this in mind, will report back.

SwoopX, 2021-03-03:
@ebaauw Is that resolved meanwhile?

ebaauw, 2021-03-03:
I haven't experienced any missed triggering of rules since v2.9.0, but it was hard to reproduce in the first place.

closed 2022-10-15 by ebaauw