dresden-elektronik/deconz-rest-plugin #4108: Is the Philips Hue Sana Basis BT compatible with deconz and phoscon?

ChrisPrefect, 2021-01-05:
It is not listed in the compatibility list on phoscon.de.

But maybe it acts just like any other RGBW hue light?

https://www.digitec.ch/de/s1/product/philips-hue-sana-basis-bt-wandleuchte-deckenleuchte-11235552

Thanks!

It makes absolutely no sense that the issue #4107 was closed!

I did provide ALL the necessary information! The template does not make any sense for this question! Why is this support page so hostile agains users?!

## Describe the question or issue you are having

## Screenshots

## Environment

 

## deCONZ Logs

## Additional context

Mimiix, 2021-01-05:
This is not how the issue template works.

Closing.

closed 2021-01-05 by Mimiix

ChrisPrefect, 2021-01-06:
So how does it work then???