dresden-elektronik/deconz-rest-plugin #4109: Is the Philips Hue Sana Basis BT compatible with deconz and phoscon?

ChrisPrefect, 2021-01-06:
## Describe the question or issue you are having

It is not listed in the compatibility list on phoscon.de.

But maybe it acts just like any other RGBW hue light?

https://www.digitec.ch/de/s1/product/philips-hue-sana-basis-bt-wandleuchte-deckenleuchte-11235552

Thanks!

## Screenshots
[image]

## Environment

 - Host system: rp4
 - Running method: Raspbian 
 - Firmware version: 26680700
 - deCONZ version: 2.09.00 / 23.12.2020
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

no logs

## Additional context

no additional context

carver7, 2021-01-07:
I believe compatibility for a device that is not currently listed in supported device list is unknown. You can request support, which is ￼ potentially more likely to happen if you include screen captures the values displayed for this device in the Phoscon application while you're trying to pair it with your Conbee. ￼￼

wvuyk, 2021-01-07:
And as it is a light it is probably supported out of the box. Just connect it and I expect it is just working like any other Philips Hue light?

Mimiix, 2021-01-07:
I agree with @carver7 and @wvuyk 

Experience with Hue devices is good. They often work out of the box without problems.

closed 2021-03-05 by github-actions[bot]