dresden-elektronik/deconz-rest-plugin #4113: iluminize 511.514 (ZGRC-TEUR-005 8)

vogtgbr, 2021-01-06:
## Device

- Product name: iluminize 511.514 Wand-Dimmer
- Manufacturer: iluminize
- Model identifier: ZGRC-TEUR-005 8
- Device type : 
  - Remote
  - Switch

## Screenshots
[image]

[image]

[image]

Mimiix, 2021-01-06:
Hi! can you provide screenshots of all clusters?

Mimiix, 2021-01-06:
Also: Thsi seems to be integrated already. What's the issue with it?

vogtgbr, 2021-01-06:
Hi!
I'm adding screenshots of clusters that contain values. Please let me know, if you need more.
The switch appears in deCONZ but not in Phoscon, so I cannot use it.

[image]
[image]
[image]

Mimiix, 2021-01-06:
aah!

That's a different thing. 

You need to ask support in the Phoscon repo. I'll move this issue to theirs.