dresden-elektronik/deconz-rest-plugin #4114: New HS1SA-M?

og9gjz567, 2021-01-06:
## Device
- Product name: HS1SA-M 
- Manufacturer: Heimann
- Model identifier: SMOK_YDLV10
- Device type :   - Sensor
## Screenshots
[image]
[image]
[image]

SwoopX, 2021-01-06:
You should have checked the supported devices list first. It is already supported.

closed 2021-01-06 by SwoopX

og9gjz567, 2021-01-06:
Thanks for your reply. But I won't get the battery status. Should I open a new issue?

Mimiix, 2021-01-06:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2373#issuecomment-595414912

That probably contains your answer.

Either way, this is not something we can fix but something specific to your issue.

SwoopX, 2021-01-06:
It might taken 1-2 hrs to get any reading...