dresden-elektronik/deconz-rest-plugin #4115: Save the Source Routing enabled-/disabled state in deconz

joggs, 2021-01-06:
## Feature request type

Save the Source Routing enabled-/disabled state in deconz, so you don't have to enable it at each restart of deconz

## Description

The enable/disable setting for Source Routing should be saved in deconz so you don't have to enter deconz and enable it each time you restart deconz. This is important as this feature has vastly improved the stability of the zigbee network.

## Considered alternatives

## Additional context

manup, 2021-01-06:
This should already be the case. The related setting can be found in `config.ini` file.

```
[source-routing]
enabled=true
max-hops=4
min-lqi=120
```

So manually enable it again shouldn't be needed.

However there is a bug in saving and restoring already discovered source routes, so often after a restart the routes need to be discovered again, this is under investigation.

joggs, 2021-01-06:
Ok,
I am running the deconz docker container in unraid and after each restart,
source routing is disabled again.
I will try to edit the ini file and see what happens.

joggs, 2021-01-07:
I had no [source-routing] section in config.ini.
If I alter the source routing settings in deconz and restart deconz, I still do not have a [source-routing] section in config.ini.
However, if I add your [source-routing]-section manually to the config.ini and start deconz, the settings are applied correctly.
If I then adjust the settings from within deconz, the settings are never updated in config.ini, so I guess there could be a permission problem or something in the code, updating the settings file.

v: 2.09.00 / 2020-12-23
fw: 26680700

manup, 2021-01-08:
Indeed looks like perhaps the file permissions aren't correct.

Can you please check them via `ls -l config.ini`

joggs, 2021-01-08:
I have tested different access levels while testing, right now chmod 777, and actually, things are written to the config file, but not to that section. 
I am not 100% clear about how this unraid/docker version is handling things, as the config.ini file is located in the docker at  /root/.local/share/dresden-elektronik/deCONZ/config.ini, but in unraid, you have a fileshare called appdata, where persisting settings are stored, as a docker upgrade means deleting the whole docker container and then installing it again as you probably know, so there is also a config.ini in this appdata share that looks different than the the one in /root/.local/share/dresden-elektronik/deCONZ , so not sure what happens behind the scenes here, but the changes I do to the source-routing part in deconz are not updated in any of these two config files.

bcutter, 2021-02-18:
I discovered this bug too in the UI of deCONZ, usually after updating deCONZ (HA Supervisor addon):

1) Enable Source Routing (set check mark)
2) Update/Restart deCONZ
3) Check deCONZ GUI:
-> expected: source routing is enabled and check mark is still set
-> reality: source routing is enabled but check mark is removed

stefanuytterhoeven, 2021-03-30:
I can't find any config.ini. 
Have the same problem here. Should be nice if the checked "enable source routing" would stay visible in the gui.
Now, I have no idea if it's enabled or not.

leonardpitzu, 2021-07-11:
if i manually edit `config.ini` to enable and configure the source routing section, upod reboot the file is "reset" just as if deconz is overwriting it.

Floyer007, 2022-02-19:
Had the same issue recently.
Part within config.ini was missing as well.

Probably I started deCONZ as sudo once, as the config.ini had rights for root:root.
Changed it via chown to my user and works fine now.

MadMonkey87, 2024-10-09:
Any updates on this issue?

mn-90, 2025-01-23:
I have the same issue, restart will delete those settings from the gui