dresden-elektronik/deconz-rest-plugin #4116: Button Map missing for 902010/25 (Bitron Plug)

stepman0, 2021-01-06:
## Describe the question or issue you are having

The logs of my deconz-container are spammed with the message "No button map for: 902010/25 ..."

I guess this is due to my Bitron Plugs, but I am not sure. What can I do?

## Screenshots

## Environment

 - Host system: Synology DS918+ 
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.09.00 / 23.12.2020
 - Device: ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: Which? Yes - an USV

## deCONZ Logs

`
16:15:53:732 [INFO] - No button map for: 902010/25 endpoint: 0x01 cluster: 0x0702 command: 0x0A payload[0]: 000

16:15:53:733 ZCL attribute report 0x00124B00167DDA9E for cluster: 0x0702, ep: 0x01, frame control: 0x18, mfcode: 0x0000

16:15:55:039 [INFO] - No button map for: 902010/25 endpoint: 0x01 cluster: 0x0702 command: 0x0A payload[0]: 000

16:15:55:039 ZCL attribute report 0x00124B001BFB87B1 for cluster: 0x0702, ep: 0x01, frame control: 0x18, mfcode: 0x0000

16:15:55:734 [INFO] - No button map for: 902010/25 endpoint: 0x01 cluster: 0x0702 command: 0x0A payload[0]: 000

16:15:55:734 ZCL attribute report 0x00124B001BFB87BD for cluster: 0x0702, ep: 0x01, frame control: 0x18, mfcode: 0x0000

16:15:58:536 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)

16:16:00:538 [INFO] - No button map for: 902010/25 endpoint: 0x01 cluster: 0x0702 command: 0x0A payload[0]: 000

16:16:00:539 ZCL attribute report 0x00124B00167DDA9E for cluster: 0x0702, ep: 0x01, frame control: 0x18, mfcode: 0x0000

16:16:03:527 [INFO] - No button map for: 902010/25 endpoint: 0x01 cluster: 0x0702 command: 0x0A payload[0]: 000

16:16:03:527 ZCL attribute report 0x00124B00167DDA9E for cluster: 0x0702, ep: 0x01, frame control: 0x18, mfcode: 0x0000

`

## Additional context

SwoopX, 2021-01-12:
Set DEBUG_INFO=0

stepman0, 2021-01-16:
Thanks, but the I can see almost no information. I was searching for a way to either skip the reoccurring no button map message or to add a button map.

closed 2021-03-05 by github-actions[bot]