dresden-elektronik/deconz-rest-plugin #4119: [Device Support Request] Sercomm - SZ-WTD02 Water Sensor

3Doty, 2021-01-06:
## Device

- Product name: Sercomm - SZ-WTD02 Water Sensor
- Manufacturer: Sercomm Corp.
- Model identifier: SZ-WTD02N_CAR
- Device type : 
  - Sensor

## Screenshots

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

SwoopX, 2021-01-17:
Can you please read the attributes from the power configuration cluster and provide the screenshot? Make sure the device is awake prior to reading. To wake it, push the physical button on the device 1-2 seconds before reading.

SwoopX, 2021-04-04:
Merged

closed 2021-04-04 by SwoopX